openwrt/packages #4569: busybox: need support for /etc/cron.daily so that packages may install into there

pprindeville, 2017-07-10:
Some packages (like `rrdtool` and `logrotate`) benefit from being able to install a script that gets run daily, for data-reduction, log rotation, temporary file cleanup, etc.

Having packages modify `/etc/crontabs/root` with `sed` seems perilous.  Easier to let each package have its own `/etc/cron.daily/$(PKG_NAME)` if it so desires.

pprindeville, 2017-07-10:
Maintainer: @nbd168

feckert, 2019-03-28:
@pprindeville 
Using this with the busybox cron does not work on OpenWrt as in other distribution (debian)
So I added a PR https://github.com/openwrt/packages/pull/8178 to use in OpenWrt the micrond from freifunk. This will do what you want.

karlp, 2019-03-28:
I added a script to my "first boot" scripts that sets this up.  You need to include run-parts from the busybox config, and cron.  
```
# Make sure that we have periodic cron support
# We plan to have
cron_periodic() {
	mkdir -p /etc/cron.d/daily
	mkdir -p /etc/cron.d/weekly
	grep -q -F "run-parts /etc/cron.d/daily" /etc/crontabs/root || {
		echo "crond daily entry not found, adding"
		local MYMIN=$(expr $(hexdump -n 2 -e '/2 "%u"' /dev/urandom) % 60)
		local MYHOUR=$(expr $(hexdump -n 2 -e '/2 "%u"' /dev/urandom) % 24)
		echo "$MYMIN $MYHOUR * * * run-parts /etc/cron.d/daily" >> /etc/crontabs/root
	}
	grep -q -F "run-parts /etc/cron.d/weekly" /etc/crontabs/root || {
		echo "crond weekly entry not found, adding"
		local MYMIN=$(expr $(hexdump -n 2 -e '/2 "%u"' /dev/urandom) % 60)
		local MYHOUR=$(expr $(hexdump -n 2 -e '/2 "%u"' /dev/urandom) % 24)
		local MYDAY=$(expr $(hexdump -n 2 -e '/2 "%u"' /dev/urandom) % 7)
		echo "$MYMIN $MYHOUR $MYDAY * * run-parts /etc/cron.d/weekly" >> /etc/crontabs/root
	}
}
```

neheb, 2020-02-09:
micrond was merged. closing.

closed 2020-02-09 by neheb

karlp, 2020-02-09:
to clarify, "no solution has been provided that uses the system cron, but a completely separate parallel daemon is now available, that would let you do this"