openwrt/packages #4579: acme: SSL_CERT_DIR environment variable, not supported by libcurl

PriceChild, 2017-07-12:
Maintainer: @tohojo 
Environment: LEDE Reboot (17.01.2, r3435-65eec8bd5f)

Description:

`/etc/init.d/acme start` fails with:
```
Wed Jul 12 12:20:34 2017 daemon.info run-acme[32088]: Running pre checks.
Wed Jul 12 12:20:34 2017 daemon.err uhttpd[32115]: Error: No sockets bound, unable to continue
Wed Jul 12 12:20:35 2017 daemon.info run-acme[32088]: Using stage api:https://acme-staging.api.letsencrypt.org
Wed Jul 12 12:20:35 2017 daemon.info run-acme[32088]: Standalone mode.
Wed Jul 12 12:20:42 2017 daemon.info run-acme[32088]: Registering account
Wed Jul 12 12:20:42 2017 daemon.err run-acme[32088]: Please refer to https://curl.haxx.se/libcurl/c/libcurl-errors.html for error code: 48
Wed Jul 12 12:20:43 2017 daemon.err run-acme[32088]: Can not connect to https://acme-staging.api.letsencrypt.org/directory to get nonce.
Wed Jul 12 12:20:43 2017 daemon.err run-acme[32088]: Register account Error:
Wed Jul 12 12:20:43 2017 daemon.err run-acme[32088]: Please add '--debug' or '--log' to check more details.
Wed Jul 12 12:20:43 2017 daemon.err run-acme[32088]: See: https://github.com/Neilpang/acme.sh/wiki/How-to-debug-acme.sh
Wed Jul 12 12:20:43 2017 daemon.err run-acme[32088]: Issuing cert for thedomain failed. Moving state to /etc/acme/thedomain.failed-1499862043
Wed Jul 12 12:20:43 2017 daemon.info run-acme[32088]: Running post checks (cleanup).
```

I modified the script to output the curl command failing and it was this one:

```
curl -L --silent --dump-header /etc/acme/http.header  -I --user-agent "acme.sh/2.6.5 (https://github.com/Neilpang/acme.sh)" -H "" -H "" -H "" -H "" -H "" "https://acme-staging.api.letsencrypt.org/directory"
```

Running that manually gave a different error (77)

`/usr/lib/acme/run-acme` however contains:
```
export SSL_CERT_DIR=/etc/ssl/certs
```
which gets the curl command above to return the same error, 48.

Removing the `--silent` then gives the following output:

```
# SSL_CERT_DIR=/etc/ssl/certs curl -L https://acme-staging.api.letsencrypt.org/directory
Warning: ignoring SSL_CERT_DIR environment variable, not supported by libcurl
curl: (48) Error
```

Package versions:
```
root@LEDE:~# opkg info curl
Package: curl
Version: 7.52.1-3
Depends: libc, libcurl
Status: install ok installed
Section: net
Architecture: mips_24kc
Size: 39280
Filename: curl_7.52.1-3_mips_24kc.ipk
Description: A client-side URL transfer utility
Installed-Time: 1499859900

root@LEDE:~# opkg info libcurl
Package: libcurl
Version: 7.52.1-3
Depends: libc, libmbedtls
Status: install ok installed
Section: libs
Architecture: mips_24kc
Size: 98639
Filename: libcurl_7.52.1-3_mips_24kc.ipk
Description: A client-side URL transfer library
Installed-Time: 1499859897

root@LEDE:~# curl --version
curl 7.52.1 (mips-openwrt-linux-gnu) libcurl/7.52.1 mbedTLS/2.5.1
Protocols: file ftp ftps http https 
Features: IPv6 Largefile SSL 
root@LEDE:~# opkg info acme
Package: acme
Version: 1.4-1
Depends: libc, curl, ca-certificates, openssl-util, netcat
Status: install user installed
Section: net
Architecture: all
Size: 29834
Filename: acme_1.4-1_all.ipk
Conffiles:
 /etc/config/acme 02af5865c8f29be58a584792c596390bcefe415ff09e7038c487ad698af1ada5
Description: A client for issuing Letsencrypt certificates.
Installed-Time: 1499859912
```

tohojo, 2017-07-12:
Hmm, I think this is because the fix for running acme with mbedtls-linked curl has not made it into the 17.01 branch. I'll see if I can fix that...

tohojo, 2017-08-13:
Sorry for the delay in fixing this; an updated package should be on its way :)

commit 788f17e9 (2017-08-13): acme: Fix for curl linked against mbed TLS. (#4254)

Use newest acme.sh release (2.6.8).
Remove dependency on ca-certificates and add dependency on ca-bundle.
Update environment variable.

Backport to 17.01 for compatibility with 17.01.2, but keep the old envvar so
it'll hopefully keep working for users who haven't upgraded.

Closes #4579, closes #4699.
closed 2017-08-13 by tohojo