openwrt/packages #4596: expat: 2.2.2 compile failure

jammmet, 2017-07-17:
Maintainer: @thess 
Environment: ramips, MT7620, trunk

Description:

Compile failure

```
CFLAGS="-O2 -I/lede/staging_dir/host/include -I/lede/staging_dir/host/usr/include -I/lede/staging_dir/hostpkg/include -I/lede/staging_dir/target-mipsel_24kc_musl/host/include" CPPFLAGS="-I/lede/staging_dir/host/include -I/lede/staging_dir/host/usr/include -I/lede/staging_dir/hostpkg/include -I/lede/staging_dir/target-mipsel_24kc_musl/host/include" CXXFLAGS="" LDFLAGS="-L/lede/staging_dir/host/lib -L/lede/staging_dir/host/usr/lib -L/lede/staging_dir/hostpkg/lib -L/lede/staging_dir/target-mipsel_24kc_musl/host/lib" make -j1 -C /lede/build_dir/hostpkg/expat-2.2.2  
make[4]: Entering directory `/lede/build_dir/hostpkg/expat-2.2.2'
/usr/bin/env bash ./libtool --verbose --mode=compile gcc -I./lib -I. -O2 -I/lede/staging_dir/host/include -I/lede/staging_dir/host/usr/include -I/lede/staging_dir/hostpkg/include -I/lede/staging_dir/target-mipsel_24kc_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes -fexceptions -fno-strict-aliasing  -I/lede/staging_dir/host/include -I/lede/staging_dir/host/usr/include -I/lede/staging_dir/hostpkg/include -I/lede/staging_dir/target-mipsel_24kc_musl/host/include -DHAVE_EXPAT_CONFIG_H -o lib/xmlparse.lo -c lib/xmlparse.c
OpenWrt-libtool: compile:  gcc -I./lib -I. -O2 -I/lede/staging_dir/host/include -I/lede/staging_dir/host/usr/include -I/lede/staging_dir/hostpkg/include -I/lede/staging_dir/target-mipsel_24kc_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes -fexceptions -fno-strict-aliasing -I/lede/staging_dir/host/include -I/lede/staging_dir/host/usr/include -I/lede/staging_dir/hostpkg/include -I/lede/staging_dir/target-mipsel_24kc_musl/host/include -DHAVE_EXPAT_CONFIG_H -c lib/xmlparse.c  -fPIC -DPIC -o lib/.libs/xmlparse.o
lib/xmlparse.c:56:3: error: #error You do not have support for any sources of high quality entropy enabled. For end user security, that is probably not what you want. Your options include: * Linux + glibc >=2.25 (getrandom): HAVE_GETRANDOM, * Linux + glibc <2.25 (syscall SYS_getrandom): HAVE_SYSCALL_GETRANDOM, * BSD / macOS (arc4random_buf): HAVE_ARC4RANDOM_BUF, * libbsd (arc4random_buf): HAVE_ARC4RANDOM_BUF + HAVE_LIBBSD, * Windows (RtlGenRandom): _WIN32. If insist on not using any of these, bypass this error by defining XML_POOR_ENTROPY; you have been warned. For CMake, one way to pass the define is: cmake -DCMAKE_C_FLAGS="-pipe -O2 -DHAVE_SYSCALL_GETRANDOM" . If you have reasons to patch this detection code away or need changes to the build system, please open a bug. Thank you!
 # error  \
   ^~~~~
make[4]: *** [lib/xmlparse.lo] Error 1
```

thess, 2017-07-17:
I have no trouble building a ramips, MT7620 config with this, but from the looks of your output, this may be a **host** build problem. Make sure your build system has support for either: 
* getrandom (Linux 3.17+, glibc 2.25+) in `<sys/random.h>`
* SYS_getrandom (Linux 3.17+) - in `<sys/syscalls.h>`

diizzyy, 2017-07-18:
Debian 8 and buildbots falls under this too, please to a full import of that pull request.

diizzyy, 2017-07-18:
@jammmet 
Your issue should be fixed in https://github.com/lede-project/source/commit/7c727c6fa427dba9e43aaf3648d12af19a4858f1

jammmet, 2017-07-19:
Thank you. I'd updated to a later kernel to get around the issue.

diizzyy, 2017-07-19:
Please close this issue then :-)

ClassyGrrrl, 2017-07-19:
I was also having the same issue. Thanks for the update. Do you know if this patch will be included in the expat Makefile for OpenWrt?

diizzyy, 2017-07-19:
The OpenWrt repo wont receive any updates (in general) until/when the merge is done. I'd highly suggest that you move to LEDE as there are several packages/libraries/* with known vulns in the OpenWrt tree.

diizzyy, 2017-07-23:
https://github.com/openwrt/packages/pull/4619
This should fix the host build error

diizzyy, 2017-07-24:
Committed, please test and close issue if successful.

closed 2017-07-27 by jammmet