openwrt/packages #4597: wireguard: default route doesn't come back after ifdown

zorun, 2017-07-17:
When configuring Wireguard with a default IPv4 route through the tunnel (0/0 in AllowedIPs), it overrides the existing "regular" default route.  But when shutting down the wireguard interface, the "regular" default route is not reinstalled.  As as result, the system has no default route anymore.

@danrl I opened this report mostly for reference, I may have a look myself if I find the time.

kim0, 2018-12-11:
Hello @zorun .. I think I'm hitting this .. Any fixed or workarounds are welcome :)

kim0, 2018-12-11:
Actually I just figured out a workaround to this. Instead of routing the whole `0.0.0.0/0` through WG, what works is to route `0/1` and `128/1` .. So in code this is
```
config wireguard_wg
        ......
	list allowed_ips '0.0.0.0/1'
	list allowed_ips '128.0.0.0/1'
```

danrl, 2018-12-11:
I remember that I used the `/1` approach myself to work around the issue. I still wonder what the *right* way to re-install the route would be. Ask `odhcpd` for an update?

diizzyy, 2019-02-09:
@thess  @hnyman @zorun 
Please close, wrong repo

closed 2019-02-09 by hnyman