openwrt/packages #4606: haproxy: Compile fail: Unsupported jump between ISA modes

hewonen, 2017-07-20:
Maintainer: @heil
Environment: Compiling HEAD in Ubuntu x86_64 for TP-Link TL-WR1043ND

Description: Compile fails as described in https://downloads.lede-project.org/snapshots/faillogs/mipsel_24kc/packages/haproxy/ssl/compile.txt

```
make[4]: Entering directory '/home/builder/lede/source/build_dir/target-mips_24kc_musl/haproxy-ssl/haproxy-1.7.5'
mips-openwrt-linux-musl-gcc  -g -o haproxy src/haproxy.o src/base64.o src/protocol.o src/uri_auth.o src/standard.o src/buffer.o src/log.o src/task.o src/chunk.o src/channel.o src/listener.o src/lru.o src/xxhash.o src/time.o src/fd.o src/pipe.o src/regex.o src/cfgparse.o src/server.o src/checks.o src/queue.o src/frontend.o src/proxy.o src/peers.o src/arg.o src/stick_table.o src/proto_uxst.o src/connection.o src/proto_http.o src/raw_sock.o src/backend.o src/tcp_rules.o src/lb_chash.o src/lb_fwlc.o src/lb_fwrr.o src/lb_map.o src/lb_fas.o src/stream_interface.o src/stats.o src/proto_tcp.o src/applet.o src/session.o src/stream.o src/hdr_idx.o src/ev_select.o src/signal.o src/acl.o src/sample.o src/memory.o src/freq_ctr.o src/auth.o src/proto_udp.o src/compression.o src/payload.o src/hash.o src/pattern.o src/map.o src/namespace.o src/mailers.o src/dns.o src/vars.o src/filters.o src/flt_http_comp.o src/flt_trace.o src/flt_spoe.o src/cli.o src/ev_poll.o src/ev_epoll.o src/ssl_sock.o src/shctx.o src/hlua.o src/hlua_fcn.o ebtree/ebtree.o ebtree/eb32tree.o ebtree/eb64tree.o ebtree/ebmbtree.o ebtree/ebsttree.o ebtree/ebimtree.o ebtree/ebistree.o   -lcrypt  -lz -ldl  -lssl -lcrypto -ldl -Wl,--export-dynamic -L/home/builder/lede/source/staging_dir/target-mips_24kc_musl/lua-5.3.3/lib -llua533 -lm -ldl -L/home/builder/lede/tl1043/staging_dir/target-mips_24kc_musl/usr//lib -lpcreposix -lpcre -lcrypto -lm
/home/builder/lede/source/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/lib/gcc/mips-openwrt-linux-musl/5.4.0/../../../../mips-openwrt-linux-musl/bin/ld: copy reloc against protected `__stack_chk_guard' is dangerous
src/hlua.o: In function `hlua_hook':
/home/builder/lede/source/build_dir/target-mips_24kc_musl/haproxy-ssl/haproxy-1.7.5/src/hlua.c:982:(.text+0x29c): Unsupported jump between ISA modes; consider recompiling with interlinking enabled
/home/builder/lede/source/build_dir/target-mips_24kc_musl/haproxy-ssl/haproxy-1.7.5/src/hlua.c:982:(.text+0x2fc): Unsupported jump between ISA modes; consider recompiling with interlinking enabled
src/hlua.o: In function `hlua_checkfunction':
/home/builder/lede/source/build_dir/target-mips_24kc_musl/haproxy-ssl/haproxy-1.7.5/src/hlua.c:256:(.text+0xc648): Unsupported jump between ISA modes; consider recompiling with interlinking enabled
src/hlua.o: In function `hlua_register_fetches':
/home/builder/lede/source/build_dir/target-mips_24kc_musl/haproxy-ssl/haproxy-1.7.5/src/hlua.c:5568:(.text+0xd2bc): Unsupported jump between ISA modes; consider recompiling with interlinking enabled
src/hlua.o: In function `hlua_get_top_error_string':
/home/builder/lede/source/build_dir/target-mips_24kc_musl/haproxy-ssl/haproxy-1.7.5/src/hlua.c:266:(.text+0xd41c): Unsupported jump between ISA modes; consider recompiling with interlinking enabled
src/hlua.o: In function `hlua_yieldk':
/home/builder/lede/source/build_dir/target-mips_24kc_musl/haproxy-ssl/haproxy-1.7.5/src/hlua.c:829:(.text+0xd440): Unsupported jump between ISA modes; consider recompiling with interlinking enabled
src/hlua_fcn.o: In function `hlua_checkboolean':
/home/builder/lede/source/build_dir/target-mips_24kc_musl/haproxy-ssl/haproxy-1.7.5/src/hlua_fcn.c:48:(.text+0x4e4): Unsupported jump between ISA modes; consider recompiling with interlinking enabled
src/hlua_fcn.o: In function `hlua_pushstrippedstring':
/home/builder/lede/source/build_dir/target-mips_24kc_musl/haproxy-ssl/haproxy-1.7.5/src/hlua_fcn.c:142:(.text+0x974): Unsupported jump between ISA modes; consider recompiling with interlinking enabled
src/hlua_fcn.o: In function `hlua_class_const_int':
/home/builder/lede/source/build_dir/target-mips_24kc_musl/haproxy-ssl/haproxy-1.7.5/src/hlua_fcn.c:157:(.text+0x9bc): Unsupported jump between ISA modes; consider recompiling with interlinking enabled
src/hlua_fcn.o: In function `hlua_class_const_str':
/home/builder/lede/source/build_dir/target-mips_24kc_musl/haproxy-ssl/haproxy-1.7.5/src/hlua_fcn.c:163:(.text+0xa00): Unsupported jump between ISA modes; consider recompiling with interlinking enabled
src/hlua_fcn.o: In function `hlua_class_function':
/home/builder/lede/source/build_dir/target-mips_24kc_musl/haproxy-ssl/haproxy-1.7.5/src/hlua_fcn.c:169:(.text+0xa48): Unsupported jump between ISA modes; consider recompiling with interlinking enabled
src/hlua_fcn.o: In function `hlua_register_metatable':
/home/builder/lede/source/build_dir/target-mips_24kc_musl/haproxy-ssl/haproxy-1.7.5/src/hlua_fcn.c:212:(.text+0xb04): Unsupported jump between ISA modes; consider recompiling with interlinking enabled
collect2: error: ld returned 1 exit status
Makefile:805: recipe for target 'haproxy' failed
make[4]: *** [haproxy] Error 1
make[4]: Leaving directory '/home/builder/lede/source/build_dir/target-mips_24kc_musl/haproxy-ssl/haproxy-1.7.5'
Makefile:181: recipe for target '/home/builder/lede/source/build_dir/target-mips_24kc_musl/haproxy-ssl/haproxy-1.7.5/.built' failed
make[3]: *** [/home/builder/lede/source/build_dir/target-mips_24kc_musl/haproxy-ssl/haproxy-1.7.5/.built] Error 2
make[3]: Leaving directory '/home/builder/lede/source/feeds/packages/net/haproxy'
package/Makefile:109: recipe for target 'package/feeds/packages/haproxy/compile' failed
make[2]: *** [package/feeds/packages/haproxy/compile] Error 2
make[2]: Leaving directory '/home/builder/lede/source'
package/Makefile:105: recipe for target '/home/builder/lede/source/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/builder/lede/source/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede/source'
/home/builder/lede/source/include/toplevel.mk:207: recipe for target 'world' failed
make: *** [world] Error 2
```

heil, 2017-07-25:
i will try to disable the lua build for this target at least

hewonen, 2017-09-29:
Happens also with Archer C7 v2, HEAD pulled on Thu Sep 28th.

What should I do to be able to compile haproxy? Will there be loss of functionality?

heil, 2017-10-15:
lua is now disabled for haproxy on mips. please try again if it works for you.

closed 2017-10-15 by heil

hewonen, 2017-10-15:
Does compile. Will soon see if it functions.

Found out that Lua is for custom scripting of the proxy. I personally don't use it, but I wonder what the actual problem was and how easy it would be to fix.

commit 2ef52e04 (2017-10-15): haproxy: disable lua on mips
 - to fix  #4606 we disable to lua build for haproxy
   depending on CONFIG_mips = y