openwrt/packages #4608: Python urllib3 ImportError: No module named urllib3

kholisrag, 2017-07-21:
Maintainer: @commodo 
Environment: OpenWRT
Description:
When I update the git repo and package repo with

```
git pull
git checkout master
./scripts/feeds update -a
./scripts/feeds install  -a
```
And I recompile OpenWRT again, the stderr is like this
```
Traceback (most recent call last):
  File "/home/kholisrag/openwrt/build_dir/target-mips_34kc_musl-1.1.16/host/Python-2.7.13/Lib/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/home/kholisrag/openwrt/build_dir/target-mips_34kc_musl-1.1.16/host/Python-2.7.13/Lib/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha224
ERROR:root:code for hash sha256 was not found.
Traceback (most recent call last):
  File "/home/kholisrag/openwrt/build_dir/target-mips_34kc_musl-1.1.16/host/Python-2.7.13/Lib/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/home/kholisrag/openwrt/build_dir/target-mips_34kc_musl-1.1.16/host/Python-2.7.13/Lib/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha256
ERROR:root:code for hash sha384 was not found.
Traceback (most recent call last):
  File "/home/kholisrag/openwrt/build_dir/target-mips_34kc_musl-1.1.16/host/Python-2.7.13/Lib/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/home/kholisrag/openwrt/build_dir/target-mips_34kc_musl-1.1.16/host/Python-2.7.13/Lib/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha384
ERROR:root:code for hash sha512 was not found.
Traceback (most recent call last):
  File "/home/kholisrag/openwrt/build_dir/target-mips_34kc_musl-1.1.16/host/Python-2.7.13/Lib/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/home/kholisrag/openwrt/build_dir/target-mips_34kc_musl-1.1.16/host/Python-2.7.13/Lib/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha512
Traceback (most recent call last):
  File "/home/kholisrag/openwrt/build_dir/target-mips_34kc_musl-1.1.16/host/Python-2.7.13/Lib/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/home/kholisrag/openwrt/build_dir/target-mips_34kc_musl-1.1.16/host/Python-2.7.13/Lib/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/home/kholisrag/openwrt/build_dir/target-mips_34kc_musl-1.1.16/host/Python-2.7.13/Lib/ensurepip/__main__.py", line 4, in <module>
    ensurepip._main()
  File "/home/kholisrag/openwrt/build_dir/target-mips_34kc_musl-1.1.16/host/Python-2.7.13/Lib/ensurepip/__init__.py", line 205, in _main
    default_pip=args.default_pip,
  File "/home/kholisrag/openwrt/build_dir/target-mips_34kc_musl-1.1.16/host/Python-2.7.13/Lib/ensurepip/__init__.py", line 108, in bootstrap
    _run_pip(args + [p[0] for p in _PROJECTS], additional_paths)
  File "/home/kholisrag/openwrt/build_dir/target-mips_34kc_musl-1.1.16/host/Python-2.7.13/Lib/ensurepip/__init__.py", line 31, in _run_pip
    import pip
  File "/tmp/tmpRHoJos/pip-9.0.1-py2.py3-none-any.whl/pip/__init__.py", line 21, in <module>
  File "/tmp/tmpRHoJos/pip-9.0.1-py2.py3-none-any.whl/pip/_vendor/requests/__init__.py", line 62, in <module>
  File "/tmp/tmpRHoJos/pip-9.0.1-py2.py3-none-any.whl/pip/_vendor/requests/packages/__init__.py", line 29, in <module>
ImportError: No module named urllib3
Makefile:927: recipe for target 'install' failed
make[4]: *** [install] Error 1
make[4]: Leaving directory '/home/kholisrag/openwrt/build_dir/target-mips_34kc_musl-1.1.16/host/Python-2.7.13'
Makefile:304: recipe for target '/home/kholisrag/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/host/stamp/.python_installed' failed
make[3]: *** [/home/kholisrag/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/host/stamp/.python_installed] Error 2
make[3]: Leaving directory '/home/kholisrag/openwrt/feeds/packages/lang/python/python'
package/Makefile:196: recipe for target 'package/feeds/packages/python/host/compile' failed
make[2]: *** [package/feeds/packages/python/host/compile] Error 2
make[2]: Leaving directory '/home/kholisrag/openwrt'
package/Makefile:193: recipe for target '/home/kholisrag/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/stamp/.package_compile' failed
make[1]: *** [/home/kholisrag/openwrt/staging_dir/target-mips_34kc_musl-1.1.16/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kholisrag/openwrt'
/home/kholisrag/openwrt/include/toplevel.mk:192: recipe for target 'world' failed
make: *** [world] Error 
```
I tried with
`pip install urllib3 --upgrade`
it's still not working

commodo, 2017-07-21:
ugh...
when i changed for LEDE I forgot about OpenWrt

so, I would fix this in OpenWrt trunk, rather than Python ;
the issue seems to be that OpenWrt relies on `libopenssl-dev` being present on the host machine ;
whereas LEDE packages `libressl` in the `tools/` subfolder ; this was done to make the build on OS X happy, but was extend to Linux for consistency reasons I guess ;

i don't want to revert the change that I recently did to Python, because that causes other issues ;
`libopenssl-dev` [on the host machine] is a bit unreliable as well, when you factor in the number of Linux distros out there; and the ability to build on OS X

i'll try to propose `libressl` in OpenWrt `tools/`
if only the merge would have happened by now.... sigh

kholisrag, 2017-07-21:
hmhm... okay, I will try compile in Lede project... this new to me...

commodo, 2017-07-21:
it's the same :)

commodo, 2017-07-21:
let's see if that PR gets accepted
it fixed the issue for me

kholisrag, 2017-07-21:
@commodo when the second time I compile lede again, stderr :

```
configure: creating ./config.status
config.status: creating lib/stdio.h
config.status: creating lib/string.h
config.status: creating ovsdb/libovsdb.sym
config.status: creating ofproto/libofproto.sym
config.status: creating lib/libsflow.sym
config.status: creating lib/libopenvswitch.sym
config.status: creating ovn/lib/libovn.sym
config.status: creating vtep/libvtep.sym
config.status: creating Makefile
config.status: creating datapath/Makefile
config.status: creating datapath/linux/Kbuild
config.status: creating datapath/linux/Makefile
config.status: creating datapath/linux/Makefile.main
config.status: creating tests/atlocal
config.status: creating lib/libopenvswitch.pc
config.status: creating lib/libsflow.pc
config.status: creating ofproto/libofproto.pc
config.status: creating ovsdb/libovsdb.pc
config.status: creating include/openvswitch/version.h
config.status: creating config.h
config.status: executing tests/atconfig commands
config.status: executing depfiles commands
config.status: executing libtool commands
config.status: executing include/openflow/openflow.h.stamp commands
config.status: executing utilities/bugtool/dummy commands
config.status: executing ovn/dummy commands
config.status: executing ovn/utilities/dummy commands
configure: WARNING: unrecognized options: --disable-nls
rm -f /home/kholisrag/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/openvswitch-2.7.1/.configured_*
touch /home/kholisrag/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/openvswitch-2.7.1/.configured_85a6bb9fceab7fabb971b4292420ed2a
rm -f /home/kholisrag/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/openvswitch-2.7.1/.built
touch /home/kholisrag/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/openvswitch-2.7.1/.built_check
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/kholisrag/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/openvswitch-2.7.1:openvswitch-2.7.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -std=gnu99  -I/home/kholisrag/lede/staging_dir/target-mips_24kc_musl/usr/include -I/home/kholisrag/lede/staging_dir/target-mips_24kc_musl/include -I/home/kholisrag/lede/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/usr/include -I/home/kholisrag/lede/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/include/fortify -I/home/kholisrag/lede/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/kholisrag/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/openvswitch-2.7.1:openvswitch-2.7.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -std=gnu99  -I/home/kholisrag/lede/staging_dir/target-mips_24kc_musl/usr/include -I/home/kholisrag/lede/staging_dir/target-mips_24kc_musl/include -I/home/kholisrag/lede/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/usr/include -I/home/kholisrag/lede/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/include/fortify -I/home/kholisrag/lede/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/include " LDFLAGS="-L/home/kholisrag/lede/staging_dir/target-mips_24kc_musl/usr/lib -L/home/kholisrag/lede/staging_dir/target-mips_24kc_musl/lib -L/home/kholisrag/lede/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/usr/lib -L/home/kholisrag/lede/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl/lib -znow -zrelro " make  -C /home/kholisrag/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/openvswitch-2.7.1/. AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap/home/kholisrag/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/openvswitch-2.7.1:openvswitch-2.7.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto -std=gnu99" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CROSS="mips-openwrt-linux-musl-" ARCH="mips" ARCH="mips" PYTHONPATH="" PYTHON="" ;
make[4]: Entering directory '/home/kholisrag/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/openvswitch-2.7.1'
sed -f ./build-aux/extract-odp-netlink-h < datapath/linux/compat/include/linux/openvswitch.h > include/odp-netlink.h
PYTHONPATH=./python":" PYTHONDONTWRITEBYTECODE=yes  ./ovsdb/ovsdb-idlc.in annotate ./vswitchd/vswitch.ovsschema ./lib/vswitch-idl.ann > lib/vswitch-idl.ovsidl.tmp && mv lib/vswitch-idl.ovsidl.tmp lib/vswitch-idl.ovsidl
/bin/bash: ./ovsdb/ovsdb-idlc.in: @PYTHON@: bad interpreter: No such file or directory
Makefile:6182: recipe for target 'lib/vswitch-idl.ovsidl' failed
make[4]: *** [lib/vswitch-idl.ovsidl] Error 126
[... cut ...]
make[2]: *** [package/feeds/packages/openvswitch/compile] Error 2
make[2]: Leaving directory '/home/kholisrag/lede'
package/Makefile:105: recipe for target '/home/kholisrag/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/kholisrag/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kholisrag/lede'
/home/kholisrag/lede/include/toplevel.mk:207: recipe for target 'world' failed
make: *** [world] Error 2
```

commodo, 2017-07-25:
that's weird ; i don't get that with lede
will check

nmaas87, 2017-07-26:
I'm experiencing the same error, I think this is due to trying to compile python-openssl(?)
OpenWRT current trunk version :)

````
Writing /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/lib/python2.7/lib-dynload/Python-2.7.13-py2.7.egg-info
rm /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/lib/python2.7/lib-dynload/_sysconfigdata.py*
/usr/bin/install -c -m 644 ./Misc/python.man \
        /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/share/man/man1/python2.7.1
if test ! -d /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/lib/pkgconfig; then \
        echo "Creating directory /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/lib/pkgconfig"; \
        /usr/bin/install -c -d -m 755 /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/lib/pkgconfig; \
fi
if test -f /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/bin/python -o -h /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/bin/python; \
then rm -f /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/bin/python; \
else true; \
fi
(cd /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/bin; ln -s python2 python)
rm -f /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/bin/python2
(cd /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/bin; ln -s python2.7 python2)
rm -f /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/bin/python2-config
(cd /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/bin; ln -s python2.7-config python2-config)
rm -f /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/bin/python-config
(cd /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/bin; ln -s python2-config python-config)
test -d /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/lib/pkgconfig || /usr/bin/install -c -d -m 755 /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/lib/pkgconfig
rm -f /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/lib/pkgconfig/python2.pc
(cd /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/lib/pkgconfig; ln -s python-2.7.pc python2.pc)
rm -f /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/lib/pkgconfig/python.pc
(cd /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/lib/pkgconfig; ln -s python2.pc python.pc)
rm -f /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/share/man/man1/python2.1
(cd /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/share/man/man1; ln -s python2.7.1 python2.1)
rm -f /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/share/man/man1/python.1
(cd /home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/share/man/man1; ln -s python2.1 python.1)
if test "xinstall" != "xno"  ; then \
        case install in \
                upgrade) ensurepip="--upgrade" ;; \
                install|*) ensurepip="" ;; \
        esac; \
         ./python -E -m ensurepip \
                $ensurepip --root=/ ; \
fi
ERROR:root:code for hash sha224 was not found.
Traceback (most recent call last):
  File "/home/user/OpenWRT/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/Python-2.7.13/Lib/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/home/user/OpenWRT/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/Python-2.7.13/Lib/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha224
ERROR:root:code for hash sha256 was not found.
Traceback (most recent call last):
  File "/home/user/OpenWRT/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/Python-2.7.13/Lib/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/home/user/OpenWRT/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/Python-2.7.13/Lib/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha256
ERROR:root:code for hash sha384 was not found.
Traceback (most recent call last):
  File "/home/user/OpenWRT/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/Python-2.7.13/Lib/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/home/user/OpenWRT/openwrt/build_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/host/Python-2.7.13/Lib/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha384
ERROR:root:code for hash sha512 was not found.
Traceback (most recent call last):
[... cut ...]
make[2]: Leaving directory '/home/user/OpenWRT/openwrt'
package/Makefile:193: recipe for target '/home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/stamp/.package_compile' failed
make[1]: *** [/home/user/OpenWRT/openwrt/staging_dir/target-arm_arm1176jzf-s+vfp_musl-1.1.16_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/OpenWRT/openwrt'
/home/user/OpenWRT/openwrt/include/toplevel.mk:192: die Regel für Ziel „world“ scheiterte
make: *** [world] Fehler 2
````

commodo, 2017-07-26:
@nmaas87 
the fix for OpenWrt should be in this PR https://github.com/openwrt/openwrt/pull/508
hopefully that will get merged soon ; or get some attention

or you could backport it locally until then

nmaas87, 2017-07-26:
@commodo Thanks for the heads-up 👍 
I think I will edit these files locally for the time being and try your PR :)

kholisrag, 2017-07-26:
@commodo sorry, just checked github,
Solved with cleaan install lede from the start... It's becaus the sudo/root issue...
I run all syntax in root mode

closed 2017-07-26 by kholisrag

commodo, 2017-07-26:
ah :)
thanks 

i did not get a chance to look into it;
i was letting an email unread [from this github issue] to remember about ;
otherwise i forget

gzivdo, 2017-10-24:
Still is issue with current openwrt.git master branch. Cannot compile python 2.7. Host system is debian.8, also tried ubuntu 16.04
`ERROR:root:code for hash sha224 was not found.
Traceback (most recent call last):
  File "/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/Python-2.7.14/Lib/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/Python-2.7.14/Lib/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha224
ERROR:root:code for hash sha256 was not found.
Traceback (most recent call last):
  File "/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/Python-2.7.14/Lib/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/Python-2.7.14/Lib/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha256
ERROR:root:code for hash sha384 was not found.
Traceback (most recent call last):
  File "/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/Python-2.7.14/Lib/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/Python-2.7.14/Lib/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha384
ERROR:root:code for hash sha512 was not found.
Traceback (most recent call last):
  File "/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/Python-2.7.14/Lib/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/Python-2.7.14/Lib/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha512
Traceback (most recent call last):
  File "/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/Python-2.7.14/Lib/runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/Python-2.7.14/Lib/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/Python-2.7.14/Lib/ensurepip/__main__.py", line 4, in <module>
    ensurepip._main()
  File "/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/Python-2.7.14/Lib/ensurepip/__init__.py", line 205, in _main
    default_pip=args.default_pip,
  File "/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/Python-2.7.14/Lib/ensurepip/__init__.py", line 108, in bootstrap
    _run_pip(args + [p[0] for p in _PROJECTS], additional_paths)
  File "/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/Python-2.7.14/Lib/ensurepip/__init__.py", line 31, in _run_pip
    import pip
  File "/tmp/tmp901Raa/pip-9.0.1-py2.py3-none-any.whl/pip/__init__.py", line 21, in <module>
  File "/tmp/tmp901Raa/pip-9.0.1-py2.py3-none-any.whl/pip/_vendor/requests/__init__.py", line 62, in <module>
  File "/tmp/tmp901Raa/pip-9.0.1-py2.py3-none-any.whl/pip/_vendor/requests/packages/__init__.py", line 29, in <module>
ImportError: No module named urllib3
Makefile:920: recipe for target 'install' failed
make[4]: *** [install] Error 1
make[4]: Leaving directory '/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/Python-2.7.14'
Makefile:312: recipe for target '/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/stamp/.python_installed' failed
make[3]: *** [/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/stamp/.python_installed] Error 2
make[3]: Leaving directory '/openwrt/feeds/packages/lang/python/python'
package/Makefile:196: recipe for target 'package/feeds/packages/python/host/compile' failed
make[2]: *** [package/feeds/packages/python/host/compile] Error 2
make[2]: Leaving directory '/openwrt'
package/Makefile:193: recipe for target '/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/stamp/.package_compile' failed
make[1]: *** [/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/openwrt'
/openwrt/include/toplevel.mk:192: recipe for target 'world' failed
make: *** [world] Error 2
`

commodo, 2017-10-24:
do you have `libopenssl-dev` installed on your system ?

commodo, 2017-10-24:
oh right ;
this is OpenWrt 

PR https://github.com/openwrt/openwrt/pull  is still open
nothing I can do about that PR [ since I don't have access ]

should be fixed when that PR is fixed, or the re-merge of LEDE & OpenWrt finishes

gzivdo, 2017-10-24:
Yes, I have libssl-dev, because without it build not starting, and there is no libressl in debian/ubuntu.

commodo, 2017-10-24:
so, the idea is that you need to add `libressl` in the OpenWrt /tools folder
like that PR [ or this patch does ]: https://github.com/openwrt/openwrt/pull/508/commits/4adba3cea3fd258f8586eea769d99a2a87956ee0

then Python will build and so will OVS

gzivdo, 2017-10-24:
Thanks, it helped

commodo, 2017-10-24:
Thanks :)
And sorry for the trouble.
I admit, I am a bit stubborn about adapting the Python build to work on OpenWrt atm.

It's hard enough [as it is] to support both LEDE & OpenWrt [in a clean manner] so I chose LEDE, since that will be the next OpenWRT :)