openwrt/packages #4611: xl2tpd - l2tp interface down on boot

kdulep, 2017-07-22:
Description:
l2tp protocol code doesn't respect luci flag 'Bring up on boot'

yousong, 2017-07-22:
Hi @kdulep 

I do not now have the environment to reproduce the issue, but I am wondering if commenting out the 'proto_add_host_dependency' line in the `/lib/netifd/proto/l2tp.sh` can workaround the issue for you.

yousong, 2017-07-22:
Can you please also share the release version of the system you are on?  Like the content of `/etc/openwrt_release`

kdulep, 2017-07-22:
i think problem go away just need to wait a bit more on startup 
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='15.05.1'
DISTRIB_REVISION='r48532'
DISTRIB_CODENAME='chaos_calmer'
DISTRIB_TARGET='ar71xx/generic'
DISTRIB_DESCRIPTION='OpenWrt Chaos Calmer 15.05.1'
DISTRIB_TAINTS=''
```

closed 2017-07-22 by kdulep