openwrt/packages #4613: sqm: Add Google's BBR for TCP congestion control

Strykar, 2017-07-22:
**Maintainer**: @tohojo 
**Environment**: Any / Linux apu 4.4.71 #0 SMP x86_64 GNU/Linux
**Firmware Version**: LEDE Reboot 17.01.2 / LuCI lede-17.01

**Description**: Feature request.
Add/offer Google's _Bottleneck Bandwidth and Round-trip propagation time_ [(BBR)](https://github.com/google/bbr) as an additional TCP congestion control algorithm in SQM.
Patching may require Linux kernel version >= 4.9
```
root@apu:~# sysctl net.ipv4.tcp_congestion_control
net.ipv4.tcp_congestion_control = cubic
```
**References**:
1. https://github.com/google/bbr
2. https://cloudplatform.googleblog.com/2017/07/TCP-BBR-congestion-control-comes-to-GCP-your-Internet-just-got-faster.html
3. https://www.cyberciti.biz/cloud-computing/increase-your-linux-server-internet-speed-with-tcp-bbr-congestion-control

tohojo, 2017-07-22:
What good would running bbr on the router do? It's a congestion control, not a queue management algorithm...

Strykar, 2017-07-22:
@tohojo If not devices behind it, wouldn't a router itself running services benefit?

tohojo, 2017-07-22:
Possibly, depending on what services the router is running. In most cases it would probably be fairly marginal. But in any case that has nothing to do with sqm-scripts; you'll have to open a bug against LEDE itself to get it included in the kernel...

anomeome, 2017-07-22:
And it's been going [on](https://github.com/lede-project/source/pull/1227)

tohojo, 2017-07-22:
Right. Closing this, then

closed 2017-07-22 by tohojo

ohsorry, 2017-12-19:
bbr is really useful, i use LEDE as proxy router to bypass GFW.
[https://forum.lede-project.org/t/tcp-bbr-support/1168/13](https://forum.lede-project.org/t/tcp-bbr-support/1168/13)

A 'proxy router' , i mean an  [openwrt-shadowsocks proxy client](https://github.com/shadowsocks/openwrt-shadowsocks) running on LEDE router.
without tcp_bbr the tcp speed is only below 20KB/s.

i have tried latest x86 build [openwrt-x86-generic-combined-ext4.img.gz](http://downloads.lede-project.org/snapshots/targets/x86/generic/openwrt-x86-generic-combined-ext4.img.gz) in vmware.  [configure bbr failed.](https://forum.lede-project.org/t/tcp-bbr-support/1168/15)