openwrt/packages #4614: ShadowSocks-libev is not running

WeiqiChenCN, 2017-07-22:
Maintainer: @yousong 
Environment: Raspberry pi 2, model B, Openwrt Master branch.

Description:

Software package
```bash
root@Raspberrypi-OpenWRT:~# opkg list | grep sha
luci-app-shadowsocks-libev - git-17.194.28316-2224714-1
shadowsocks-libev-config - 3.0.6-2
shadowsocks-libev-ss-local - 3.0.6-2
shadowsocks-libev-ss-redir - 3.0.6-2
shadowsocks-libev-ss-rules - 3.0.6-2
shadowsocks-libev-ss-tunnel - 3.0.6-2
```

And The luci `http://raspberrypi-openwrt.lan/cgi-bin/luci/admin/services/shadowsocks-libev` page show

```
ShadowSocks-libev
ShadowSocks-libev is not running
Global Setting
This section contains no values yet

 UDP Relay
This section contains no values yet

 UDP Forward
This section contains no values yet

 Access Control
This section contains no values yet
```

My configure file:
`/etc/config/shadowsocks-libev`
```
config ss_local
        option disabled 1
        option server 'sss0'
        option local_address '0.0.0.0'
        option local_port '1080'
        option timeout '30'

config ss_tunnel
        option disabled 1
        option server 'sss0'
        option local_address '0.0.0.0'
        option local_port '1090'
        option tunnel_address 'example.com:80'
        option mode 'tcp_and_udp'
        option timeout '60'

config ss_redir hi
        option disabled 1
        option server 'sss0'
        option local_address '0.0.0.0'
        option local_port '1100'
        option mode 'tcp_and_udp'
        option timeout '60'
        option fast_open 1
        option verbose 1
        option reuse_port 1

config ss_redir hj
        option disabled 1
        option server 'sss0'
        option local_address '0.0.0.0'
        option local_port '1100'
        option mode 'tcp_and_udp'
        option timeout '60'
        option fast_open 1
        option verbose 1
        option reuse_port 1

config ss_rules 'ss_rules'
        option disabled 1
        option redir_tcp 'hi'
        option redir_udp 'hi'
        option src_default 'bypass'
        option local_default 'checkdst'
        list src_ips_forward '192.168.1.4'
        list dst_ips_forward '8.8.8.8'

config server 'sss0'
        option disabled 1
        option server '192.168.1.3'
        option server_port '9001'
        option password '********'
        option method 'aes-256-cfb'

config ss_server
        option disabled 1
        option server_port '9001'
        option password '********'
        option method 'aes-256-cfb'
        option bind_address '192.168.7.72'

```

`/etc/config/shadowsocks-opkg`
```
config general
        option startup_delay '0'

config transparent_proxy
        option main_server 'nil'
        option udp_relay_server 'nil'
        option local_port '1234'

config socks5_proxy
        option server 'nil'
        option local_port '1080'

config port_forward
        option server 'nil'
        option local_port '5300'
        option destination '8.8.4.4:53'

config servers
        option auth '0'
        option fast_open '0'
        option server '127.0.0.1'
        option server_port '8388'
        option timeout '60'
        option password 'barfoo!'
        option encrypt_method 'rc4-md5'

config access_control
        option self_proxy '1'
```

I tried `/etc/init.d/shadowsocks-libev start` but nothing changes.

yousong, 2017-07-22:
The `shadowsocks-libev` package in this repo has been rewritten only about 20 days ago and the corresponding rewritten version of `luci-app-shadowsocks-libev` package has only been ready only the day before yesterday and it's still in the luci repo for review and feedback (https://github.com/openwrt/luci/pull/1242).

I would suggest giving the openwrt/luci#1242 a try.  It should work.  Otherwise, you need to manually enabling/configuring the relevant config sections in the `/etc/config/shadowsocks-libev`.  There is a [README.md](https://github.com/openwrt/packages/blob/master/net/shadowsocks-libev/README.md) file to guide you through this

The `/etc/config/shadowsocks-opkg` file is probably from shadowsocks/luci-app-shadowsocks which is not relevant here.

WeiqiChenCN, 2017-07-23:
I `rm feeds/luci/applications/luci-app-shadowsocks-libev/ -r` and copy another one from `https://github.com/yousong/luci/commit/4e6b8ae9dfa862abc85a2194fc788390114ed597`.

I recompiled, installed it on my raspberry pi , and reboot my pi.

Now I seem that `luci-app-shadowsocks-libev` can add _Local Instances_ and configure _Remote Servers_.

But the _Running_ status is _no_ , I can't get it _yes_.

[image]

[image]

> The _Server_'s content is ip address of my $$ server. (Can I write it's hostname in that Text box??)

WeiqiChenCN, 2017-07-23:
Before this time, I used 
```
git clone --depth=1 https://github.com/shadowsocks/openwrt-feeds.git package/feeds
git clone --depth=1 https://github.com/shadowsocks/openwrt-shadowsocks.git package/shadowsocks-libev
git clone --depth=1 https://github.com/shadowsocks/luci-app-shadowsocks.git package/luci-app-shadowsocks
git clone --depth=1 https://github.com/clowwindy/ChinaDNS.git package/ChinaDNS

./scripts/feeds update -a
./scripts/feeds install -a
make menuconfig
make  package/luci-app-shadowsocks/compile package/shadowsocks-libev/compile V=99
```

to compile ss for openwrt. 
Now I found ss in openwrt's trunk  so I want to try that out.

> Anyway, I can't compile the code clone below in the sdk that is compiled by me. It seem that something wrong in the package `mbedtls-2.5.1` when it compiled in a sdk. (https://github.com/shadowsocks/luci-app-shadowsocks/issues/131)

yousong, 2017-07-23:
> But the Running status is no , I can't get it yes.

That's because the instances are in "Disabled" state in the screenshot.  Click the button to toggle this state, then "Save & apply".  I will add a line of text for this in the description text.  Thanks for the feedback.

> The Server's content is ip address of my $$ server. (Can I write it's hostname in that Text box??)

Yes, the datatype of that field is `host` and it will accept ipv4, ipv6, hostnames

WeiqiChenCN, 2017-07-23:
I tried that before, And it's the same.

[image]

```bash
root@Raspberrypi-OpenWRT:~# ss-local --help

shadowsocks-libev 3.0.6

  maintained by Max Lv <max.c.lv@gmail.com> and Linus Yang <laokongzi@gmail.com>

  usage:

    ss-local

       -s <server_host>           Host name or IP address of your remote server.
       -p <server_port>           Port number of your remote server.
       -l <local_port>            Port number of your local server.
       -k <password>              Password of your remote server.
       -m <encrypt_method>        Encrypt method: rc4-md5,
                                  aes-128-gcm, aes-192-gcm, aes-256-gcm,
                                  aes-128-cfb, aes-192-cfb, aes-256-cfb,
                                  aes-128-ctr, aes-192-ctr, aes-256-ctr,
                                  camellia-128-cfb, camellia-192-cfb,
                                  camellia-256-cfb, bf-cfb,
                                  chacha20-ietf-poly1305,
                                  xchacha20-ietf-poly1305,
                                  salsa20, chacha20 and chacha20-ietf.
                                  The default cipher is rc4-md5.

       [-a <user>]                Run as another user.
       [-f <pid_file>]            The file path to store pid.
       [-t <timeout>]             Socket timeout in seconds.
       [-c <config_file>]         The path to config file.
       [-n <number>]              Max number of open files.
       [-i <interface>]           Network interface to bind.
       [-b <local_address>]       Local address to bind.

       [-u]                       Enable UDP relay.
       [-U]                       Enable UDP relay and disable TCP relay.

       [--reuse-port]             Enable port reuse.
       [--fast-open]              Enable TCP fast open.
                                  with Linux kernel > 3.7.0.
       [--acl <acl_file>]         Path to ACL (Access Control List).
       [--mtu <MTU>]              MTU of your network interface.
       [--mptcp]                  Enable Multipath TCP on MPTCP Kernel.
       [--key <key_in_base64>]    Key of your remote server.
       [--plugin <name>]          Enable SIP003 plugin. (Experimental)
       [--plugin-opts <options>]  Set SIP003 plugin options. (Experimental)

       [-v]                       Verbose mode.
       [-h, --help]               Print this message.

```

yousong, 2017-07-23:
It will take a few seconds before procd starts up the instances and before that the page wil be rendered with running being off.  Have you tried refreshing the Local Instances page?

Is there any output if you manually start the service with /etc/init.d/shadowsocks-libev.  If all goes well, the generated json config file will bein directory /var/etc/shadowsocks-libev

WeiqiChenCN, 2017-07-23:
```bash
root@Raspberrypi-OpenWRT:~# /etc/init.d/shadowsocks-libev restart
root@Raspberrypi-OpenWRT:~# ls /var/etc/shadowsocks-libev/
root@Raspberrypi-OpenWRT:~# ps | grep ss
 2706 root      1012 S    grep ss
root@Raspberrypi-OpenWRT:~#
```

Is there a log or other thing that I can use it do debug what is happening?

yousong, 2017-07-23:
Weiqi and I have just finished a Teamviewer session and the issue turns out to be caused by `validate_data` being too old in OpenWrt 15.01 branch causing compatibility issues with shadowsocks-libev in `master` branch here.  I have just pushed the fix.

Thanks for the feedback and patience.

closed 2017-07-23 by yousong

no1be4me, 2017-08-18:
yousong,

Can you help, I have shadowsocks-libev_3.0.8-1 and shadowsocks-libev-3.0.8-1 server also installed on Lede 17.0.1. The Kernel is 4.4.79, when I tried loading luci-app-shadowsocks_1.8.1-1 it gives the following error:
 
Installing luci-app-shadowsocks (1.8.1-1) to root...
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-shadowsocks:
 *      kernel (= 4.4.71-1-840c23c3f9dc6160d8c7bd302c6f494d) *  kernel (= 4.4.71-1-840c23c3f9dc6160d8c7bd302c6f494d) *
 * opkg_install_cmd: Cannot install package luci-app-shadowsocks.
Can to say how to get this install on this Kernel?

yousong, 2017-08-19:
The error indicates that the package was built against a kernel of different version from the currently running one.  Please consider giving shadowsocks-libev related packages in LEDE master branch a try.

For further info, please open a new issue and report there.  Keep the old, closed discussion clean and intact.  Thanks