openwrt/packages #4620: postfix: post install script

flipreverse, 2017-07-24:
Hi folks,

Since issue #582 is fixed, the postfix post install script sets the right permissions for every file needed by postfix. Even though the permissions are already correct, it sets them. The kernel therefore copies the file, which gets chmodded, to the overlayfs. Since the config partition is pretty small, even on an ARCHER C2600, the postfix files consume the whole space leaving no space for any config files.

Might it be possible to patch the post-install script to check whether a chmod is necessary?
If not, it steps over leaving the file as it is, and don't waste any space.

Cheers,
Alex

hnyman, 2017-07-24:
cc @lucize @Shulyaka

Shulyaka, 2017-07-24:
Hi Alex

Is it the 'postfix set-permissions' command that makes the trouble?

flipreverse, 2017-07-24:
@Shulyaka Yes.

Shulyaka, 2017-07-24:
Ok, will try to fix it.

Shulyaka, 2017-08-02:
@flipreverse Are you able to test the fix #4661, whether it solves the problem for you?

flipreverse, 2017-08-05:
@Shulyaka The issue seems to be fixed. :) Thank you!

closed 2017-08-05 by hnyman

Shulyaka, 2017-08-05:
Thank you for testing!