openwrt/packages #4633: postfix: symlink to sendmail is missing

flipreverse, 2017-07-26:
Hi folks,

I need sendmail to make crond send mails in case of an error. I already have postfix up and running. I've selected sendmail (from busybox), but the binary /usr/sbin/sendmail does not exist.
Only sendmail.{old,postfix} exist. After I grepped through the lede tree, I found the postfix post install script (build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root.orig-ipq806x/usr/lib/opkg/info/postfix.postinst-pkg) which seems to be responsible for providing /usr/sbin/sendmail. Somehow it doesn't do it's job.
Line 8 of the aforementioned script contains the necessary line which creates a symlink to sendmail.postfix. However, it is not executed. :(
Can you confirm this behavior?

Cheers,
Alex

Shulyaka, 2017-08-02:
You are right, there is an error on that line, it should be
```
ln -s "$(sendmail_path)$(ln_suffix)" "$${IPKG_INSTROOT}$(sendmail_path)"
```
instead of 
```
ln -s "$${IPKG_INSTROOT}$(sendmail_path)$(ln_suffix)" "$(sendmail_path)"
```
I will make a fix later today. In the meanwhile, please create the symlink for sendmail, newaliases and mailq manually on your device.

Thank you for the issue!

Best regards,
Denis Shulyaka

flipreverse, 2017-08-05:
@Shulyaka The issue seems to be fixed. :) Thank you!

closed 2017-08-05 by hnyman