openwrt/packages #4635: fwknopd: /lib/functions.sh not found

aasgit, 2017-07-26:
Maintainer: @jp-bennett 
Environment: 

```
NAME="LEDE"
VERSION="SNAPSHOT, Reboot"
ID="lede"
ID_LIKE="lede openwrt"
PRETTY_NAME="LEDE Reboot SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="http://lede-project.org/"
BUG_URL="http://bugs.lede-project.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r4619-d72371e"
LEDE_BOARD="mvebu/generic"
LEDE_ARCH="arm_cortex-a9_vfpv3"
LEDE_TAINTS="no-all"
LEDE_DEVICE_MANUFACTURER="LEDE"
LEDE_DEVICE_MANUFACTURER_URL="http://lede-project.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="LEDE Reboot SNAPSHOT r4619-d72371e"
```

Description:

I'm trying to build an image with LEDE Image Builder and I get the following error:

```
lede-imagebuilder-mvebu.Linux-x86_64/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/root-mvebu/etc/init.d/fwknopd: line 7: /lib/functions.sh: No such file or directory  
```

Latest image tried is the current one:

```
lede-imagebuilder-mvebu.Linux-x86_64.tar.xz| 4c89b71d8f4a4133140c83d20fd00b692fed48a81f9e759660c61a4017ef2522 | 33790.8 KB| Wed Jul 26 09:53:33 2017  
```
@jow-  suggested this fix:

```
diff --git a/net/fwknop/files/fwknopd.init b/net/fwknop/files/fwknopd.init
index baf81b8d..5fd2656e 100644
--- a/net/fwknop/files/fwknopd.init
+++ b/net/fwknop/files/fwknopd.init
@@ -4,13 +4,13 @@
 # Copyright (C) 2009-2014 fwknop developers and contributors. For a full
 # list of contributors, see the file 'CREDITS'.
 #
-. /lib/functions.sh
 START=95
 
 FWKNOPD_BIN=/usr/sbin/fwknopd
 
 start()
 {
+	. /lib/functions.sh
 	gen_confs
 	if [ $UCI_ENABLED ]; then
         	$FWKNOPD_BIN -c /var/etc/fwknopd.conf -a /var/etc/access.conf 
@@ -34,6 +34,7 @@ restart()
 
 reload()
 {
+	. /lib/functions.sh
 	gen_confs
         $FWKNOPD_BIN -R
 }

```

but that would be valid only if I wasbuilding from source.

I'm available for further info or tests.

pprindeville, 2017-07-26:
You shouldn't need to invoke it explicitly at all.

Your wrapper `/etc/rc.common` is already including it.

jp-bennett, 2017-07-26:
I'll take a look at this.  Thanks for the report!

jp-bennett, 2017-07-29:
@aasgit Hey, I'm having trouble replicating this.  Can you re-test with the Jul 28th or later snapshot.  Alternatively, what is the exact command you're running that triggers this error?

aasgit, 2017-07-31:
@jp-bennett I'm ging to test it now and I'll let you know.

aasgit, 2017-07-31:
@jp-bennett 

tested just now with

```
lede-imagebuilder-mvebu.Linux-x86_64.tar.xz| 49bafa7ae2399fe85d401a442b38711bb0f9040ff3117885ab47a7a03a7de60a | 31512.4 KB| Mon Jul 31 12:26:51 2017
```

```
make image PROFILE="linksys-wrt3200acm" PACKAGES="adblock bash bind-dig block-mount ca-bundle ca-certificates coreutils-base64 coreutils-basename coreutils-cat coreutils-chcon coreutils
-chgrp coreutils-chmod coreutils-chown coreutils-chroot coreutils-cksum coreutils-comm coreutils-cp coreutils-csplit coreutils-cut coreutils-date coreutils-dd coreutils-dir coreutils-di
rcolors coreutils-dirname coreutils-du coreutils-echo coreutils-env coreutils-expand coreutils-expr coreutils-factor coreutils-false coreutils-fmt coreutils-fold coreutils-groups coreut
ils-head coreutils-hostid coreutils-id coreutils-install coreutils-join coreutils-kill coreutils-link coreutils-ln coreutils-logname coreutils-ls coreutils-md5sum coreutils-mkdir coreut
ils-mkfifo coreutils-mknod coreutils-mktemp coreutils-mv coreutils-nice coreutils-nl coreutils-nohup coreutils-nproc coreutils-od coreutils-paste coreutils-pathchk coreutils-pinky coreu
tils-pr coreutils-printenv coreutils-printf coreutils-ptx coreutils-pwd coreutils-readlink coreutils-realpath coreutils-rm coreutils-rmdir coreutils-runcon coreutils-seq coreutils-sha1s
um coreutils-sha224sum coreutils-sha256sum coreutils-sha384sum coreutils-sha512sum coreutils-shred coreutils-shuf coreutils-sleep coreutils-sort coreutils-split coreutils-stat coreutils
-stdbuf coreutils-stty coreutils-sum coreutils-sync coreutils-tac coreutils-tail coreutils-tee coreutils-test coreutils-timeout coreutils-touch coreutils-tr coreutils-true coreutils-tru
ncate coreutils-tsort coreutils-tty coreutils-uname coreutils-unexpand coreutils-uniq coreutils-unlink coreutils-uptime coreutils-users coreutils-vdir coreutils-wc coreutils-who coreuti
ls-whoami coreutils-yes curl diffutils dnscrypt-proxy -dnsmasq dnsmasq-full dropbear e2fsprogs fdisk file git git-http gnupg gnupg-utils haveged htop iftop ip-full lftp luci luci-app-ad
block luci-app-clamav luci-app-ddns luci-app-fwknopd luci-app-openvpn luci-app-sqm luci-app-uhttpd luci-app-watchcat luci-base luci-ssl-openssl msmtp msmtp-mta mutt netdata nginx odhcp6
c openvpn-openssl ppp-mod-pppoa procps-ng-pkill procps-ng-pmap procps-ng-ps procps-ng-pwdx procps-ng-skill procps-ng-slabtop procps-ng-snice procps-ng-tload procps-ng-vmstat procps-ng-w
 procps-ng-watch rsync shadow-chage shadow-chfn shadow-chpasswd shadow-chsh shadow-common shadow-expiry shadow-faillog shadow-gpasswd shadow-groupadd shadow-groupdel shadow-groupmems sh
adow-groupmod shadow-groups shadow-lastlog shadow-login shadow-newgrp shadow-nologin shadow-passwd shadow-su shadow-useradd shadow-userdel shadow-usermod shadow-utils shadow-vipw sudo s
wap-utils sysstat tmux unzip usbutils vim-full vim-runtime wget xzgrep" FILES="files/" EXTRA_IMAGE_NAME="cga-2017-07-31-1525"
```

I still get the error:

```
[...]
Copying extra files                                                                                                                           
                                                                                                                                                
Cleaning up                                                                                                                                                                              
                                                                                                                                               
Activating init scripts                                                                                                                      
/bin/bash                                                                                                                                                                                
/bin/rbash                                                                                                                                     
/bin/rbash                                                                                                                                             
/bin/rbash                                                                                                                                                                               
/bin/rbash                                                                                                                                       
/bin/rbash                                                                                                                                                 
/home/andrea/LEDE/ImageBuilding/lede-buildwrapper/lede-imagebuilder-mvebu.Linux-x86_64/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/root-mvebu/etc/init.d/fwknopd: line 7: /lib/functio
ns.sh: No such file or directory                                                                                                                        
which: no uci in (/home/andrea/LEDE/ImageBuilding/lede-buildwrapper/lede-imagebuilder-mvebu.Linux-x86_64/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.4.0_musl_eabi/bin:/home/andrea/L
EDE/ImageBuilding/lede-buildwrapper/lede-imagebuilder-mvebu.Linux-x86_64/staging_dir/host/bin:/home/andrea/LEDE/ImageBuilding/lede-buildwrapper/lede-imagebuilder-mvebu.Linux-x86_64/stag
[... cut ...]
DE/ImageBuilding/lede-buildwrapper/lede-imagebuilder-mvebu.Linux-x86_64/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-5.4.0_musl_eabi/bin:/home/andrea/LEDE/ImageBuilding/lede-buildwrapp
er/lede-imagebuilder-mvebu.Linux-x86_64/staging_dir/host/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl)
                                                                                                                                                                     
Building images...                                              
[...]
```

aasgit, 2017-08-31:
@jp-bennett  a little note to say that this still happens on latest builds with the same commands. the last build built successfully (and running on my router right now) is from two days ago.

jp-bennett, 2017-10-14:
Was finally able to reproduce this issue.  It looks like simply removing the functions.sh line will get it working.  I'll test a bit more and send in the pull request to get it fixed.

aasgit, 2017-10-14:
@jp-bennett good stuff 👍 

only one question: is @jow- patch good to fix this?

jp-bennett, 2017-10-14:
@aasgit The patch by @jow- does fix it.  Removing the line altogether seems to also work.  I prefer the solution that reduces LOC rather than adds to it, if they're otherwise identical. :)

There is an existing pull request that touches this file.  Once it lands I'll make this change as well.

aasgit, 2017-10-14:
@jp-bennett I agree that less LOC and DRY are solid and fair principles. However, although I'm not the dev/maintainer here, wouldn't those lib functions be needed (hence the @jow- patch suggestion)? my2c.

jp-bennett, 2017-10-16:
This should be fixed with #4951.  I believe the needed include is always present due to the rc.common wrapper.  Let me know if you run into problems.

aasgit, 2017-10-16:
I'll build again once this is fixed/closed and provide feedback. thank you :)

dibdot, 2018-08-17:
should be fixed.

closed 2018-08-17 by dibdot