openwrt/packages #4638: package/feeds/packages/python/python-cffi/Makefile' does not implement a 'host' build type

build000, 2017-07-27:
```
dom@machina:~/Pulpit/Projekty/WITI-LEDE-REBOOT-small$ git pull
remote: Counting objects: 4, done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 4 (delta 3), reused 0 (delta 0)
Unpacking objects: 100% (4/4), done.
From https://git.lede-project.org/source
   9551d91..bf5d32a  master     -> origin/master
Updating 9551d91..bf5d32a
Fast-forward
 scripts/package-metadata.pl | 31 +++++++++++++++++++++++++------
 1 file changed, 25 insertions(+), 6 deletions(-)
dom@machina:~/Pulpit/Projekty/WITI-LEDE-REBOOT-small$ make package/symlinks V=s
./scripts/feeds update -a
Updating feed 'packages' from 'https://git.lede-project.org/feed/packages.git' ...
remote: Counting objects: 42, done.
remote: Compressing objects: 100% (40/40), done.
remote: Total 42 (delta 35), reused 3 (delta 1)
Unpacking objects: 100% (42/42), done.
From https://git.lede-project.org/feed/packages
   20ebb47..5a58466  master     -> origin/master
Updating 20ebb47..5a58466
Fast-forward
 lang/python/python/patches/001-enable-zlib.patch                                  |  7 +------
 lang/python/python/patches/002-do-not-add-include-dirs-when-cross-compiling.patch |  4 +---
 lang/python/python/patches/003-do-not-compile-tests-at-build.patch                |  6 ++----
 lang/python/python/patches/004-do-not-write-bytes-codes.patch                     |  6 ++----
 lang/python/python/patches/006-remove-multi-arch-and-local-paths.patch            |  2 --
 lang/python/python/patches/008-distutils-use-python-sysroot.patch                 |  8 ++------
 lang/python/python/patches/009-do-not-use-dblib_dir-when-cross-compiling.patch    |  8 +++-----
 lang/python/python/patches/010-do-not-add-rt-lib-dirs-when-cross-compiling.patch  |  4 +---
 lang/python/python/patches/011-remove-setupterm-definition.patch                  |  2 --
 lang/python/python3/patches/001-enable-zlib.patch                                 |  7 +------
 lang/python/python3/patches/003-do-not-run-distutils-tests.patch                  | 12 +++++-------
 lang/python/python3/patches/004-do-not-write-bytes-codes.patch                    |  6 ++----
 lang/python/python3/patches/006-remove-multi-arch-and-local-paths.patch           |  2 --
 lang/python/python3/patches/007-distutils-do-not-adjust-path.patch                |  6 +++---
 lang/python/python3/patches/008-fix-distutils-path-creation.patch                 |  2 --
 lang/python/python3/patches/010-do-not-add-rt-lib-dirs-when-cross-compiling.patch |  4 +---
 lang/python/python3/patches/011-fix-ncursesw-definition-colisions.patch           |  2 --
 lang/python/python3/patches/012-disable-wcsftime-detection.patch                  |  4 +---
 lang/python/python3/patches/014-remove-platform-so-suffix.patch                   |  8 ++------
 lang/python/python3/patches/015-abort-on-failed-modules.patch                     |  4 +---
 lang/python/python3/patches/016-adjust-config-paths.patch                         | 22 ++++++----------------
 sound/mpg123/Makefile                                                             |  4 ++--
 22 files changed, 36 insertions(+), 94 deletions(-)
Create index file './feeds/packages.index' 
Collecting package info: done
Updating feed 'luci' from 'https://git.lede-project.org/project/luci.git' ...
Already up-to-date.
Create index file './feeds/luci.index' 
Updating feed 'routing' from 'https://git.lede-project.org/feed/routing.git' ...
Already up-to-date.
Create index file './feeds/routing.index' 
Updating feed 'telephony' from 'https://git.lede-project.org/feed/telephony.git' ...
Already up-to-date.
Create index file './feeds/telephony.index' 
./scripts/feeds install -a
Collecting package info: done
WARNING: Makefile 'package/feeds/packages/python/python-cryptography/Makefile' has a build dependency on 'python-cffi/host' but 'package/feeds/packages/python/python-cffi/Makefile' does not implement a 'host' build type
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
dom@machina:~/Pulpit/Projekty/WITI-LEDE-REBOOT-small$ scripts/getver.sh
r4634-bf5d32a
```

hnyman, 2017-07-27:
No connection to #4637 , but caused by #4601

The missing dependency caused the phase2 buildbot to fail completely due to package dependency metadata logic error, but @jow- fixed that with https://github.com/lede-project/source/commit/bf5d32af2a675f7577b388b5eef2a11e6ce042eb

Now there is this new error message that clearly tells the problem.

commodo, 2017-07-27:
well, i have a changeset ready for this
will submit it

closed 2017-07-27 by hnyman