openwrt/packages #4641: xl2tp client fails to reconnect

rainbowjose, 2017-07-27:
When im manually trying to reconnect in luci - l2tp shows this error AND not tries to reconnect.

Thu Jul 27 14:30:31 2017 daemon.info xl2tpd[1412]: Disconnecting from 83.102.254.213, Local: 7779, Remote: 59015
Thu Jul 27 14:30:31 2017 daemon.debug xl2tpd[1412]: Terminating pppd: sending TERM signal to pid 9404
Thu Jul 27 14:30:31 2017 daemon.info xl2tpd[1412]: Connection 59015 closed to 83.102.254.213, port 1701 (Goodbye!)
Thu Jul 27 14:30:31 2017 daemon.info pppd[9404]: Terminating on signal 15
Thu Jul 27 14:30:31 2017 daemon.info pppd[9404]: Connect time 1145.6 minutes.
Thu Jul 27 14:30:31 2017 daemon.info pppd[9404]: Sent 378117980 bytes, received 2462457188 bytes.
Thu Jul 27 14:30:31 2017 daemon.notice netifd: Network device 'l2tp-VPN' link is down
Thu Jul 27 14:30:31 2017 daemon.notice pppd[9404]: Connection terminated.
Thu Jul 27 14:30:31 2017 daemon.info xl2tpd[1412]: Can not find tunnel 7779 (refhim=0)
Thu Jul 27 14:30:31 2017 daemon.debug xl2tpd[1412]: network_thread: unable to find call or tunnel to handle packet.  call = 59666, tunnel = 7779 Dumping.
Thu Jul 27 14:30:31 2017 daemon.err miniupnpd[2833]: Failed to get IP for interface l2tp-VPN
Thu Jul 27 14:30:31 2017 daemon.warn miniupnpd[2833]: SendNATPMPPublicAddressChangeNotification: cannot get public IP address, stopping
Thu Jul 27 14:30:31 2017 daemon.notice netifd: Interface 'VPN' is now down
Thu Jul 27 14:30:31 2017 daemon.notice netifd: Interface 'VPN' is setting up now
Thu Jul 27 14:30:31 2017 daemon.info dnsmasq[1582]: reading /tmp/resolv.conf.auto
Thu Jul 27 14:30:31 2017 daemon.info dnsmasq[1582]: using local addresses only for domain lan
Thu Jul 27 14:30:31 2017 daemon.info dnsmasq[1582]: using nameserver 213.234.192.8#53
Thu Jul 27 14:30:31 2017 daemon.info dnsmasq[1582]: using nameserver 85.21.192.3#53
Thu Jul 27 14:30:31 2017 daemon.info pppd[9404]: Exit.
Thu Jul 27 14:30:31 2017 daemon.notice xl2tpd[1412]: Connecting to host 83.102.254.213, port 1701
Thu Jul 27 14:30:31 2017 daemon.info xl2tpd[1412]: control_finish: Connection closed to 83.102.254.213, port 1701 (Reached configured max-tunnels limit), Local: 10032, Remote: 39679

yousong, 2017-07-27:
Looks like an old bug to me.  Please share also the xl2tpd full version in use with opkg.

smay4finger, 2017-08-06:
I can confirm that. Version of xl2tpd is devel-20151125-3

rainbowjose, 2017-08-06:
it happens when yo re using luci, not pure "network" config for connection.
In "network" you can type, for example "redial = 20" and then it work, but
luci works in other way, which i can understand.

rainbowjose, 2017-08-06:
*cant

On Sun, Aug 6, 2017 at 6:48 PM, Jose Rainbow <youlold@gmail.com> wrote:

> it happens when yo re using luci, not pure "network" config for
> connection. In "network" you can type, for example "redial = 20" and then
> it work, but luci works in other way, which i can understand.
>

yousong, 2017-08-07:
Hi, @rainbowjose , As with https://github.com/openwrt/packages/issues/1334#issuecomment-108875095, I could not find `Reached configured max-tunnels limit` anywhere in the source code of xl2tpd, can you find where those bits come from?

Please consider giving xl2tpd in LEDE 17.01 release a try as OpenWrt 15.01 is mostly in unmaintained state now.

@smay4finger , Can you post the relevant log please?

yousong, 2017-08-17:
Closing this for now.  Feel free to update this issue with more info then we can proceed.  Thanks

closed 2017-08-17 by yousong