openwrt/packages #4642: curl: https is broken (at least on image builder)

aasgit, 2017-07-27:
Maintainer:  could not find it sorry.

Environment:
```
root@lede-wyn:~# cat /etc/os-release                                                                                                                                                    
NAME="LEDE"
VERSION="SNAPSHOT, Reboot"
ID="lede"
ID_LIKE="lede openwrt"
PRETTY_NAME="LEDE Reboot SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="http://lede-project.org/"
BUG_URL="http://bugs.lede-project.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r4619-d72371e"
LEDE_BOARD="mvebu/generic"
LEDE_ARCH="arm_cortex-a9_vfpv3"
LEDE_TAINTS="no-all"
LEDE_DEVICE_MANUFACTURER="LEDE"
LEDE_DEVICE_MANUFACTURER_URL="http://lede-project.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="LEDE Reboot SNAPSHOT r4619-d72371e"
```

Description:

i build my images with the official Image Builder and, since about 10 days, the cURL package is compiled without 'https' and should link to mbedtls but it's broken (at least on image generated with image builder). In fact, any cURL with https fails because https it's not compiled for cURL. I was told that the package should link dynamically to mbedtls or something but (unless I'm blind) it's not there.

I was told that the compiled version works fine but, still, an offical tool seems to be broken and should be fixed. This is why I'm pointing this out, after trying to reach out in the IRC channel without luck. Every image I tried to build after the 18th of July has this issue.

Here's some info:

I noticed when DDNS stopped working:

```
root@lede-wyn:~# service ddns restart
root@lede-wyn:~#  221106 ERROR : cURL: libcurl compiled without https support - TERMINATE
```

cURL info:

```
root@lede-wyn:~# curl --version      
curl 7.54.1 (arm-openwrt-linux-gnu) libcurl/7.54.1
Release-Date: 2017-06-14
Protocols: file ftp http 
Features: IPv6 Largefile 

root@lede-wyn:~# ldd /usr/bin/curl   
	/lib/ld-musl-armhf.so.1 (0x7f620000)
	libcurl.so.4 => /usr/lib/libcurl.so.4 (0xb6ed8000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0xb6ebd000)
	libc.so => /lib/ld-musl-armhf.so.1 (0x7f620000)

root@lede-wyn:~# ldd /usr/lib/libcurl*
	ldd (0x7f5f9000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0xb6f71000)
	libc.so => ldd (0x7f5f9000)
```

Every package/dependency is installed correctly but still, the ssl/tls part is not working.

```
root@lede-wyn:~# opkg list-installed | egrep "ssl|curl|mbed|ddns"    
curl - 7.54.1-1
ddns-scripts - 2.7.6-16
libcurl - 7.54.1-1
libmbedtls - 2.5.1-1
libopenssl - 1.0.2k-1
libustream-openssl - 2016-07-02-ec80adaa-2
luci-app-ddns - 2.4.8-2
luci-ssl-openssl - git-17.194.28316-2224714-1
openssl-util - 1.0.2k-1
```

I'm available for more details and tests if needed.

I hope this is helpful enough and that the issue can be solved easily.

jammmet, 2017-08-17:
I experienced a similar issue when attempting to download a file from an https server directly with `curl`:
`curl: (77) Error reading ca cert file /etc/ssl/certs/ca-certificates.crt - mbedTLS: (-0x3E00) PK - Read/write of file failed`

install the `ca-bundle` package fixed it:
`opkg install ca-bundle`

aasgit, 2017-08-28:
hi @jammmet ca-bundle is installed already and I did still get the error. but now curl is working correcty , out of the blue. it's really weird. I have not changed anything. I'm going to install a fresh build in a minute and update this again afterwards.

aasgit, 2017-08-31:
all: curl has been working fine for the last few days and I can't explain/reproduce the issue anymore. closing.

closed 2017-08-31 by aasgit