openwrt/packages #4644: transmission: [rw]mem_max are set too low by default

neheb, 2017-07-27:
transmission throws an error when rmem_max and wmem_max are set too low (16Kb by default). eg. 

UDP Failed to set receive buffer: requested 4194304, got 262142 (tr-udp.c:75)
UDP Please add the line "net.core.rmem_max = 4194304" to /etc/sysctl.conf (tr-udp.c:80)
UDP Failed to set send buffer: requested 1048576, got 262142 (tr-udp.c:86)
UDP Please add the line "net.core.wmem_max = 1048576" to /etc/sysctl.conf (tr-udp.c:91)

Since openwrt is for embedded clients, I suggest adding a .conf file in sysctl.d with exactly these minimum values (higher is recommended).

RogueScholar, 2017-08-02:
My thought on this was slightly different. I was curious why the buffer sizes weren't handled with preargs in the init.d file that echo the larger default values to the appropriate kernel parameters in net.core? Even if they were commented out originally for the sake of maximizing backwards compatibility with memory-poor devices, a corresponding comment in the packaged settings file would suffice to make concerned users aware of their availability to resolve the syslog errors. That saves from adding another file to the package while still tying the buffer size changes only to when transmission is started.

neheb, 2017-08-02:
At the end of the day, leaving the default values cripples transmission. I just tested it and got way better speeds with a sysctl file. (From ~3 to ~8 MB/s).

closed 2017-11-22 by neheb

soursir1, 2018-04-29:
Do I need to adjust this setting and add both of those settings to /etc/sysctl.conf? My LEDE build is consistently running low on memory. 

Is this fixed in an upcoming update?

poranje, 2018-05-28:
This seems something for @Rosen Penev.

neheb, 2018-05-28:
I will do some rudimentary testing to see.

neheb, 2018-05-28:
I just tested. Removing the sysctl file just makes downloading slower. It still runs out of memory on a 512MB device though.

I don't have a good solution to this. Use a USB drive as swap?

RogueScholar, 2018-06-01:
@neheb That's exactly how I did it. It's inelegant, especially with the pitiful throughput from the controller in my R7000, but the dog will hunt.