openwrt/packages #4657: Unescaped left brace in regex is illegal here in regex

DocMAX, 2017-07-30:
Building on Arch Linux.
OpenWrt trunk

make[3]: Entering directory '/mnt/htpc/data-hdd/Sources/openwrt/trunk/feeds/packages/libs/sqlite3'
CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -mdsp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /mnt/htpc/data-hdd/Sources/openwrt/trunk/build_dir/target-mips_34kc+dsp_musl-1.1.16/sqlite-autoconf-3190300:sqlite-autoconf-3190300 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -DSQLITE_ENABLE_UNLOCK_NOTIFY=1 -DHAVE_ISNAN=1 -DHAVE_MALLOC_USABLE_SIZE=1  -I/mnt/htpc/data-hdd/Sources/openwrt/trunk/staging_dir/target-mips_34kc+dsp_musl-1.1.16/usr/include -I/mnt/htpc/data-hdd/Sources/openwrt/trunk/staging_dir/target-mips_34kc+dsp_musl-1.1.16/include -I/mnt/htpc/data-hdd/Sources/openwrt/trunk/staging_dir/toolchain-mips_34kc+dsp_gcc-5.3.0_musl-1.1.16/usr/include -I/mnt/htpc/data-hdd/Sources/openwrt/trunk/staging_dir/toolchain-mips_34kc+dsp_gcc-5.3.0_musl-1.1.16/include/fortify -I/mnt/htpc/data-hdd/Sources/openwrt/trunk/staging_dir/toolchain-mips_34kc+dsp_gcc-5.3.0_musl-1.1.16/include " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -mdsp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap /mnt/htpc/data-hdd/Sources/openwrt/trunk/build_dir/target-mips_34kc+dsp_musl-1.1.16/sqlite-autoconf-3190300:sqlite-autoconf-3190300 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -DSQLITE_ENABLE_UNLOCK_NOTIFY=1 -DHAVE_ISNAN=1 -DHAVE_MALLOC_USABLE_SIZE=1  -I/mnt/htpc/data-hdd/Sources/openwrt/trunk/staging_dir/target-mips_34kc+dsp_musl-1.1.16/usr/include -I/mnt/htpc/data-hdd/Sources/openwrt/trunk/staging_dir/target-mips_34kc+dsp_musl-1.1.16/include -I/mnt/htpc/data-hdd/Sources/openwrt/trunk/staging_dir/toolchain-mips_34kc+dsp_gcc-5.3.0_musl-1.1.16/usr/include -I/mnt/htpc/data-hdd/Sources/openwrt/trunk/staging_dir/toolchain-mips_34kc+dsp_gcc-5.3.0_musl-1.1.16/include/fortify -I/mnt/htpc/data-hdd/Sources/openwrt/trunk/staging_dir/toolchain-mips_34kc+dsp_gcc-5.3.0_musl-1.1.16/include " LDFLAGS="-L/mnt/htpc/data-hdd/Sources/openwrt/trunk/staging_dir/target-mips_34kc+dsp_musl-1.1.16/usr/lib -L/mnt/htpc/data-hdd/Sources/openwrt/trunk/staging_dir/target-mips_34kc+dsp_musl-1.1.16/lib -L/mnt/htpc/data-hdd/Sources/openwrt/trunk/staging_dir/toolchain-mips_34kc+dsp_gcc-5.3.0_musl-1.1.16/usr/lib -L/mnt/htpc/data-hdd/Sources/openwrt/trunk/staging_dir/toolchain-mips_34kc+dsp_gcc-5.3.0_musl-1.1.16/lib -znow -zrelro " make -j1 -C /mnt/htpc/data-hdd/Sources/openwrt/trunk/build_dir/target-mips_34kc+dsp_musl-1.1.16/sqlite-autoconf-3190300/. AR="mips-openwrt-linux-musl-gcc-ar" AS="mips-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=34kc -mdsp -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -iremap /mnt/htpc/data-hdd/Sources/openwrt/trunk/build_dir/target-mips_34kc+dsp_musl-1.1.16/sqlite-autoconf-3190300:sqlite-autoconf-3190300 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fpic -DSQLITE_ENABLE_UNLOCK_NOTIFY=1 -DHAVE_ISNAN=1 -DHAVE_MALLOC_USABLE_SIZE=1" LD=mips-openwrt-linux-musl-ld NM="mips-openwrt-linux-musl-gcc-nm" CC="mips-openwrt-linux-musl-gcc" GCC="mips-openwrt-linux-musl-gcc" CXX="mips-openwrt-linux-musl-g++" RANLIB="mips-openwrt-linux-musl-gcc-ranlib" STRIP=mips-openwrt-linux-musl-strip OBJCOPY=mips-openwrt-linux-musl-objcopy OBJDUMP=mips-openwrt-linux-musl-objdump SIZE=mips-openwrt-linux-musl-size CROSS="mips-openwrt-linux-musl-" ARCH="mips" ;
make[4]: Entering directory '/mnt/htpc/data-hdd/Sources/openwrt/trunk/build_dir/target-mips_34kc+dsp_musl-1.1.16/sqlite-autoconf-3190300'
 cd . && /bin/sh /mnt/htpc/data-hdd/Sources/openwrt/trunk/build_dir/target-mips_34kc+dsp_musl-1.1.16/sqlite-autoconf-3190300/missing automake-1.15 --foreign
Unescaped left brace in regex is illegal here in regex; marked by <-- HERE in m/\${ <-- HERE ([^ \t=:+{}]+)}/ at /mnt/htpc/data-hdd/Sources/openwrt/trunk/staging_dir/host/bin/automake-1.15 line 3938.
make[4]: *** [Makefile:387: Makefile.in] Error 1
make[4]: Leaving directory '/mnt/htpc/data-hdd/Sources/openwrt/trunk/build_dir/target-mips_34kc+dsp_musl-1.1.16/sqlite-autoconf-3190300'
make[3]: *** [Makefile:109: /mnt/htpc/data-hdd/Sources/openwrt/trunk/build_dir/target-mips_34kc+dsp_musl-1.1.16/sqlite-autoconf-3190300/.built] Error 2
make[3]: Leaving directory '/mnt/htpc/data-hdd/Sources/openwrt/trunk/feeds/packages/libs/sqlite3'
make[2]: *** [package/Makefile:197: package/feeds/packages/sqlite3/compile] Error 2
make[2]: Leaving directory '/mnt/htpc/data-hdd/Sources/openwrt/trunk'
make[1]: *** [package/Makefile:193: /mnt/htpc/data-hdd/Sources/openwrt/trunk/staging_dir/target-mips_34kc+dsp_musl-1.1.16/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/htpc/data-hdd/Sources/openwrt/trunk'
make: *** [/mnt/htpc/data-hdd/Sources/openwrt/trunk/include/toplevel.mk:194: world] Error 2

zorun, 2017-07-30:
This is a known incompatibility with recent versions of Perl.

It has been fixed in LEDE: https://git.lede-project.org/?p=source.git;a=commit;h=92c80f38cff3c20388f9ac13d5196f2745aeaf77

DocMAX, 2017-07-30:
OK, but this is not LEDE here...(?)

zorun, 2017-07-30:
This package feed is supposed to be compatible with both OpenWrt and LEDE :)

But in practice, the OpenWrt tree is more or less abandoned for now, so this kind of issue gets fixed much slowly (if at all).  I guess you can backport the fix to OpenWrt or switch to LEDE.

hnyman, 2017-07-30:
> OK, but this is not LEDE here...(?)

This is shared feed. Both for LEDE and the semi-deprecated old Openwrt code base. 

Openwrt main source has not been maintained/updated for more than a year and will be overwritten with LEDE when the merge happens. If you want up-to-date main source now, start using LEDE.

pprindeville, 2017-07-30:
Looking at the fix, I don't understand why the fix was `[{]` and not `\{` instead.

DocMAX, 2017-07-30:
ok switched to LEDE, works fine

closed 2017-08-05 by hnyman