openwrt/packages #4685: MWAN3 port forwarding and 6in4 connection not working properly

wackejohn, 2017-08-09:
-------------------------------

Maintainer: @feckert
Environment: 
                      LEDE Reboot SNAPSHOT r4683-370aacf532 / LuCI Master (git-17.219.56366-102199f) 
                      Running on an armada 370 device witch two pppoe wan interface,two tinc vpn connection and one 6in4 connection.
                
Description:
                     After update to the latest version of MWAN3,I found that all the ports forwarding to device behind LEDE router not working properly,but forwarding to router itself works.
                    And the 6in4 connection was broken after MWAN3 started,seems that it was the self loopback interface issue.The 6in4 connection always need to first **ifdown** the self interface and then **ifup** the self interface.After that the 6in4 connection working properly.

Trubleshooting Data:
```
Software versions : 

OpenWrt - LEDE Reboot SNAPSHOT r4683-370aacf532
LuCI - git-17.219.56366-102199f

mwan3 - 2.6-1
mwan3-luci - git-17.219.56366-102199f-1

Output of "cat /etc/config/mwan3" : 

config rule 'google_dns'
	option proto 'all'
	option family 'ipv4'
	option dest_ip '8.8.8.8,8.8.4.4,208.67.222.222,208.67.220.220,199.91.73.222,178.79.131.110'
	option sticky '1'
	option timeout '600'
	option use_policy 'wana_wanb'

config rule 'IPV6_tunnel_0'
	option dest_ip '216.218.221.6'
	option family 'ipv4'
	option proto 'all'
	option sticky '0'
	option use_policy 'wan2_only'

config rule 'IPV6_Tunnel_1'
	option family 'ipv4'
	option proto 'all'
	option sticky '0'
	option use_policy 'wan1_only'
	option dest_ip '66.220.7.82'

config rule 'ChinaMobile_DNS'
	option dest_ip '112.4.0.55,221.131.143.69'
	option family 'ipv4'
	option proto 'all'
	option sticky '0'
	option use_policy 'wan2_only'

config rule 'DNS'
	option dest_port '53'
	option family 'ipv4'
	option sticky '0'
	option proto 'udp'
	option use_policy 'wan1_only'

config rule 'gfwlist'
	option proto 'all'
	option ipset 'gfwlist'
	option family 'ipv4'
	option timeout '300'
	option use_policy 'wana_wanb'
	option sticky '0'

config rule 'HOMEServer_IPv6'
	option dest_ip 'tb.ipv6.sparqnet.net'
	option family 'ipv4'
[... 1098 lines cut ...]
64 bytes from 2607:f8b0:4007:807::200e: seq=0 ttl=57 time=229.371 ms
64 bytes from 2607:f8b0:4007:807::200e: seq=2 ttl=57 time=222.147 ms
64 bytes from 2607:f8b0:4007:807::200e: seq=3 ttl=57 time=208.618 ms
64 bytes from 2607:f8b0:4007:807::200e: seq=4 ttl=57 time=207.440 ms
64 bytes from 2607:f8b0:4007:807::200e: seq=5 ttl=57 time=225.007 ms
^C
--- ipv6.google.com ping statistics ---
7 packets transmitted, 5 packets received, 28% packet loss
round-trip min/avg/max = 207.440/218.516/229.371 ms
```
Prot forwording issue:
```
root@HOME-Router:~# telnet 192.168.1.2 22
SSH-2.0-OpenSSH_7.2
^C
Console escape. Commands are:

 l      go to line mode
 c      go to character mode
 z      suspend telnet
 e      exit telnet
```
As the config above,I forworded the 22 port to wan 223 port,but the 223 port at wan was timed out.

Need help to fix above issues.Thanks.

feckert, 2017-08-09:
What was your version of mwan3, when this setup worked without issues?

wackejohn, 2017-08-09:
Thanks for reply.
I used the 1.6.3 version of mwan3 and I want to upgrade to mwan3 2.x.

feckert, 2017-08-09:
There were a lot of changes made since version 1.6.3 and the maintainer was not me at this time!
So I will have a closer look what are the difference are and under stand your setup!

wackejohn, 2017-08-09:
Thanks again.
The major change was the **the_routable_loopback_self**,See https://wiki.openwrt.org/doc/howto/mwan3#the_routable_loopback_self.
All the strange problem was caused by this loopback interface.

feckert, 2017-08-09:
Only a first hind! I have seen that your self interface has the metric 5 and the route to 172.93.42.206 has the metric 0. Could you change this? Self metric 0 and 172.93.42.206 metric 5. Why do you need this

wackejohn, 2017-08-10:
I have change the self metric 0,and delete all the static route.
main routing table
```
root@HOME-Router:~# route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         192.168.0.1     0.0.0.0         UG    0      0        0 lo
default         1.23.227.114.br 0.0.0.0         UG    10     0        0 pppoe-wan1
default         112.3.84.1      0.0.0.0         UG    20     0        0 pppoe-wan2
default         192.168.100.1   0.0.0.0         UG    30     0        0 tinc_tun0
default         192.168.200.1   0.0.0.0         UG    40     0        0 tinc_tun1
default         192.168.195.1   0.0.0.0         UG    100    0        0 zt0
66.220.7.82     192.168.0.1     255.255.255.255 UGH   0      0        0 lo
112.3.84.1      *               255.255.255.255 UH    0      0        0 pppoe-wan2
114.227.23.1    *               255.255.255.255 UH    0      0        0 pppoe-wan1
192.168.1.0     *               255.255.255.0   U     0      0        0 br-lan
192.168.100.0   *               255.255.255.0   U     30     0        0 tinc_tun0
192.168.195.0   *               255.255.255.0   U     100    0        0 zt0
192.168.196.0   192.168.196.2   255.255.255.0   UG    0      0        0 tun0
192.168.196.2   *               255.255.255.255 UH    0      0        0 tun0
192.168.200.0   *               255.255.255.0   U     40     0        0 tinc_tun1
root@HOME-Router:~# 
```
```
root@HOME-Router:~# ping 66.220.7.82
PING 66.220.7.82 (66.220.7.82): 56 data bytes
^C
--- 66.220.7.82 ping statistics ---
2 packets transmitted, 0 packets received, 100% packet loss
root@HOME-Router:~# traceroute 66.220.7.82
traceroute to 66.220.7.82 (66.220.7.82), 30 hops max, 38 byte packets
 1  *  *  *
 2  *  *  *
 3  *  *  *
 4  *  *  *
 5  *  *  *
 6  *  *  *
 7  *  *  *
 8  *  *  *
 9  *  *  *
10  *  *  *
11  *  *  *
12  *  *  *
13  *  *  *
14  *  *  *
15  *  *  *
16  *  *  *
17  *  *  *
18  *  *  *
19  *  *  *
20  *  *  *
21  *  *  *
22  *  *  *
23  *  *  *
24  *  *  *
25  *  *  *
26  *  *  *
27  *  *  *
28  *  *  *
29  *  *  *
30  *  *  *
```

As you see,the 6in4 script always add a route to the main routing table use the lowest metric interface.
If I don't add the static route and change the metric lower than the self interface,the 66.220.7.82 will always use the main routing table which is the lo interface.
The 172.93.42.206 I have removed,seems that the tinc won't add route to the main routing table,I used
shadowvpn for my vpn connection which will add route to main routing table as the 6in4 script.

But after change the metric,the 6in4 script and port forwarding still not working.

feckert, 2017-08-10:
I have added a PR #4688 with some features and some bugfixes.
Can you try this and let me know if it es working.

Your use is very complex i need some time to review your setup.
I have never used 6to4! I will have a look what this service is doing to understand your issue :-)

wackejohn, 2017-08-10:
thanks,I will try later

feckert, 2017-08-10:
Could you also send me your firewall config of the router with "iptables-save > /tmp/iptables.txt".
I have a script which visualize the table in a graph with graphvz so it is easier for me to understand the package flow.

wackejohn, 2017-08-10:
the firewall config:
[iptables.txt](https://github.com/openwrt/packages/files/1214912/iptables.txt)

feckert, 2017-08-10:
This is the output of my script.

[image]

It LGTM i am nor sure if this has something to do with the mangle table. Could you remove the self interface from the firewall "lan" zone? Because the self interface only adds an ip to the "lo" which is not in a zone.

wackejohn, 2017-08-10:
I have remove the "self" zone in the firewall.
new firewall config:
[iptables.txt](https://github.com/openwrt/packages/files/1214970/iptables.txt)

wackejohn, 2017-08-10:
By the way, I have install the 2.6.2-2 version of mwan3,how to set local_source to "lan"?

feckert, 2017-08-10:
Add a new config section in mwan3 config and remove self interface from network config.
After 'lan' interface is configured by netifd the hotplug script of mwan3 is called and set the ip of lan to the loopback device.

```
config globals 'globals'
         option local_source 'lan'

```

wackejohn, 2017-08-10:
I have add the globals section,and restart the mwan3,but not working?need to reboot the router?

feckert, 2017-08-10:
You have to 
`ifdown lan `
then
`ifup lan`

or reboot

wackejohn, 2017-08-10:
I've did a reboot,seems that 6in4 and the router originated packets works fine,but the port forwarding still not woking.
```
root@HOME-Router:~# route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         192.168.1.1     0.0.0.0         UG    0      0        0 lo
default         180.115.229.1   0.0.0.0         UG    10     0        0 pppoe-wan1
default         1.36.66.223.sta 0.0.0.0         UG    20     0        0 pppoe-wan2
default         192.168.100.1   0.0.0.0         UG    30     0        0 tinc_tun0
default         192.168.200.1   0.0.0.0         UG    40     0        0 tinc_tun1
default         192.168.195.1   0.0.0.0         UG    100    0        0 zt0
66.220.7.82     180.115.229.1   255.255.255.255 UGH   10     0        0 pppoe-wan1
66.220.7.82     192.168.100.1   255.255.255.255 UGH   30     0        0 tinc_tun0
180.115.229.1   *               255.255.255.255 UH    0      0        0 pppoe-wan1
192.168.1.0     *               255.255.255.0   U     0      0        0 br-lan
192.168.100.0   *               255.255.255.0   U     30     0        0 tinc_tun0
192.168.195.0   *               255.255.255.0   U     100    0        0 zt0
192.168.196.0   192.168.196.2   255.255.255.0   UG    0      0        0 tun0
192.168.196.2   *               255.255.255.255 UH    0      0        0 tun0
192.168.200.0   *               255.255.255.0   U     40     0        0 tinc_tun1
223.66.36.1     *               255.255.255.255 UH    0      0        0 pppoe-wan2
root@HOME-Router:~# 
```
```
root@HOME-Router:~# ifdown wanc
root@HOME-Router:~# ifup wanc
root@HOME-Router:~# ping6 ipv6.google.com
PING ipv6.google.com (2607:f8b0:4007:80c::200e): 56 data bytes
64 bytes from 2607:f8b0:4007:80c::200e: seq=0 ttl=57 time=321.137 ms
64 bytes from 2607:f8b0:4007:80c::200e: seq=1 ttl=57 time=320.273 ms
64 bytes from 2607:f8b0:4007:80c::200e: seq=2 ttl=57 time=320.466 ms
^C
--- ipv6.google.com ping statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 320.273/320.625/321.137 ms
root@HOME-Router:~# 
```
```
root@HOME-Router:~# traceroute 8.8.8.8
traceroute to 8.8.8.8 (8.8.8.8), 30 hops max, 38 byte packets
 1  192.168.100.1 (192.168.100.1)  153.716 ms  155.459 ms  155.848 ms
 2  *  *  *
 3  te-3-0-4.cr2.lax2.us.zenlayer.net (192.254.84.145)  161.259 ms  te-2-0-4.cr2.lax2.us.zenlayer.net (192.254.84.141)  158.478 ms  192.254.84.157 (192.254.84.157)  157.784 ms
 4  192.254.81.27 (192.254.81.27)  156.623 ms  192.254.81.31 (192.254.81.31)  156.705 ms  156.270 ms
 5  google.as15169.any2ix.coresite.com (206.72.210.41)  156.135 ms  157.610 ms  161.729 ms
 6  108.170.247.193 (108.170.247.193)  158.410 ms  108.170.247.129 (108.170.247.129)  156.435 ms  108.170.247.161 (108.170.247.161)  156.372 ms
 7  216.239.62.101 (216.239.62.101)  156.407 ms  72.14.233.225 (72.14.233.225)  156.251 ms  209.85.255.191 (209.85.255.191)  158.043 ms
 8  google-public-dns-a.google.com (8.8.8.8)  156.209 ms  155.875 ms  156.256 ms
root@HOME-Router:~# 
```

feckert, 2017-08-10:
Nice that some issue are fixed :-)
But could the last remaining problem an iptables ipv4/ipv6 filter/nat problem?
Could you reach the port over another connection not 6to4?

tpham3783, 2017-08-10:
Florian,

Do you mind sharing the script that generated the iptables graph described
here:

https://user-images.githubusercontent.com/553091/29170973-1c503940-7ddb-11e7-8f78-f541109a40b1.png

I am tired of tracing the firewall rules within a command line terminal..

Thank you in advance,

Toan

wackejohn, 2017-08-10:
I tested from another router,can't reach the port from wan directly,but reach the port from tinc vpn conneciton.
```
root@HOME-Router_HB:~# telnet 180.115.229.26 223

^C
root@HOME-Router_HB:~# telnet 192.168.100.2 223
SSH-2.0-OpenSSH_7.2

Protocol mismatch.
Connection closed by foreign host
root@HOME-Router_HB:~# 
root@HOME-Router_HB:~# telnet 180.115.229.26 810

^C
root@HOME-Router_HB:~# telnet 192.168.100.2 810

HTTP/1.1 400 Bad Request
Content-Type: text/html
Connection: close
Date: Thu, 10 Aug 2017 14:09:01 GMT
Content-Length: 94

<HTML><HEAD>
<TITLE>400 Bad Request</TITLE>
</HEAD><BODY>
<H1>Bad Request</H1>
</BODY></HTML>
Connection closed by foreign host
```
But if there is no lo interface, the port forwarding works fine.

feckert, 2017-08-10:
@tpham3783 sure
This is not my own script i found it on the web
[graph-iptables.txt](https://github.com/openwrt/packages/files/1215197/graph-iptables.txt)

feckert, 2017-08-10:
@wackejohn mm this is strange! What is the difference! Will have to think about it.

wackejohn, 2017-08-10:
@feckert I found another issue,my openvpn server on the router not working,also because of the self interface...

before    ip route del default via 192.168.1.1 dev lo
```
Thu Aug 10 23:30:14 2017 daemon.notice openvpn(vpn_server)[2793]: 117.136.46.11:14076 TLS: Initial packet from [AF_INET]117.136.46.11:14076, sid=bd796cad 3e9a0fb0
Thu Aug 10 23:30:30 2017 daemon.err openvpn(vpn_server)[2793]: 192.168.1.225:48283 TLS Error: TLS key negotiation failed to occur within 60 seconds (check your network connectivity)
Thu Aug 10 23:30:30 2017 daemon.err openvpn(vpn_server)[2793]: 192.168.1.225:48283 TLS Error: TLS handshake failed
Thu Aug 10 23:30:30 2017 daemon.notice openvpn(vpn_server)[2793]: 192.168.1.225:48283 SIGUSR1[soft,tls-error] received, client-instance restarting
```

after      ip route del default via 192.168.1.1 dev lo

```
Thu Aug 10 23:31:14 2017 daemon.notice openvpn(vpn_server)[2793]: 117.136.46.11:14076 SIGUSR1[soft,tls-error] received, client-instance restarting
Thu Aug 10 23:34:27 2017 daemon.notice openvpn(vpn_server)[2793]: 117.136.66.54:53576 TLS: Initial packet from [AF_INET]117.136.66.54:53576, sid=09fc3e36 1defe6b6
Thu Aug 10 23:34:27 2017 daemon.notice openvpn(vpn_server)[2793]: 117.136.66.54:53576 VERIFY OK: depth=1, CN=Easy-RSA CA
Thu Aug 10 23:34:27 2017 daemon.notice openvpn(vpn_server)[2793]: 117.136.66.54:53576 VERIFY OK: depth=0, CN=client
Thu Aug 10 23:34:27 2017 daemon.notice openvpn(vpn_server)[2793]: 117.136.66.54:53576 peer info: IV_VER=2.5_master
Thu Aug 10 23:34:27 2017 daemon.notice openvpn(vpn_server)[2793]: 117.136.66.54:53576 peer info: IV_PLAT=android
Thu Aug 10 23:34:27 2017 daemon.notice openvpn(vpn_server)[2793]: 117.136.66.54:53576 peer info: IV_PROTO=2
Thu Aug 10 23:34:27 2017 daemon.notice openvpn(vpn_server)[2793]: 117.136.66.54:53576 peer info: IV_NCP=2
Thu Aug 10 23:34:27 2017 daemon.notice openvpn(vpn_server)[2793]: 117.136.66.54:53576 peer info: IV_LZ4=1
Thu Aug 10 23:34:27 2017 daemon.notice openvpn(vpn_server)[2793]: 117.136.66.54:53576 peer info: IV_LZ4v2=1
Thu Aug 10 23:34:27 2017 daemon.notice openvpn(vpn_server)[2793]: 117.136.66.54:53576 peer info: IV_LZO=1
Thu Aug 10 23:34:27 2017 daemon.notice openvpn(vpn_server)[2793]: 117.136.66.54:53576 peer info: IV_COMP_STUB=1
Thu Aug 10 23:34:27 2017 daemon.notice openvpn(vpn_server)[2793]: 117.136.66.54:53576 peer info: IV_COMP_STUBv2=1
Thu Aug 10 23:34:27 2017 daemon.notice openvpn(vpn_server)[2793]: 117.136.66.54:53576 peer info: IV_TCPNL=1
Thu Aug 10 23:34:27 2017 daemon.notice openvpn(vpn_server)[2793]: 117.136.66.54:53576 peer info: IV_GUI_VER=de.blinkt.openvpn_0.6.73
Thu Aug 10 23:34:27 2017 daemon.notice openvpn(vpn_server)[2793]: 117.136.66.54:53576 Control Channel: TLSv1.2, cipher TLSv1/SSLv3 ECDHE-RSA-AES256-GCM-SHA384, 2048 bit RSA
Thu Aug 10 23:34:27 2017 daemon.notice openvpn(vpn_server)[2793]: 117.136.66.54:53576 [client] Peer Connection Initiated with [AF_INET]117.136.66.54:53576
Thu Aug 10 23:34:27 2017 daemon.notice openvpn(vpn_server)[2793]: client/117.136.66.54:53576 MULTI_sva: pool returned IPv4=192.168.196.6, IPv6=(Not enabled)
Thu Aug 10 23:34:27 2017 daemon.notice openvpn(vpn_server)[2793]: client/117.136.66.54:53576 MULTI: Learn: 192.168.196.6 -> client/117.136.66.54:53576
Thu Aug 10 23:34:27 2017 daemon.notice openvpn(vpn_server)[2793]: client/117.136.66.54:53576 MULTI: primary virtual IP for client/117.136.66.54:53576: 192.168.196.6
Thu Aug 10 23:34:29 2017 daemon.notice openvpn(vpn_server)[2793]: client/117.136.66.54:53576 PUSH: Received control message: 'PUSH_REQUEST'
Thu Aug 10 23:34:29 2017 daemon.notice openvpn(vpn_server)[2793]: client/117.136.66.54:53576 SENT CONTROL [client]: 'PUSH_REPLY,dhcp-option DNS 192.168.1.1,route 0.0.0.0 0.0.0.0,redirect-gateway,route 192.168.196.0 255.255.255.0,topology net30,ping 10,ping-restart 120,ifconfig 192.168.196.6 192.168.196.5,peer-id 0,cipher AES-256-GCM' (status=1)
Thu Aug 10 23:34:29 2017 daemon.notice openvpn(vpn_server)[2793]: client/117.136.66.54:53576 Data Channel: using negotiated cipher 'AES-256-GCM'
Thu Aug 10 23:34:29 2017 daemon.notice openvpn(vpn_server)[2793]: client/117.136.66.54:53576 Data Channel Encrypt: Cipher 'AES-256-GCM' initialized with 256 bit key
Thu Aug 10 23:34:29 2017 daemon.notice openvpn(vpn_server)[2793]: client/117.136.66.54:53576 Data Channel Decrypt: Cipher 'AES-256-GCM' initialized with 256 bit key
```

feckert, 2017-08-11:
@wackejohn if you set in the "section mwan3 globals" the "option local_source" to "none" then the no ip will be set to the lo interface and so the router initiated traffic will always use the next default route he can find.
Short:
routing will not be made over the mwan3_hook mangle table

feckert, 2017-08-11:
> @feckert need new firewall config?
iptables.txt

Do you need a new picture? if so you could use the script that I have already posted in this thread:
https://github.com/openwrt/packages/files/1215197/graph-iptables.txt

To make the mwan3 working with openvpn i have add a PR https://github.com/lede-project/source/pull/1103 which will add an netifd-protohandler for openvpn. This could solve some issues with openvpn because the interface will be handled by netifd and so could be added as an interface to mwan3.
But I have only tested the client side, so router is working as an road warrior and not as an server.

wackejohn, 2017-08-11:
@feckert Thanks,I will try your PR to see the openvpn server work properly or not.
 But how to fix the port forwarding issue?

wackejohn, 2017-08-11:
@feckert I've tried your openvpn version,seems that not working.

feckert, 2017-08-11:
@wackejohn Because your setup is very complex. You should start with a small configuration.
For example 2 simple wan interfaces and then add some rules (mwan3 rules/ firewall protforwarding).
If this is working you can add some more interfaces zerotier or 6to4 .

I think it is more constructive to make it more systematical.

wackejohn, 2017-08-11:
But if I do ``` ip route del default via 192.168.1.1 dev lo```  openvpn and portforwarding both working...
And I'll try simple config to see the port forwarding can working with the lo interface or not.

wackejohn, 2017-08-12:
I've tried the simple config, prot forwarding working only if do the ``` ip route del default via 192.168.1.1 dev lo```

mwan3 troubleshooting data:
```
Software versions : 

OpenWrt - LEDE Reboot SNAPSHOT r4683-370aacf532
LuCI - git-17.219.56366-102199f

mwan3 - 2.6.2-2
mwan3-luci - git-17.222.49750-afe2d2c-1
git-17.219.56366-102199f-1

Output of "cat /etc/config/mwan3" : 

config globals 'globals'
         option local_source 'lan'

config rule 'loadbalance'
	option family 'ipv4'
	option proto 'all'
	option equalize '1'
	option sticky '0'
	option use_policy 'loadbalanced'

config policy 'wan1_only'
	option last_resort 'unreachable'
	option family 'ipv4'
	list use_member 'wan1_m1_w1'

config policy 'wan2_only'
	list use_member 'wan2_m1_w1'
	option family 'ipv4'
	option last_resort 'unreachable'

config policy 'loadbalanced'
	option last_resort 'unreachable'
	option family 'ipv4'
	list use_member 'wan1_m1_w1'
	list use_member 'wan2_m1_w1'

config member 'wan1_m1_w1'
	option metric '1'
	option interface 'wan1'
	option weight '1'

config member 'wan2_m1_w1'
	option interface 'wan2'
	option weight '1'
	option metric '1'

config interface 'wan1'
	option enabled '1'
	option family 'ipv4'
	list track_ip '114.114.114.114'
	list track_ip '114.114.115.115'
	option down '2'
	option up '2'
	option reliability '2'
	option size '56'
	option count '3'
	option timeout '2'
	option interval '1'
	option failure_interval '1'
	option recovery_interval '1'
	option flush_conntrack 'never'

config interface 'wan2'
	option enabled '1'
	option family 'ipv4'
[... 295 lines cut ...]
    0     0 MARK       all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xff00 /* wan2 1 1 */ MARK xset 0x200/0xff00

Chain mwan3_rules (1 references)
 pkts bytes target     prot opt in     out     source               destination         
  585 50664 mwan3_policy_loadbalanced  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xff00 /* loadbalance */

Chain qos_Default (0 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 CONNMARK   all  --  *      *       0.0.0.0/0            0.0.0.0/0            CONNMARK restore mask 0xf
    0     0 qos_Default_ct  all  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf
    0     0 MARK       udp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf0 length 0:500 MARK xset 0x22/0xff
    0     0 MARK       icmp --  *      *       0.0.0.0/0            0.0.0.0/0            MARK xset 0x11/0xff
    0     0 MARK       tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf0 tcp spts:1024:65535 dpts:1024:65535 MARK xset 0x44/0xff
    0     0 MARK       udp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf0 udp spts:1024:65535 dpts:1024:65535 MARK xset 0x44/0xff
    0     0 CONNMARK   all  --  *      *       0.0.0.0/0            0.0.0.0/0            CONNMARK save mask 0xff

Chain qos_Default_ct (1 references)
 pkts bytes target     prot opt in     out     source               destination         
    0     0 MARK       tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf tcp multiport ports 22,53 /* ssh, dns */ MARK xset 0x11/0xff
    0     0 MARK       udp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf udp multiport ports 22,53 /* ssh, dns */ MARK xset 0x11/0xff
    0     0 MARK       tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf tcp multiport ports 20,21,25,80,110,443,993,995 /* ftp, smtp, http(s), imap */ MARK xset 0x33/0xff
    0     0 MARK       tcp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf tcp multiport ports 5190 /* AOL, iChat, ICQ */ MARK xset 0x22/0xff
    0     0 MARK       udp  --  *      *       0.0.0.0/0            0.0.0.0/0            mark match 0x0/0xf udp multiport ports 5190 /* AOL, iChat, ICQ */ MARK xset 0x22/0xff
    0     0 CONNMARK   all  --  *      *       0.0.0.0/0            0.0.0.0/0            CONNMARK save mask 0xff
```

wackejohn, 2017-08-13:
@feckert 
The port forwarding issue was fixed by trun off the "Masquerading" in the firewall "lan" zone.
Now only the openvpn server not working properly....

tpham3783, 2017-08-13:
I saw this similar issue a few months back, but never had the time to look
deep into it.  I my case, I could not forward to the Lan's subnet, but it
worked fine if I were to redirect to 127.0.0.1

Btw, that's strange that your default route goes to the LAn's ip but on the
loopback interface. That route would not be reachable unless you run ipsec
tunnel

feckert, 2017-08-13:
@wackejohn there a lot of up and down and ping lost events on the interfaces! Why? You should change the interfaces settings. My are

> interval 30
> failure_interval 3
> recovery_interval 3

Additional it depends on the router because mwan3 is script!!

wackejohn, 2017-08-14:
@feckert Thanks.I'll try to change the interfaces settings later...Thanks again.

wackejohn, 2017-08-14:
@tpham3783 that lo interface is used to traffic initiated by the router itself from version 2.0 of mwan3.
see https://wiki.openwrt.org/doc/howto/mwan3#the_routable_loopback_self.
I don't know a lot about routing table,the self interface should not conflict with the lan interface,and this caused a lot of problems,but now with @feckert latest commits,all function working without issue.

closed 2017-08-14 by wackejohn

feckert, 2017-08-14:
@wackejohn Thanks for testing and feedback :-)
are there any issues left? openvpn?

wackejohn, 2017-08-14:
@feckert In my case  the mwan3 now working without issues(long time to ifup interface,i will teset later),the openvpn server issue have fixed by change the listen ip to "192.168.1.1".
see https://wiki.openwrt.org/doc/howto/mwan3#openvpn

tinysun, 2020-03-30:
i have the similar  issuse. I just one line pppoe and oneline map-e. I cannot do anything. mwan is not working,and wan I turn off the firewall,the router is down. I think map-e can be working,but there was many issuses in package.  map-e is long time to fix.it just use in a few area not like pppoe.
if good news please tell me you router setting, thank you.
by the way,I am set pppoe and map-e in two routers now.