openwrt/packages #4686: Strongswan: strongswan-mod-forecast

dulemis, 2017-08-09:
It would be nice to add this feature , as it makes routing multicast and/or broadcast over VPN possible. This could obviously be well used for clients browsing Windows network shares, or different streaming clients like Plex.

Strykar, 2017-08-15:
@dulemis It exists on LEDE Reboot 17.01.2
```
root@apu:~# opkg list|grep strongswan-mod-forecast
strongswan-mod-forecast - 5.5.1-1 - StrongSwan forward multi/broadcast traffic plugin
```

dulemis, 2017-08-16:
Hi, Im using Entware and when asked there i got answer that i should request it upstream (here).

stintel, 2017-08-23:
strongswan-mod-forecast exists for LEDE (both master and 17.01), and for OpenWrt snapshots.

closed 2017-08-23 by stintel

dulemis, 2017-08-23:
Hello again

This is how it looks for me on my device.

# opkg update
Downloading http://pkg.entware.net/binaries/armv7/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/packages

# opkg list | grep strongswan
strongswan - 5.5.2-1 - StrongSwan is an OpenSource IPsec implementation for the Linux operating system. This package contains shared libraries and scripts.
strongswan-charon - 5.5.2-1 - StrongSwan is an OpenSource IPsec implementation for the Linux operating system. This package contains charon, an IKEv2 keying daemon.
strongswan-libtls - 5.5.2-1 - StrongSwan is an OpenSource IPsec implementation for the Linux operating system. This package contains libtls for strongSwan plugins eap-tls, eap-ttls, eap-peap, tnc-tnccs
strongswan-mod-addrblock - 5.5.2-1 - StrongSwan RFC 3779 address block constraint support plugin
strongswan-mod-aes - 5.5.2-1 - StrongSwan AES crypto plugin
strongswan-mod-af-alg - 5.5.2-1 - StrongSwan AF_ALG crypto interface to Linux Crypto API plugin
strongswan-mod-agent - 5.5.2-1 - StrongSwan SSH agent signing plugin
strongswan-mod-attr - 5.5.2-1 - StrongSwan file based config plugin
strongswan-mod-attr-sql - 5.5.2-1 - StrongSwan SQL based config plugin
strongswan-mod-blowfish - 5.5.2-1 - StrongSwan Blowfish crypto plugin
strongswan-mod-ccm - 5.5.2-1 - StrongSwan CCM AEAD wrapper crypto plugin
strongswan-mod-cmac - 5.5.2-1 - StrongSwan CMAC crypto plugin
strongswan-mod-constraints - 5.5.2-1 - StrongSwan advanced X509 constraint checking plugin
strongswan-mod-coupling - 5.5.2-1 - StrongSwan IKEv2 plugin to couple peer certificates permanently to authentication plugin
strongswan-mod-ctr - 5.5.2-1 - StrongSwan Counter Mode wrapper crypto plugin
strongswan-mod-curl - 5.5.2-1 - StrongSwan cURL fetcher plugin plugin
strongswan-mod-des - 5.5.2-1 - StrongSwan DES crypto plugin
strongswan-mod-dhcp - 5.5.2-1 - StrongSwan DHCP based attribute provider plugin
strongswan-mod-dnskey - 5.5.2-1 - StrongSwan DNS RR key decoding plugin
strongswan-mod-duplicheck - 5.5.2-1 - StrongSwan advanced duplicate checking plugin
strongswan-mod-eap-identity - 5.5.2-1 - StrongSwan EAP identity helper plugin
strongswan-mod-eap-md5 - 5.5.2-1 - StrongSwan EAP MD5 (CHAP) EAP auth plugin
strongswan-mod-eap-mschapv2 - 5.5.2-1 - StrongSwan EAP MS-CHAPv2 EAP auth plugin
strongswan-mod-eap-radius - 5.5.2-1 - StrongSwan EAP RADIUS auth plugin
strongswan-mod-eap-tls - 5.5.2-1 - StrongSwan EAP TLS auth plugin
strongswan-mod-farp - 5.5.2-1 - StrongSwan fake arp respsonses plugin
strongswan-mod-fips-prf - 5.5.2-1 - StrongSwan FIPS PRF crypto plugin
strongswan-mod-gcm - 5.5.2-1 - StrongSwan GCM AEAD wrapper crypto plugin
strongswan-mod-gcrypt - 5.5.2-1 - StrongSwan libgcrypt plugin
strongswan-mod-gmp - 5.5.2-1 - StrongSwan libgmp plugin
strongswan-mod-gmpdh - 5.5.2-1 - StrongSwan DH-Groups; no libgmp dep plugin
strongswan-mod-ha - 5.5.2-1 - StrongSwan high availability cluster plugin
strongswan-mod-hmac - 5.5.2-1 - StrongSwan HMAC crypto plugin
strongswan-mod-kernel-libipsec - 5.5.2-1 - StrongSwan libipsec kernel interface plugin
strongswan-mod-kernel-netlink - 5.5.2-1 - StrongSwan netlink kernel interface plugin
strongswan-mod-ldap - 5.5.2-1 - StrongSwan LDAP plugin
strongswan-mod-led - 5.5.2-1 - StrongSwan LED blink on IKE activity plugin
strongswan-mod-load-tester - 5.5.2-1 - StrongSwan load testing plugin
strongswan-mod-md4 - 5.5.2-1 - StrongSwan MD4 crypto plugin
strongswan-mod-md5 - 5.5.2-1 - StrongSwan MD5 crypto plugin
strongswan-mod-mysql - 5.5.2-1 - StrongSwan MySQL database interface plugin
strongswan-mod-nonce - 5.5.2-1 - StrongSwan nonce genereation plugin
strongswan-mod-openssl - 5.5.2-1 - StrongSwan OpenSSL crypto plugin
strongswan-mod-pem - 5.5.2-1 - StrongSwan PEM decoding plugin
strongswan-mod-pgp - 5.5.2-1 - StrongSwan PGP key decoding plugin
strongswan-mod-pkcs1 - 5.5.2-1 - StrongSwan PKCS1 key decoding plugin
strongswan-mod-pkcs11 - 5.5.2-1 - StrongSwan PKCS11 key decoding plugin
strongswan-mod-pkcs12 - 5.5.2-1 - StrongSwan PKCS12 key decoding plugin
strongswan-mod-pkcs7 - 5.5.2-1 - StrongSwan PKCS7 key decoding plugin
strongswan-mod-pkcs8 - 5.5.2-1 - StrongSwan PKCS8 key decoding plugin
strongswan-mod-pubkey - 5.5.2-1 - StrongSwan raw public key plugin
strongswan-mod-random - 5.5.2-1 - StrongSwan RNG plugin
strongswan-mod-rc2 - 5.5.2-1 - StrongSwan RC2 crypto plugin
strongswan-mod-resolve - 5.5.2-1 - StrongSwan DNS resolver plugin
strongswan-mod-revocation - 5.5.2-1 - StrongSwan X509 CRL/OCSP revocation plugin
strongswan-mod-sha1 - 5.5.2-1 - StrongSwan SHA1 crypto plugin
strongswan-mod-sha2 - 5.5.2-1 - StrongSwan SHA2 crypto plugin
strongswan-mod-smp - 5.5.2-1 - StrongSwan SMP configuration and control interface plugin
strongswan-mod-socket-default - 5.5.2-1 - StrongSwan default socket implementation for charon plugin
strongswan-mod-socket-dynamic - 5.5.2-1 - StrongSwan dynamic socket implementation for charon plugin
strongswan-mod-sql - 5.5.2-1 - StrongSwan SQL database interface plugin
strongswan-mod-sqlite - 5.5.2-1 - StrongSwan SQLite database interface plugin
strongswan-mod-sshkey - 5.5.2-1 - StrongSwan SSH key decoding plugin
strongswan-mod-stroke - 5.5.2-1 - StrongSwan Stroke plugin
strongswan-mod-test-vectors - 5.5.2-1 - StrongSwan crypto test vectors plugin
strongswan-mod-unity - 5.5.2-1 - StrongSwan Cisco Unity extension plugin
strongswan-mod-updown - 5.5.2-1 - StrongSwan updown firewall plugin
strongswan-mod-whitelist - 5.5.2-1 - StrongSwan peer identity whitelisting plugin
strongswan-mod-x509 - 5.5.2-1 - StrongSwan x509 certificate plugin
strongswan-mod-xauth-eap - 5.5.2-1 - StrongSwan EAP XAuth backend plugin
strongswan-mod-xauth-generic - 5.5.2-1 - StrongSwan generic XAuth backend plugin
strongswan-mod-xcbc - 5.5.2-1 - StrongSwan xcbc crypto plugin
strongswan-utils - 5.5.2-1 - StrongSwan is an OpenSource IPsec implementation for the Linux operating system. This package contains the pki & scepclient utilities.

stintel, 2017-08-23:
This is an Entware problem, they seem to be using an outdated clone of this repo. Please take it up with them.

dulemis, 2017-08-23:
Oh, ok ,thank you!

dulemis, 2017-08-23:
One more question, is there any way possible to circumvent this and install it manually directly from source...?

dulemis, 2017-08-23:
One more question, is there any way possible to circumvent this and install it manually directly from your source...?

stintel, 2017-08-23:
Maybe you can add one of the arm package targets from https://downloads.lede-project.org/snapshots/packages/ to /etc/opkg/customfeeds.conf. You'll have to figure out which ones are compatible with your device.

Or you could simply use LEDE instead of Entware.

dulemis, 2017-10-03:
Sorry for my late reply. Thank you, I have looked through the link but unsure what's compatible. I'm using a Netgear R6250 router (	ARMv7 Processor rev 0 (v7l)), but can't find anything matching ARMv7?