openwrt/packages #4694: mwan3: second wan tracking gets killed when first goes down

nileshgr, 2017-08-12:
Maintainer: @feckert 

Description:

When first interfaces goes down, mwan3 kills the second interface's tracking daemon (mwan3track). 

Reason: 

```
Line 452 in lib/mwan3/mwan3.sh
kill $(pgrep -f "mwan3track $1") &> /dev/null
```

So if you have named your interfaces as wan & wan2, when wan (1) goes down, it will kill wan2 because pgrep matches wan2 daemon too (pattern matching).

Possible easiest solution:
Update documentation and sample configuration files to use interfaces as wan1 and wan2.

feckert, 2017-08-12:
I have already fixed this issue. https://github.com/openwrt/packages/commit/0b44ca505bdb882b7b2d68431ca8b2a739f6cc97

nileshgr, 2017-08-12:
> I have already fixed this issue. 0b44ca5

That's weird. I've been using trunk version and pulled about 2-3 days back.
You fixed it 12 days ago.

feckert, 2017-08-12:
I made the fix 12 days ago on my forked repository.
But i have pushed the changes on Thursday after testing.
The date use see on the commit are the changes when they are committed on the forked repository!
The date will not changed if the get merged in the upstream repository.

nileshgr, 2017-08-12:
Well since you have tested it I am closing this issue. I have already renamed my interfaces to wan1 & wan2. Not sufficient changes in core / packages for me to update my image. Saving flash cycles by not updating. 😀

closed 2017-08-12 by nileshgr

feckert, 2017-08-14:
@nileshgr thanks for feedback :-)