openwrt/packages #4705: unixodbc: fails to compile

michael-dev, 2017-08-13:
Maintainer: @dangowrt
Environment: LEDE df3295f50e54909090846de12f7deb3ff8de6557 , P1020WLAN

Description:

Failing to compile since commit 5cc52f06. Reverting that commit fixes issue for me.
See  https://github.com/openwrt/packages/commit/5cc52f06aa5499cabbce6311a3212aff1c731084#commitcomment-23623273

dangowrt, 2017-08-13:
Please pick up https://github.com/dangowrt/packages/commit/5f7f7cfa654d0acd780d2cc354e2beed46dfac21 and report back whether that fixes your build.

michael-dev, 2017-08-14:
Nope does not fix it

michael-dev, 2017-08-14:
Looking for __nldbl_printf it looks like it is not present in my host libc-2.19.so but is present in lede staging toolchain libc-2.25.so. So that is compiling using staging toolchain libc headers but not linking against a matching libc.

michael-dev, 2017-08-14:
Moving host build to a separate host target fixes this for me.

See https://github.com/michael-dev/packages/commit/4602f8683abb90bc38f128e2827b119299e355f3

dangowrt, 2017-08-14:
On Mon, Aug 14, 2017 at 08:14:32AM +0000, michael-dev wrote:
> Moving host build to a separate host target fixes this for me.
> 
> See https://github.com/michael-dev/packages/commit/4602f8683abb90bc38f128e2827b119299e355f3

Sure, but it doesn't solve the original problem which is that
odbc-config needs to be built against the config.h generated for the
target and hence return values which were detected by configure for
the target. But run on the host.
It's kinda from an pre pkg-config world and we hardly got that not
very cross-compile friendly mechanics anywhere (but in unixodbc and
postgresql).

michael-dev, 2017-08-14:
> Sure, but it doesn't solve the original problem which is that
> odbc-config needs to be built against the config.h generated for the
> target and hence return values which were detected by configure for
> the target. But run on the host.

Hm okay. So we need a HOST_BUILD_DEPEND=unixodbc to force the target version to be compiled first.
Then we can just copy over config.h in Host/Prepare after configure has run.
If some package needs odbc_config, it will need to depend on unixodbc/host. I cannot add this to PKG_BUILD_DEPEND as this would cause a loop. And adding the relevant $(Call ...) to force a host build from within a package build was not obvious to me.

See https://github.com/michael-dev/packages/commit/c06af7069611a29fb94d3fbbd9ca8d4ef1ec42c9

michael-dev, 2017-08-16:
See https://github.com/michael-dev/packages/commit/e002628cc27deb94a010955b10e443510d1a55ac

dangowrt, 2017-08-30:
thanks @michael-dev

commit b8f0e2ca (2017-08-30): unixodbc: fix host build

Fixed #4705

Reported-by: Michael Braun <michael-dev@fami-braun.de>
closed 2017-08-30 by dangowrt