openwrt/packages #4720: mwan3 script issue

kuoruan, 2017-08-17:
https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/mwan3.sh#L787
```[ -n "$($IP rule | awk '$1 == "'$(($id+1000)):'"')"i ]``` is always true.
and there are many ```-o``` ```-a``` in script which is not well defined in POSIX.

Do not use ```==``` to compare strings. It is only valid in a limited set of shells, and will produce unspecified behavior.

http://pubs.opengroup.org/onlinepubs/009695399/utilities/test.html

kuoruan, 2017-08-17:
@feckert

feckert, 2017-08-17:
@kuoruan 
I am not the original author of mwan3. I am the maintainer since this year.
Because the original author Adze1502 did not want to maintain the packages any more.

I know that there are some problems with POSIX ( for example == ).
If we change this we have to do testing. Some checks could be wrong after the change to POSIX!

My new added improvements will be checked with shellcheck https://github.com/koalaman/shellcheck/wiki. And should be fine

> https://github.com/openwrt/packages/blob/master/net/mwan3/files/lib/mwan3/mwan3.sh#L787

I think we should use the new ubus interface to check if interface is online/offline.
Feel free to add a PR.

closed 2017-08-18 by kuoruan