openwrt/packages #4734: libffmpeg-mini: not enabled to support MPEG-1/2 codec?

maitung, 2017-08-19:
Maintainer: @thess @antonlacon 
Environment:
  - Router: TPLINK TL-WR1043ND v4
  - Arch: mips_24kc
  - Target: ar71xx
  - LEDE: 17.01.2 stable release
  - libffmpeg-mini version 3.2.2-1
  - minidlna version 1.1.5-1

Hi thess and antonlacon,

After installing and configuring everything, I can start minidlna service successfully. The media server can stream various formats without a problem: mkv, avi, mp4, mp3. But my vcd and dvd clips are not shown on the database. I looked into the log file and saw these lines:

metadata.c:691: warn: Opening /mnt/sda1/sample.mpg failed! [Invalid data found when processing input]
scanner.c:490: warn: Unsuccessful getting details for /mnt/sda1/sample.mpg!

I decided to remove libffmpeg-mini and installed libffmpeg-full. Now everything works fine. So is this a bug? As I know libffmpeg-mini supports quite many popular formats, it should have support MPEG-1/2 too. If I am wrong, please help me how to configure/enable MPEG-1/2 codec.

Thank you very much for your time.

thess, 2017-08-19:
Hi - I say your posting of this issue in our forum and started to investigate but I don't have an environment to test it in. Perhaps Ian can help.

libffmpeg-mini does indeed have support for MPEG1/2. For vcd/vob formats, it is missing Program Stream (MPEGPS) support and possibly a related parser. I can make a proposed fix, but I would need testing help.

maitung, 2017-08-19:
Thanks for your quick reply!

I've tried removing and installing libffmpeg-mini several times but still yielded the same result, so I am pretty sure the library is installed successfully. My thought is the same with you, libffmpeg-mini should have enabled MPEG-1/2 codec by default, it is more enough than my need.

I have another router with USB port but I can't get it until the next few days. I will try to set up minidlna and see the result.

thess, 2017-08-19:
There is nothing wrong with the way you are installing libffmpeg-mini. The main reason you are unable to see your mpeg videos is a mis-configuration of the build which we are working on fixing. Stay tuned

maitung, 2017-08-20:
Thanks. I'm looking forward to your fix.

thess, 2017-08-22:
3.2.7-2 has a fix to include mpegps (Program Stream) support. You could help us out a bit here by either making your `sample.mpg` available for testing or post the output of something like __Mediainfo__ on this file.

I didn't mean to close this issue until it was fixed and verified by you - sorry.

maitung, 2017-08-22:
VLC provides media info as following:

### VCD clip
- Stream 0:
  Type: Video
  Codec: MPEG-1/2 Video (mpgv)
  Resolution: 352x256
  Display resolution: 352x240
  Frame rate: 29.970030
  Decoded format: Planar 4:2:0 YUV
- Stream 1:
  Type: Audio
  Codec: MPEG Audio layer 1/2 (mpga)
  Channels: Stereo
  Sample rate: 44100 Hz
  Bitrate: 224 kb/s

### DVD clip
- Stream 0:
  Type: Video
  Codec: MPEG-1/2 Video (mpgv)
  Resolution: 720x480
  Display resolution: 720x480
  Frame rate: 29.970030
  Decoded format: Planar 4:2:0 YUV
- Stream 1:
  Type: Audio
  Codec: A52 Audio (aka AC3) (a52 )
  Channels: Stereo
  Sample rate: 48000 Hz
  Bitrate: 448 kb/s

It's interesting that 2 vcd clips now can be played, but the rest are not. In the mean time, I'll make some sample clips for you to test and post here later.

maitung, 2017-08-22:
Please have a look at the attachment for some sample clips. They can't be played with version 3.2.7-2

[libffmpeg-mini_SampleTest.zip](https://github.com/openwrt/packages/files/1242932/libffmpeg-mini_SampleTest.zip)

antonlacon, 2017-08-22:
Did these samples / the originals work with libffmpeg-full?

The VCD sample may be failing because of the use of mpeg1 layer 2 audio, which isn't included in libffmpeg-mini. if the size increase is minimal, it can probably be added.

The DVD sample I'm unsure on and will need to find time to do testing.

maitung, 2017-08-23:
Hi @antonlacon 

libffmpeg-full works fine, the issue only happens with the mini lib.
The media info of the working vcd files is exactly the same with the one which I posted earlier. I don't understand why only 2 of them works with version 3.2.7-2 while the rest don't. Anyway, all my vcd files share the same media info, so do dvd files.

thess, 2017-08-29:
The vcd sample has mpeg1 layer 2 audio. Adding mpeg layer 2 audio is ~120bytes added to libavcodec.
See PR #4785.

thess, 2017-08-30:
@maitung - please re-test when libffmpeg-mini 3.2.7-3 is available.

I have tested this build with your 2 samples and they are now recognized correctly.

Please close this after you verify the results.

maitung, 2017-08-31:
I received this message: "Package libffmpeg-mini version 3.2.7-3 has no valid architecture, ignoring."
I tried editing opkg config by adding this line: "arch all 100" but had no success. opkg showed all installed packages name "has no valid architecture, ignoring." message.

Could you please add target, architecture like version 3.2.7-2?

thess, 2017-08-31:
You may have got ahead of the buildbots. Do an ``opkg update`` and try again. It looks like the package was built for your platform early this morning.

maitung, 2017-08-31:
I wget via this [url](http://downloads.lede-project.org/snapshots/packages/mipsel_24kc/packages/libffmpeg-mini_3.2.7-3_mipsel_24kc.ipk). Is this the correct build?
I'm just curious about the filename. Does 3.2.2-1_mips have any difference with 3.2.7-3_mipsel?
Thanks.

thess, 2017-08-31:
In a word - endianness 
Your TL router is big endian, mipsel is little endian - will not work on your system.

maitung, 2017-08-31:
I'm sorry for downloading the wrong build. I have 2 tabs open for release and snapshot url and didn't notice I had browsed on the wrong one.

Now it works perfectly. Thanks for your time and effort.

I have 1 more question. When is usually the fix be merged into the release branch?

PR #4735 (2017-08-20): ffmpeg: fixup mpeg support in libffmpeg-mini
Maintainer: me & @thess 
Compile tested: mips/ar71xx, LEDE trunk
Run tested: None

Description:

The demuxer mpegps is for VOB support and mpegvideo is for raw mpeg video.
mpegps was intended to be included with libffmpeg-mini; mpegvideo was not.

Should resolve #4734
closed 2017-08-31 by maitung