openwrt/packages #4747: <netcat> in listen mode udp - Illegal instruction

kofec, 2017-08-22:
@Adam Gensler
Always have below error:

root@LEDEnsa310Grawerska:~# tcpdump -vvv -l -u -p 162
tcpdump: syntax error in filter expression: syntax error
root@LEDEnsa310Grawerska:~# tcpdump -v -l -u -p 162
tcpdump: syntax error in filter expression: syntax error
root@LEDEnsa310Grawerska:~# tcpdump -vlu -p 162
tcpdump: syntax error in filter expression: syntax error
root@LEDEnsa310Grawerska:~# tcpdump -lu -p 162
tcpdump: syntax error in filter expression: syntax error
root@LEDEnsa310Grawerska:~# tcpdump -l -p 162
tcpdump: syntax error in filter expression: syntax error
root@LEDEnsa310Grawerska:~# nc -vvv -l -u -p 162
Listening on any address 162 (snmp-trap)
Received packet from 192.168.11.2:65534 -> 192.168.1.60:162 (local)
Illegal instruction
root@LEDEnsa310Grawerska:~# nc -vvv -lxu -p 162
Listening on any address 162 (snmp-trap)
Received packet from 192.168.11.2:65534 -> 192.168.1.60:162 (local)
Illegal instruction
root@LEDEnsa310Grawerska:~# nc -lxu -p 162
Illegal instruction
root@LEDEnsa310Grawerska:~# nc -lnxu -p 162
Illegal instruction
root@LEDEnsa310Grawerska:~# nc -lnxu -p 162 -o alala.txt
Illegal instruction
root@LEDEnsa310Grawerska:~# nc -vvlnxu -p 162
Listening on any address 162 (snmp-trap)
Received packet from 192.168.11.2:65534 -> 192.168.1.60:162 (local)
Illegal instruction

diizzyy, 2017-08-23:
a workaround is to use ncat in nmap-suit which also should be your preferred choice :-)

kofec, 2017-08-23:
@diizzyy thanks a lot for advice - I will test it today. Close the ticket

closed 2017-08-23 by kofec

kofec, 2017-08-23:
@diizzyy working great - thanks a lot

diizzyy, 2017-08-23:
@kofec You're welcome!