openwrt/packages #4748: bind: configure: error: found lmdb include but not library

stintel, 2017-08-23:
Maintainer: @nmeyerhans / cc @pprindeville 
Environment: LEDE master and 17.01

Description:
Commit eab56b6bee5d578ccf79f732317e612d91c24025 breaks build of bind(-tools).
```
stijn@taz ~/Development/LEDE/source $ make package/bind/{clean,compile} V=s
make[1]: Entering directory '/home/stijn/Development/LEDE/source'
make[2]: Entering directory '/home/stijn/Development/LEDE/openwrt-packages/net/bind'
rm -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/stamp/.bind_installed
rm -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/packages/bind.list /home/stijn/Development/LEDE/source/staging_dir/host/packages/bind.list
rm -rf /home/stijn/Development/LEDE/source/build_dir/target-mips64_octeon_64_musl-1.1.16/bind-9.11.2
make[2]: Leaving directory '/home/stijn/Development/LEDE/openwrt-packages/net/bind'
make[1]: Leaving directory '/home/stijn/Development/LEDE/source'
make[1]: Entering directory '/home/stijn/Development/LEDE/source'
make[2]: Entering directory '/home/stijn/Development/LEDE/source/package/libs/toolchain'
if [ -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean ]; then rm -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean; fi
echo "libc" >> /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install
if [ -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean ]; then rm -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean; fi
echo "libgcc" >> /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install
if [ -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean ]; then rm -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean; fi
echo "libatomic" >> /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install
if [ -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean ]; then rm -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean; fi
if [ -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean ]; then rm -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean; fi
echo "libstdcpp" >> /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install
if [ -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean ]; then rm -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean; fi
echo "libpthread" >> /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install
if [ -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean ]; then rm -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean; fi
if [ -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean ]; then rm -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean; fi
echo "librt" >> /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install
if [ -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean ]; then rm -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean; fi
if [ -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean ]; then rm -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean; fi
if [ -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean ]; then rm -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/toolchain.default.install.clean; fi
make[2]: Leaving directory '/home/stijn/Development/LEDE/source/package/libs/toolchain'
make[2]: Entering directory '/home/stijn/Development/LEDE/source/package/libs/openssl'
echo '1.0.2k' | cmp -s - /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/libopenssl.version || echo '1.0.2k' > /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/libopenssl.version
if [ -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/openssl.default.install.clean ]; then rm -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/openssl.default.install /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/openssl.default.install.clean; fi
echo "libopenssl" >> /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/openssl.default.install
if [ -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/openssl.default.install.clean ]; then rm -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/openssl.default.install /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/openssl.default.install.clean; fi
echo "openssl-util" >> /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/openssl.default.install
make[2]: Leaving directory '/home/stijn/Development/LEDE/source/package/libs/openssl'
make[2]: Entering directory '/home/stijn/Development/LEDE/source/package/libs/zlib'
if [ -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/zlib.default.install.clean ]; then rm -f /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/zlib.default.install /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/zlib.default.install.clean; fi
echo "zlib" >> /home/stijn/Development/LEDE/source/staging_dir/target-mips64_octeon_64_musl-1.1.16/pkginfo/zlib.default.install
[... cut ...]
configure: error: found lmdb include but not library.
make[2]: *** [Makefile:217: /home/stijn/Development/LEDE/source/build_dir/target-mips64_octeon_64_musl-1.1.16/bind-9.11.2/.configured_nyyyyyyyynyyyy] Error 1
make[2]: Leaving directory '/home/stijn/Development/LEDE/openwrt-packages/net/bind'
make[1]: *** [package/Makefile:106: package/feeds/packages/bind/compile] Error 2
make[1]: Leaving directory '/home/stijn/Development/LEDE/source'
make: *** [/home/stijn/Development/LEDE/source/include/toplevel.mk:200: package/bind/compile] Error 2
```

stintel, 2017-08-23:
@nmeyerhans Can I get your ACK for 5c9bc0d?

closed 2017-08-24 by stintel

pprindeville, 2017-08-24:
@stintel: thanks for fixing that!

commit fd709402 (2017-08-24): bind: disable lmdb

When building on hosts with lmdb installed, bind configure phase fails:

  configure: error: found lmdb include but not library.

Solve this by disabling lmdb. Fixes #4748.

Fixes: eab56b6bee5d ("bind: version update to 9.11.2")