openwrt/packages #4752: p910nd - Printer server doesnt start at boot anymore

Plasmaglow, 2017-08-23:
THIS: [https://dev.openwrt.org/ticket/14909](url)
No matter what, it's still broken! p910nd has to be disabled and re enabled via luci after EVERY reboot/power outtage or it won't work. Tested in LEDE 17 & OPENWRT Chaos Chalmer. Router: TP LINK 842 v1.

pprindeville, 2017-08-23:
What's in your `/etc/rc.d/` directory?

pprindeville, 2017-08-23:
For instance:

`ls -l /etc/rc.d/[SK]??p910nd`

Plasmaglow, 2017-08-23:
Hello Philip, i have to type ls -l /etc/rc.d/[SK]??p910nd via SSH and report outcome?

pprindeville, 2017-08-23:
It would definitely help, yes.

Plasmaglow, 2017-08-24:
sorry for the delay, got this:
[image]

pprindeville, 2017-08-24:
It's not enabled.

Try:

`enable p910nd`

and reboot and tell me if you still have this problem.

Plasmaglow, 2017-08-24:
[image]

pprindeville, 2017-08-24:
It doesn't need to be a screen shot.  Text is fine.

And what does `/var/log` look like?

`ls -ltr /var/log`

Plasmaglow, 2017-08-24:
sorry, my bad.
`root@LEDE:~# ls -ltr /var/log
-rw-r--r--    1 root     root             0 Aug  6 12:59 wtmp
-rw-r--r--    1 root     root             0 Aug  6 12:59 lastlog
`

pprindeville, 2017-08-24:
Okay, and what about:

`ls /etc/init.d /etc/rc.d`

I'm wondering what your logging service is...

Plasmaglow, 2017-08-24:
root@LEDE:~# ls /etc/init.d /etc/rc.d
/etc/init.d:
boot          dnsmasq       dropbear      gpio_switch   log           odhcpd        rpcd          sysfixtime    system        umount
cron          done          firewall      led           network       p910nd        sysctl        sysntpd       uhttpd        urandom_seed

/etc/rc.d:
K10gpio_switch   K89log           K98boot          S10boot          S12log           S19firewall      S50cron          S50uhttpd        S98gpio_switch
K50dropbear      K90network       K99umount        S10system        S12rpcd          S20network       S50dropbear      S95done          S98sysntpd
K85odhcpd        K90sysfixtime    S00sysfixtime    S11sysctl        S19dnsmasq       S35odhcpd        S50p910nd        S96led           S99urandom_seed

pprindeville, 2017-08-24:
Weird that your logger isn't running.  It's hard to tell what else might be misbehaving without that.

Do you have a serial console?

Any chance of capturing the boot-time messages from the console?

champtar, 2017-08-24:
Hey guys, not sure what you are trying to do, but you should just use ```logread``` no?

pprindeville, 2017-08-24:
We could try that.  Can you attach the output of `logread`?

Plasmaglow, 2017-08-24:
It's that the JTAG thing? If so then no :(
I'm convinced there is something wrong with p910nd not auto starting like it should. It's been like this forever but it's not a major problem just trying to give something back.

root@LEDE:~# logread
Wed Aug 23 20:35:52 2017 daemon.info dnsmasq-dhcp[3264]: DHCPOFFER(br-lan) 192.168.2.125 a4:70:d6:ba:2f:8e
Wed Aug 23 20:35:52 2017 daemon.info dnsmasq-dhcp[3264]: DHCPREQUEST(br-lan) 192.168.2.125 a4:70:d6:ba:2f:8e
Wed Aug 23 20:35:52 2017 daemon.info dnsmasq-dhcp[3264]: DHCPACK(br-lan) 192.168.2.125 a4:70:d6:ba:2f:8e android-c002c893f7f9ff6c
Wed Aug 23 20:35:53 2017 daemon.info odhcpd[710]: Using a RA lifetime of 0 seconds on br-lan
Wed Aug 23 20:38:53 2017 daemon.info hostapd: wlan0: STA 40:e2:30:a3:52:7f WPA: group key handshake completed (RSN)
Wed Aug 23 20:38:53 2017 daemon.info hostapd: wlan0: STA a4:70:d6:ba:2f:8e WPA: group key handshake completed (RSN)
Wed Aug 23 20:38:53 2017 daemon.info hostapd: wlan0: STA 5c:51:88:42:f0:f5 WPA: group key handshake completed (RSN)
Wed Aug 23 20:39:39 2017 daemon.info odhcpd[710]: Using a RA lifetime of 0 seconds on br-lan
Wed Aug 23 20:47:40 2017 daemon.info dnsmasq-dhcp[3264]: DHCPDISCOVER(br-lan) a4:70:d6:ba:2f:8e
Wed Aug 23 20:47:40 2017 daemon.info dnsmasq-dhcp[3264]: DHCPOFFER(br-lan) 192.168.2.125 a4:70:d6:ba:2f:8e
Wed Aug 23 20:47:40 2017 daemon.info dnsmasq-dhcp[3264]: DHCPREQUEST(br-lan) 192.168.2.125 a4:70:d6:ba:2f:8e
Wed Aug 23 20:47:40 2017 daemon.info dnsmasq-dhcp[3264]: DHCPACK(br-lan) 192.168.2.125 a4:70:d6:ba:2f:8e android-c002c893f7f9ff6c
Wed Aug 23 20:47:41 2017 daemon.info odhcpd[710]: Using a RA lifetime of 0 seconds on br-lan
Wed Aug 23 20:48:53 2017 daemon.info hostapd: wlan0: STA 40:e2:30:a3:52:7f WPA: group key handshake completed (RSN)
Wed Aug 23 20:48:53 2017 daemon.info hostapd: wlan0: STA 5c:51:88:42:f0:f5 WPA: group key handshake completed (RSN)
Wed Aug 23 20:48:56 2017 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED a4:70:d6:ba:2f:8e
Wed Aug 23 20:49:01 2017 daemon.info hostapd: wlan0: STA a4:70:d6:ba:2f:8e IEEE 802.11: deauthenticated due to local deauth request
Wed Aug 23 20:49:34 2017 daemon.info odhcpd[710]: Using a RA lifetime of 0 seconds on br-lan
Wed Aug 23 20:58:53 2017 daemon.info hostapd: wlan0: STA 40:e2:30:a3:52:7f WPA: group key handshake completed (RSN)
Wed Aug 23 20:58:53 2017 daemon.info hostapd: wlan0: STA 5c:51:88:42:f0:f5 WPA: group key handshake completed (RSN)
Wed Aug 23 20:59:02 2017 daemon.info odhcpd[710]: Using a RA lifetime of 0 seconds on br-lan
Wed Aug 23 21:07:14 2017 daemon.info odhcpd[710]: Using a RA lifetime of 0 seconds on br-lan
Wed Aug 23 21:08:53 2017 daemon.info hostapd: wlan0: STA 40:e2:30:a3:52:7f WPA: group key handshake completed (RSN)
Wed Aug 23 21:08:53 2017 daemon.info hostapd: wlan0: STA 5c:51:88:42:f0:f5 WPA: group key handshake completed (RSN)
Wed Aug 23 21:14:27 2017 daemon.info odhcpd[710]: Using a RA lifetime of 0 seconds on br-lan
Wed Aug 23 21:17:55 2017 daemon.info odhcpd[710]: Using a RA lifetime of 0 seconds on br-lan
Wed Aug 23 21:17:59 2017 daemon.info dnsmasq-dhcp[3264]: DHCPDISCOVER(br-lan) a4:70:d6:ba:2f:8e
Wed Aug 23 21:17:59 2017 daemon.info dnsmasq-dhcp[3264]: DHCPOFFER(br-lan) 192.168.2.125 a4:70:d6:ba:2f:8e
Wed Aug 23 21:17:59 2017 daemon.info dnsmasq-dhcp[3264]: DHCPREQUEST(br-lan) 192.168.2.125 a4:70:d6:ba:2f:8e
Wed Aug 23 21:17:59 2017 daemon.info dnsmasq-dhcp[3264]: DHCPACK(br-lan) 192.168.2.125 a4:70:d6:ba:2f:8e android-c002c893f7f9ff6c
Wed Aug 23 21:18:00 2017 daemon.info odhcpd[710]: Using a RA lifetime of 0 seconds on br-lan
Wed Aug 23 21:18:53 2017 daemon.info hostapd: wlan0: STA 40:e2:30:a3:52:7f WPA: group key handshake completed (RSN)
Wed Aug 23 21:18:53 2017 daemon.info hostapd: wlan0: STA 5c:51:88:42:f0:f5 WPA: group key handshake completed (RSN)
Wed Aug 23 21:19:16 2017 daemon.info dnsmasq-dhcp[3264]: DHCPDISCOVER(br-lan) 1c:56:fe:7c:d1:98
Wed Aug 23 21:19:16 2017 daemon.info dnsmasq-dhcp[3264]: DHCPOFFER(br-lan) 192.168.2.195 1c:56:fe:7c:d1:98
Wed Aug 23 21:19:16 2017 daemon.info dnsmasq-dhcp[3264]: DHCPREQUEST(br-lan) 192.168.2.195 1c:56:fe:7c:d1:98
Wed Aug 23 21:19:16 2017 daemon.info dnsmasq-dhcp[3264]: DHCPACK(br-lan) 192.168.2.195 1c:56:fe:7c:d1:98 android-93ecade77a8b6e16
Wed Aug 23 21:19:17 2017 daemon.info odhcpd[710]: Using a RA lifetime of 0 seconds on br-lan
Wed Aug 23 21:24:41 2017 daemon.info dnsmasq-dhcp[3264]: DHCPDISCOVER(br-lan) 5c:51:88:42:f0:f5
Wed Aug 23 21:24:41 2017 daemon.info dnsmasq-dhcp[3264]: DHCPOFFER(br-lan) 192.168.2.131 5c:51:88:42:f0:f5
Wed Aug 23 21:24:41 2017 daemon.info dnsmasq-dhcp[3264]: DHCPREQUEST(br-lan) 192.168.2.131 5c:51:88:42:f0:f5
Wed Aug 23 21:24:41 2017 daemon.info dnsmasq-dhcp[3264]: DHCPACK(br-lan) 192.168.2.131 5c:51:88:42:f0:f5 android-ce1283b7c669c83b
Wed Aug 23 21:24:42 2017 daemon.info odhcpd[710]: Using a RA lifetime of 0 seconds on br-lan
Wed Aug 23 21:27:52 2017 daemon.info odhcpd[710]: Using a RA lifetime of 0 seconds on br-lan
Wed Aug 23 21:28:53 2017 daemon.info hostapd: wlan0: STA 40:e2:30:a3:52:7f WPA: group key handshake completed (RSN)
Wed Aug 23 21:28:56 2017 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 5c:51:88:42:f0:f5
Wed Aug 23 21:29:01 2017 daemon.info hostapd: wlan0: STA 5c:51:88:42:f0:f5 IEEE 802.11: deauthenticated due to local deauth request
Wed Aug 23 21:33:18 2017 daemon.info odhcpd[710]: Using a RA lifetime of 0 seconds on br-lan
[... cut ...]
Thu Aug 24 18:25:24 2017 daemon.info dnsmasq-dhcp[3611]: DHCPACK(br-lan) 192.168.2.102 10:a5:d0:34:53:73 android-108f488455d24bb2
Thu Aug 24 18:25:28 2017 daemon.info odhcpd[710]: Using a RA lifetime of 0 seconds on br-lan
Thu Aug 24 18:26:03 2017 daemon.info odhcpd[710]: Using a RA lifetime of 0 seconds on br-lan
Thu Aug 24 18:31:21 2017 daemon.info odhcpd[710]: Using a RA lifetime of 0 seconds on br-lan
Thu Aug 24 18:41:00 2017 daemon.info odhcpd[710]: Using a RA lifetime of 0 seconds on br-lan
root@LEDE:~#

pprindeville, 2017-08-24:
Sorry, my fault for not being more explict: please attach the output of `logread` immediately after a reboot.

Plasmaglow, 2017-08-24:
ok, the printer is most probably off (doing all this via teamviewer) but i can get someone turn it on if necessary. Will return with new logread.

Plasmaglow, 2017-08-24:
Thu Aug 24 10:35:03 2017 kern.notice kernel: [    0.000000] Linux version 4.4.71                                                                              (buildbot@builds-02.infra.lede-project.org) (gcc version 5.4.0 (LEDE GCC 5.4.0                                                                              r3101-bce140e) ) #0 Wed Jun 7 19:24:41 2017
Thu Aug 24 10:35:03 2017 kern.debug kernel: [    0.000000] MyLoader: sysp=f0f0f0                                                                             f0, boardp=f0f0f0f0, parts=f0f0f0f0
Thu Aug 24 10:35:03 2017 kern.info kernel: [    0.000000] bootconsole [early0] e                                                                             nabled
Thu Aug 24 10:35:03 2017 kern.info kernel: [    0.000000] CPU0 revision is: 0001                                                                             9374 (MIPS 24Kc)
Thu Aug 24 10:35:03 2017 kern.info kernel: [    0.000000] SoC: Atheros AR7241 re                                                                             v 1
Thu Aug 24 10:35:03 2017 kern.info kernel: [    0.000000] Determined physical RA                                                                             M map:
Thu Aug 24 10:35:03 2017 kern.info kernel: [    0.000000]  memory: 02000000 @ 00                                                                             000000 (usable)
Thu Aug 24 10:35:03 2017 kern.info kernel: [    0.000000] Initrd not found or em                                                                             pty - disabling initrd
Thu Aug 24 10:35:03 2017 kern.warn kernel: [    0.000000] No valid device tree f                                                                             ound, continuing without
Thu Aug 24 10:35:03 2017 kern.info kernel: [    0.000000] Zone ranges:
Thu Aug 24 10:35:03 2017 kern.info kernel: [    0.000000]   Normal   [mem 0x0000                                                                             000000000000-0x0000000001ffffff]
Thu Aug 24 10:35:03 2017 kern.info kernel: [    0.000000] Movable zone start for                                                                              each node
Thu Aug 24 10:35:03 2017 kern.info kernel: [    0.000000] Early memory node rang                                                                             es
Thu Aug 24 10:35:03 2017 kern.info kernel: [    0.000000]   node   0: [mem 0x000                                                                             0000000000000-0x0000000001ffffff]
Thu Aug 24 10:35:03 2017 kern.info kernel: [    0.000000] Initmem setup node 0 [                                                                             mem 0x0000000000000000-0x0000000001ffffff]
Thu Aug 24 10:35:03 2017 kern.debug kernel: [    0.000000] On node 0 totalpages:                                                                              8192
Thu Aug 24 10:35:03 2017 kern.debug kernel: [    0.000000] free_area_init_node:                                                                              node 0, pgdat 803d04d0, node_mem_map 81000000
Thu Aug 24 10:35:03 2017 kern.debug kernel: [    0.000000]   Normal zone: 64 pag                                                                             es used for memmap
Thu Aug 24 10:35:03 2017 kern.debug kernel: [    0.000000]   Normal zone: 0 page                                                                             s reserved
Thu Aug 24 10:35:03 2017 kern.debug kernel: [    0.000000]   Normal zone: 8192 p                                                                             ages, LIFO batch:0
Thu Aug 24 10:35:03 2017 kern.warn kernel: [    0.000000] Primary instruction ca                                                                             che 64kB, VIPT, 4-way, linesize 32 bytes.
Thu Aug 24 10:35:03 2017 kern.warn kernel: [    0.000000] Primary data cache 32k                                                                             B, 4-way, VIPT, cache aliases, linesize 32 bytes
Thu Aug 24 10:35:03 2017 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32                                                                             768 u32768 alloc=1*32768
Thu Aug 24 10:35:03 2017 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Thu Aug 24 10:35:03 2017 kern.info kernel: [    0.000000] Built 1 zonelists in Z                                                                             one order, mobility grouping on.  Total pages: 8128
Thu Aug 24 10:35:03 2017 kern.notice kernel: [    0.000000] Kernel command line:                                                                              board=TL-MR3420  console=ttyS0,115200 rootfstype=squashfs,jffs2 noinitrd
Thu Aug 24 10:35:03 2017 kern.info kernel: [    0.000000] PID hash table entries                                                                             : 128 (order: -3, 512 bytes)
Thu Aug 24 10:35:03 2017 kern.info kernel: [    0.000000] Dentry cache hash tabl                                                                             e entries: 4096 (order: 2, 16384 bytes)
Thu Aug 24 10:35:03 2017 kern.info kernel: [    0.000000] Inode-cache hash table                                                                              entries: 2048 (order: 1, 8192 bytes)
Thu Aug 24 10:35:03 2017 kern.info kernel: [    0.000000] Writing ErrCtl registe                                                                             r=00000000
[... cut ...]
Thu Aug 24 19:06:09 2017 kern.info kernel: [   36.553050] br-lan: port 2(wlan0)                          entered forwarding state
Thu Aug 24 19:06:28 2017 kern.notice kernel: [   56.507065] random: nonblocking                          pool is initialized
Thu Aug 24 19:07:27 2017 authpriv.info dropbear[1192]: Child connection from 192                         .168.2.223:59983
Thu Aug 24 19:07:34 2017 authpriv.notice dropbear[1192]: Password auth succeeded                          for 'root' from 192.168.2.223:59983

pprindeville, 2017-08-25:
Okay, not seeing anything related to p910nd.

What do you see if you do:

```
/etc/init.d/p910nd start
logtail | tail -n20
```

Plasmaglow, 2017-08-27:
```
root@LEDE:~# /etc/init.d/p910nd start
root@LEDE:~# logtail | tail -n20
-ash: logtail: not found
```

pprindeville, 2017-08-27:
Sorry, meant `logread | tail -n20` right after doing the `start`.

Plasmaglow, 2017-09-08:
I'm back.

```
root@LEDE:~# /etc/init.d/p910nd start
root@LEDE:~# logread | tail -n20
```
Fri Sep  8 17:01:12 2017 daemon.info dnsmasq-dhcp[2850]: DHCPREQUEST(br-lan) 192.168.2.176 7c:91:22:ee:d4:4e
Fri Sep  8 17:01:12 2017 daemon.info dnsmasq-dhcp[2850]: DHCPACK(br-lan) 192.168.2.176 7c:91:22:ee:d4:4e android-55585e1b6d704299
Fri Sep  8 17:01:13 2017 daemon.info odhcpd[716]: Using a RA lifetime of 0 seconds on br-lan
Fri Sep  8 17:01:15 2017 daemon.info dnsmasq-dhcp[2850]: DHCPREQUEST(br-lan) 192.168.2.176 7c:91:22:ee:d4:4e
Fri Sep  8 17:01:15 2017 daemon.info dnsmasq-dhcp[2850]: DHCPACK(br-lan) 192.168.2.176 7c:91:22:ee:d4:4e android-55585e1b6d704299
Fri Sep  8 17:01:16 2017 daemon.info odhcpd[716]: Using a RA lifetime of 0 seconds on br-lan
Fri Sep  8 17:01:23 2017 daemon.info dnsmasq-dhcp[2850]: DHCPREQUEST(br-lan) 192.168.2.176 7c:91:22:ee:d4:4e
Fri Sep  8 17:01:23 2017 daemon.info dnsmasq-dhcp[2850]: DHCPACK(br-lan) 192.168.2.176 7c:91:22:ee:d4:4e android-55585e1b6d704299
Fri Sep  8 17:01:48 2017 daemon.info dnsmasq-dhcp[2850]: DHCPREQUEST(br-lan) 192.168.2.246 d0:a6:37:c6:f6:84
Fri Sep  8 17:01:48 2017 daemon.info dnsmasq-dhcp[2850]: DHCPACK(br-lan) 192.168.2.246 d0:a6:37:c6:f6:84 iPhonedrancisco
Fri Sep  8 17:01:48 2017 daemon.info odhcpd[716]: Using a RA lifetime of 0 seconds on br-lan
Fri Sep  8 17:01:52 2017 daemon.info odhcpd[716]: Using a RA lifetime of 0 seconds on br-lan
Fri Sep  8 17:01:56 2017 daemon.info odhcpd[716]: Using a RA lifetime of 0 seconds on br-lan
Fri Sep  8 17:03:25 2017 daemon.info dnsmasq-dhcp[2850]: DHCPREQUEST(br-lan) 192.168.2.246 d0:a6:37:c6:f6:84
Fri Sep  8 17:03:25 2017 daemon.info dnsmasq-dhcp[2850]: DHCPACK(br-lan) 192.168.2.246 d0:a6:37:c6:f6:84 iPhonedrancisco
Fri Sep  8 17:03:25 2017 daemon.info odhcpd[716]: Using a RA lifetime of 0 seconds on br-lan
Fri Sep  8 17:03:29 2017 daemon.info odhcpd[716]: Using a RA lifetime of 0 seconds on br-lan
Fri Sep  8 17:03:34 2017 daemon.info odhcpd[716]: Using a RA lifetime of 0 seconds on br-lan
Fri Sep  8 17:06:56 2017 daemon.info hostapd: wlan0: STA 40:e2:30:a3:52:7f WPA: group key handshake completed (RSN)
Fri Sep  8 17:08:47 2017 daemon.info odhcpd[716]: Using a RA lifetime of 0 seconds on br-lan

pprindeville, 2017-09-11:
Without getting into the box, it's a little hard to diagnose.

Can you get me access to it?

diizzyy, 2017-12-12:
I think we can close this due to reporter timeout.
p910nd starts fine on my boxes at least.

FrancYescO, 2018-07-18:
i have the same issue on r7490, p910 is not starting have to manually start it each reboot, it will start with no issue ```
root@Xiaomi_Mir3G:~# /etc/init.d/p910nd start
root@Xiaomi_Mir3G:~# ps | grep p910nd
 9493 root       908 S    {p910nd} /usr/sbin/p9100d -d -b -f /dev/usb/lp0 -i 1
 9511 root      1188 S    grep p910nd
```
but will just not autorun on boot

diizzyy, 2018-07-18:
https://github.com/openwrt/packages/blob/master/net/p910nd/files/p910nd.init#L3
In the init script, can you change start to 99 instead? 
There might be a race condition where the printer isn't detected when p910nd starts

FrancYescO, 2018-07-18:
Moved start to 99 and rebooted, not started
gone to the start page of luci and p910 was show as disabled, so enabled and rebooted, p910 started on boot!
but was to easy, so i moved back start to 50 (correctly shown as enabled in luci "start" page), rebooted and now is starting on boot

its just trolling me?

FrancYescO, 2018-07-19:
just done a sysupgrade to a newer snapshot and again the p910 deamon does not start on boot, so probably the correct fix should be to put the start to 99

pkerling, 2018-07-23:
@FrancYescO can you make a PR?

FrancYescO, 2018-07-23:
Jesus Travis. i just need to change two characters LOL

yousong, 2018-07-29:
Should be fixed by #6557

closed 2018-07-29 by yousong

paucoma, 2020-11-11:
I would like to add that in my particular scenario, with p910nd print server attached to interface **wan**, even with START=99 in the init.d script, when the wan connection is inexistant/missing at boot of the device the service fails to load and does not reload anymore.

 1. First I setup some additional p910nd logging [as described here](https://openwrt.org/docs/guide-user/services/print_server/p910nd.server#p910nd_process_stdout_to_system_log).
 2. Then I saw how the service crashed when it attempted to start without having the wan interface up:
```
11:10:00 daemon.info p910nd[1379]: bind: Address not available
11:10:00 daemon.info p910nd[1379]: accept: Bad file descriptor
11:10:11 daemon.info p910nd[1513]: bind: Address not available
11:10:11 daemon.info p910nd[1513]: accept: Bad file descriptor
11:10:16 daemon.info p910nd[1514]: bind: Address not available
11:10:16 daemon.info p910nd[1514]: accept: Bad file descriptor
11:10:21 daemon.info p910nd[1515]: bind: Address not available
11:10:21 daemon.info p910nd[1515]: accept: Bad file descriptor
11:10:26 daemon.info p910nd[1516]: bind: Address not available
11:10:26 daemon.info p910nd[1516]: accept: Bad file descriptor
11:10:26 daemon.info procd: Instance p910nd::p9100d s in a crash loop 6 crashes, 0 seconds since last crash
```
3. Restarting the service manually once the wan interface is up gets the print server up and running again.

My solution was to add the following to the end of my /etc/init.d/p910nd script:
```
service_triggers(){
        procd_add_interface_trigger "interface.*.up" wan "/etc/init.d/p910nd" restart
        logger -t p910nd "Restarting p910nd service due to up trigger of wan"
}
```
This now restarts the p910nd service when the wan interface goes up

I have documented my findings [on this other open issue](https://github.com/openwrt/packages/issues/7831#issuecomment-725573479)