openwrt/packages #4753: php-fpm:got sagement fault when i install and load 'php7-pecl-http'

805447391, 2017-08-23:
yesterday. i deploy nextcloud on my Raspberry Pi 3b (Firmare 17.0.2 LEDE musl, 32bit system run on aarch64 cpu),i have installed nginx and mysql-server,then i find if i do ```opkg install php7-pecl-http```and run php-* programs, i would get Sagement Fault ~but if i remove this package, it works normally.so, can you find out where is problem?thanks very much!

mhei, 2017-08-25:
@805447391: could you please post the output when you run `opkg list_installed` and the output when you run `ls -la /etc/php7/*` via SSH (or debug console). Thanks

805447391, 2017-08-26:
i have fixed this problem by rebuilding package php-pecl-http using my lede sdk, it works normally now, thanks very much, best wishes!

closed 2017-08-26 by 805447391