openwrt/packages #4754: gcc:

805447391, 2017-08-23:
target 17.0.2 bcm2710
32bit system in aarch64 cpu

problem : gcc could not compile helloworld.c
ps: if i run ```gcc hellowprld.c --mfloat-abi=hard -mfpu=neon-vfpv4``` is ok.
can you give me a good idea to avoid append these paramters or fix this 'bug' ?

805447391, 2017-08-24:
i have fixed this problem after reinstalling gcc. but i dont know what is the reason of this .

805447391, 2017-08-24:
sorry, i just set a alias for gcc, this problem still exist...

805447391, 2017-08-24:
but when i use snapshots mirror :  http://downloads.lede-project.org/snapshots/packages/arm_cortex-a53_neon-vfpv4/packages/gcc_5.4.0-2_arm_cortex-a53_neon-vfpv4.ipk
IT IS OKAY ~WTF

805447391, 2017-08-24:
i guess there is a small bug in gcc which in package17.0.1

diizzyy, 2017-12-12:
@champtar @hnyman 
If anything, this is a toolchain issue. Can you close?

closed 2017-12-12 by 805447391