openwrt/packages #4759: SOX did not install on ReSpeaker

algomusic, 2017-08-25:
Maintainer: @thess 
Environment: OpenWRT/LEDE version - (Chaos Calmer, v0.9.99)

Description:

Tried to install the SOX package on the ReSpeaker Core and got this error:

 * check_data_file_clashes: Package sox wants to install file /usr/bin/play
	But that file is already provided by package  * respeaker-os-file
 * opkg_install_cmd: Cannot install package sox.

thess, 2017-08-25:
Apparently, the core ReSpeaker os images already have a 'play' utility installed. You can overwrite it with
__--force-overwrite__ option to opkg or. better yet, rename the existing utility to another name before installing SOX.

There is not much guaranteed (or managed) file-name uniqueness between open-source applications. I think it is up to you to sort it out.

thess, 2017-08-28:
Closed - not a bug.

closed 2017-08-28 by thess