openwrt/packages #4762: wireguard: breaks network_find_wan shell function (from /lib/functions/network.sh)

stangri, 2017-08-26:
Maintainer: @danrl 
Environment: (ipq806x, Linksys EA8500, LEDE master snapshot)

Description:
When wireguard tunnel is established, it breaks the network_find_wan shell function (from /lib/functions/network.sh) -- it returns the name of the wireguard interface rather than the true WAN interface (thus breaking ```openvpn-policy-routing``` and potentially other packages).

If it's intentional/by-design/unavoidable, I'd like to know what other methods could be used to determine the name of the real WAN interface with wireguard running.

If that's a not expected behaviour, could anyone suggest how this can be fixed?

danrl, 2017-08-26:
Thanks for opening the bug, stan. I had a brief look at the function in question and it seems to take routes into account (via the network_gateway) function. Could you provide us with a minimal example configuration that produces reported bug? Please also include the routes of thw wireguard interface(s) that are wrongfully returned by the function.

As discussed via mail, I suspect the only factor to determine if an interface is a WAN interface is the default route. Depending on the definition, this would be expected behavior. However, I would like to verify if WireGuard interfaces behave differently than other tunnel interfaces with the same routes.

/cc @zorun @zx2c4

jow-, 2017-08-26:
This is a non-issue. The `network_find_wan` procedure is indeed meant to find the itnerface carrying the default route.

stangri, 2017-08-26:
@jow- However with the running OpenVPN tunnel with routing set to go to it, I still get valid WAN interface name from `network_find_wan`.

@danrl I'll switch from release to snapshots, run some tests and e-mail you later today.

stangri, 2017-08-27:
Non-issue, it was a misconfigured wireguard interface on my part which @danrl helped me to configure properly.

closed 2017-08-27 by stangri