openwrt/packages #4778: python: failed to build.

paulo-raca, 2017-08-28:
Maintainer: @commodo 
Environment: sunxi

Description:
I'm trying to build a system image with python3 package, but the build fails.

My .config is simply
```
CONFIG_TARGET_sunxi=y
CONFIG_PACKAGE_python3=y
```
(Followed by `make defconfig`)

Output:
```
[...]
running install_scripts
changing mode of /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/bin/2to3 to 755
changing mode of /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/bin/idle to 755
changing mode of /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/bin/pydoc to 755
changing mode of /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/bin/smtpd.py to 755
running install_egg_info
Removing /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/lib/python2.7/lib-dynload/Python-2.7.13-py2.7.egg-info
Writing /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/lib/python2.7/lib-dynload/Python-2.7.13-py2.7.egg-info
rm /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/lib/python2.7/lib-dynload/_sysconfigdata.py*
/usr/bin/install -c -m 644 ./Misc/python.man \
        /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/share/man/man1/python2.7.1
if test ! -d /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/lib/pkgconfig; then \
        echo "Creating directory /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/lib/pkgconfig"; \
        /usr/bin/install -c -d -m 755 /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/lib/pkgconfig; \
fi
if test -f /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/bin/python -o -h /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/bin/python; \
then rm -f /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/bin/python; \
else true; \
fi
(cd /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/bin; ln -s python2 python)
rm -f /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/bin/python2
(cd /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/bin; ln -s python2.7 python2)
rm -f /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/bin/python2-config
(cd /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/bin; ln -s python2.7-config python2-config)
rm -f /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/bin/python-config
(cd /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/bin; ln -s python2-config python-config)
test -d /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/lib/pkgconfig || /usr/bin/install -c -d -m 755 /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/lib/pkgconfig
rm -f /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/lib/pkgconfig/python2.pc
(cd /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/lib/pkgconfig; ln -s python-2.7.pc python2.pc)
rm -f /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/lib/pkgconfig/python.pc
(cd /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/lib/pkgconfig; ln -s python2.pc python.pc)
rm -f /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/share/man/man1/python2.1
(cd /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/share/man/man1; ln -s python2.7.1 python2.1)
rm -f /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/share/man/man1/python.1
(cd /home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/share/man/man1; ln -s python2.1 python.1)
if test "xinstall" != "xno"  ; then \
        case install in \
                upgrade) ensurepip="--upgrade" ;; \
                install|*) ensurepip="" ;; \
        esac; \
         ./python -E -m ensurepip \
                $ensurepip --root=/ ; \
fi
ERROR:root:code for hash sha224 was not found.
Traceback (most recent call last):
  File "/home/openwrt/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/Python-2.7.13/Lib/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
  File "/home/openwrt/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/Python-2.7.13/Lib/hashlib.py", line 97, in __get_builtin_constructor
    raise ValueError('unsupported hash type ' + name)
ValueError: unsupported hash type sha224
ERROR:root:code for hash sha256 was not found.
Traceback (most recent call last):
  File "/home/openwrt/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/Python-2.7.13/Lib/hashlib.py", line 147, in <module>
    globals()[__func_name] = __get_hash(__func_name)
[... 27 lines cut ...]
    _run_pip(args + [p[0] for p in _PROJECTS], additional_paths)
  File "/home/openwrt/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/Python-2.7.13/Lib/ensurepip/__init__.py", line 31, in _run_pip
    import pip
  File "/tmp/tmpFDNNKh/pip-9.0.1-py2.py3-none-any.whl/pip/__init__.py", line 21, in <module>
  File "/tmp/tmpFDNNKh/pip-9.0.1-py2.py3-none-any.whl/pip/_vendor/requests/__init__.py", line 62, in <module>
  File "/tmp/tmpFDNNKh/pip-9.0.1-py2.py3-none-any.whl/pip/_vendor/requests/packages/__init__.py", line 29, in <module>
ImportError: No module named urllib3
Makefile:927: recipe for target 'install' failed
make[4]: *** [install] Error 1
make[4]: Leaving directory '/home/openwrt/openwrt/build_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/Python-2.7.13'
Makefile:304: recipe for target '/home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/stamp/.python_installed' failed
make[3]: Leaving directory '/home/openwrt/openwrt/feeds/packages/lang/python/python'
make[3]: *** [/home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/host/stamp/.python_installed] Error 2
package/Makefile:196: recipe for target 'package/feeds/packages/python/host/compile' failed
make[2]: Leaving directory '/home/openwrt/openwrt'
make[2]: *** [package/feeds/packages/python/host/compile] Error 2
package/Makefile:193: recipe for target '/home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/stamp/.package_compile' failed
make[1]: Leaving directory '/home/openwrt/openwrt'
make[1]: *** [/home/openwrt/openwrt/staging_dir/target-arm_cortex-a8+vfpv3_musl-1.1.16_eabi/stamp/.package_compile] Error 2
make: *** [world] Error 2
/home/openwrt/openwrt/include/toplevel.mk:192: recipe for target 'world' failed
make: Leaving directory '/home/openwrt/openwrt'
The command '/bin/sh -c sudo -iu openwrt make -C openwrt -j1 V=s' returned a non-zero code: 2

```

commodo, 2017-08-28:
Could you try using the changes from this PR ? 
https://github.com/openwrt/openwrt/pull/508

I pushed that PR a while back, but it did not get much attention

paulo-raca, 2017-08-28:
Yup, that worked!

Sorry I missed it when looking for this issue

commodo, 2017-08-29:
No worries.
It's not a a visible PR it seems.

commodo, 2017-09-14:
@paulo-raca  
is there something else to do on this issue ? :)
if not, can we close ? 

thanks
Alex

paulo-raca, 2017-09-14:
Should we wait for https://github.com/openwrt/openwrt/pull/508 to close this?

If not, go on and close it.
Thanks again! :)

commodo, 2017-09-14:
ah ; right
forgot about it

sure ; let's wait :)
i was just grep-ing through issues that have `python` keyword in it
though, it feels like the re-merge of LEDE & OpenWrt will happen sooner than that PR
let's see

commodo, 2018-01-02:
happy new year :)

@paulo-raca 
OpenWrt was now replaced by former LEDE tree:
https://github.com/openwrt/openwrt/commits/master

LibreSSL is part of OpenWrt now;
https://github.com/openwrt/openwrt/tree/master/tools/libressl

this can be closed whenever :)

paulo-raca, 2018-01-04:
Sweet, thank you very much!
Closing

closed 2018-01-04 by paulo-raca