openwrt/packages #4795: LuCI GUI generates incorrect fwknopd conf parameters for dropdowns

leonyu, 2017-09-02:
Maintainer: @jp-bennett 
Environment: LEDE 17.01.2 on Linksys WRT1900AC

Description:

LuCI GUI generates incorrect fwknopd conf parameters for dropdowns -- specifically `Key type` & `HMAC Key type`.

Although benign for my use, it produces following error: 
```
Sat Sep  2 00:31:54 2017 daemon.err fwknopd[1844]: [*] Ignoring unknown access parameter: 'keytype' in /var/etc/access.conf
Sat Sep  2 00:31:54 2017 daemon.err fwknopd[1844]: [*] Ignoring unknown access parameter: 'hkeytype' in /var/etc/access.conf
```

With some minor digging, problem maybe in `fwknopd.lua` file.

jp-bennett, 2017-09-02:
I believe these parameters are stored for the luci interface, and are injected into the config file as a byproduct.  I'll take a closer look, but if I remember correctly, this is expected behavior.

dibdot, 2018-08-17:
See also https://github.com/openwrt/luci/issues/2062

closed 2018-08-17 by dibdot