openwrt/packages #4796: gen-qr.sh produces error output in BusyBox ash

leonyu, 2017-09-02:
Maintainer: @jp-bennett
Environment: LEDE 17.01.2 on Linksys WRT1900AC

Description:

When opening the LuCI config page for `fwknopd`, following error is produced: 
```
Sat Sep  2 00:50:11 2017 daemon.err uhttpd[1817]: uci: Entry not found
Sat Sep  2 00:50:11 2017 daemon.err uhttpd[1817]: uci: Entry not found
Sat Sep  2 00:50:11 2017 daemon.err uhttpd[1817]: sh: : unknown operand
Sat Sep  2 00:50:11 2017 daemon.err uhttpd[1817]: sh: : unknown operand
```
It appears the error originate from `/usr/sbin/gen-qr.sh`, as I am able to get the same error running that shell script in a terminal. 

I think the `unknown operand` error is caused by not quoting of the variable in comparison: 
```
if [ $key_base64 != "" ]; then
```
to 
```
if [ "$key_base64" != "" ]; then
```

I am not sure what causes `Entry not found` error.

hnyman, 2017-09-02:
Sounds like a possible duplicate of #3325 for luci-app-fwknopd, but that bug was supposed to be fixed in January.

jp-bennett, 2017-10-16:
The missing quotation marks is causing the unknown operator, but only when the key variables are empty.  This is essentially harmless noise, but for correctness I'll add the quotation marks as suggested.

jp-bennett, 2017-10-16:
Pull request to fix this is over at https://github.com/openwrt/luci/pull/1393

feckert, 2019-09-07:
This was fixed in https://github.com/openwrt/luci/pull/1393 and was merged so I will close this

closed 2019-09-07 by feckert