openwrt/packages #4803: ddns-scripts: Bugged interface prefix readout

LinuxfarmerHH, 2017-09-04:
Version: ddns-scripts | 2.7.6-17

Using dynv6.com for DynDNS services and got it to run with ssl. The ddns section is set to read the IP address from interface 'wan6'.

Overview for wan6
Type: dhcpv6-pd
Prefix Delegated: 2a02:xxxx:xxxx:xxxx::/56

Received at dynv6.com
Address/Prefix 2a02:xxxx:xxxx::1:f8

Logfile from ddns

ddns.myddns_ipv6.check_interval='30'
ddns.myddns_ipv6.domain='abcdef.dynv6.net'
ddns.myddns_ipv6.enabled='1'
ddns.myddns_ipv6.force_ipversion='1'
ddns.myddns_ipv6.interface='wan6'
ddns.myddns_ipv6.ip_network='wan6'
ddns.myddns_ipv6.ip_source='network'
ddns.myddns_ipv6.lookup_host='ipv6.dynv6.com'
ddns.myddns_ipv6.password='some_incrediblepass'
ddns.myddns_ipv6.retry_count='10'
ddns.myddns_ipv6.service_name='dynv6.com'
ddns.myddns_ipv6.use_https='1'
ddns.myddns_ipv6.use_ipv6='1'
ddns.myddns_ipv6=service
 180211       : verbose mode  : 0 - run normal, NO console output
 180211       : check interval: 1800 seconds
 180211       : force interval: 259200 seconds
 180211       : retry interval: 60 seconds
 180211       : retry counter : 10 times
 180211       : No old process
 180211       : last update: 2017-09-04 17:57
 180211       : Detect registered/public IP
 180211       : #> /usr/bin/host -t AAAA -6 ipv6.dynv6.com  >/var/run/ddns/myddns_ipv6.dat 2>/var/run/ddns/myddns_ipv6.err
 180211       : Registered IP '2a03:4000:xxxx:1db::' detected
 180211  info : Starting main loop at 2017-09-04 18:02
 180211       : Detect local IP on 'network'
 180211       : Local IP '2a02:xxxx:xxxx::1:f8' detected on network 'wan6'
 180211       : Update needed - L: '2a02:xxxx:xxxx:0000:0000:0000:0001:00f8' <> R: '2a03:4000:xxxx:01db:0000:0000:0000:0000'
 180212       : #> /usr/bin/wget-ssl -nv -t 1 -O /var/run/ddns/myddns_ipv6.dat -o /var/run/ddns/myddns_ipv6.err -6 --no-proxy 'https://dynv6.com/api/update?hostname=abcdefg.dynv6.net&token=some_incrediblepass&ipv6=2a02:xxxx:xxxx:0000:0000:0000:0001:00f8'
 180212       : DDNS Provider answered: addresses unchanged

chris5560, 2017-09-05:
You miss-configured your lookup_host.
You set to "ipv6.dynv6.com" instead of your "registered" hostname.

option lookup_host ""
# FQDN of ONE of your defined host at DDNS provider
# REQUIRED to verify what the current IP at DNS using nslookup/host command
# default: none

LinuxfarmerHH, 2017-09-05:
Have corrected this, but main problem did not change.
Local IP '2a02:xxxx:xxxx::1:f8' detected on network 'wan6'

This is sent to dynv6, in the config i cannot switch to br-lan what show the external IP within ifconfig.

chris5560, 2017-09-05:
Hi Eric,

sorry but you nowhere noted what is the correct IPv6.

Have you tried to detect your local IP via option ip_source='web' and setting corresponding option ip_url ?

LinuxfarmerHH, 2017-09-05:
Ah got it, ip_address_source needs to be 'interface' and interface need setting 'br-lan' to work correct. Now the dynv6 links to the right IPv6, thank you.

closed 2017-09-05 by LinuxfarmerHH