openwrt/packages #4810: 'ostiary' can not be installed on OpenWrt Chaos Calmer 15.05.1 (r49484)

oleg-sedletskyi, 2017-09-07:
Maintainer: Don't know (I contacted author of the software and he responded that someone associated with OpenWRT created the package. http://ingles.homeunix.net/)  
Environment: OpenWrt Chaos Calmer 15.05.1

Description:
I would like to install ostiary software on my router, but it seems that **ostiary** package is not available in the repositories anymore.
Here are some details: 
```
opkg update
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/base/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_base.
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/base/Packages.sig.
Signature check passed.
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/telephony/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_telephony.
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/telephony/Packages.sig.
Signature check passed.
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/packages/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_packages.
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/packages/Packages.sig.
Signature check passed.
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/routing/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_routing.
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/routing/Packages.sig.
Signature check passed.
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/luci/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_luci.
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/luci/Packages.sig.
Signature check passed.
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/management/Packages.gz.
Updated list of available packages in /var/opkg-lists/chaos_calmer_management.
Downloading http://downloads.openwrt.org/chaos_calmer/15.05.1/ar71xx/generic/packages/management/Packages.sig.
Signature check passed.
Downloading http://dl.eko.one.pl/chaos_calmer/ar71xx/packages/Packages.gz.
Updated list of available packages in /var/opkg-lists/eko1.
Downloading http://dl.eko.one.pl/chaos_calmer/ar71xx/packages/Packages.sig.
Signature check passed.
root@OpenWrt:~# opkg install ostiary
Unknown package 'ostiary'.
Collected errors:
 * opkg_install_cmd: Cannot install package ostiary.

```
I can remember I installed this package previously on my other router probably on previous release of OpenWRT.
In case this is the wrong place to raise the issue ‒ please let me know the right one.

hnyman, 2017-09-07:
Never heard of "ostiary" package. To my knowledge it has not been in the official packages feed repository. 

It seems to be still in the old abandoned "oldpackages" repository, that contains old packages that have not yet been imported here (since year 2014).
https://git.lede-project.org/?p=openwrt/packages.git;a=tree;f=admin/ostiary;h=79637ae91c9b413c7751d168691a310cf6c91495;hb=HEAD

Nobody has volunteered to be the maintainer of that package, so nobody has written a pull request to get it imported here. Note that the transfer here happened three years ago, so the package has not been missed by active users since then.

If you want the package here, volunteer to be the maintainer, import source code, compile and test the package (with current LEDE 17.01 or newer), and then write a pull request to get it added to the official repo.  (Note that it will never get backported to the old 15.05 branch, but be available just in the current LEDE master and the semi-deprecated Openwrt master)

Alternative is to edit your feeds.conf.default, and activate the oldpackages feed repo there and compile the package by yourself for your own build.

closed 2017-09-07 by hnyman

oleg-sedletskyi, 2017-09-08:
Thanks for information. 
I was able to install the package from the old repository:
http://downloads.openwrt.org/attitude_adjustment/12.09/ar71xx/generic/packages/
with the command:
opkg install http://downloads.openwrt.org/attitude_adjustment/12
.09/ar71xx/generic/packages/ostiary_4.0-1_ar71xx.ipk
I'm going to investigate how to migrate the package to the current release. 
Ostiary package is mentioned on the following page:
https://wiki.openwrt.org/doc/howto/secure.access
But I suspect that current development is LEDE and I may need to switch to that.