openwrt/packages #4819: collectd-mod-uptime: Reboot does not reset uptime value

rafwes, 2017-09-12:
Maintainer: @hnyman
Environment: ar71xx TP-Link WR842NDv1 LEDE 7.01.2

Description:

Uptime plugin values sent over the network plugin should reset to zero upon reboot. They don't.

hnyman, 2017-09-12:
Hmm. I guess that the uptime value in the rebooted router actually decreases to 0, but as you see in the network stats host the averaged time series results (of the last day or so), it does not seem to go down to 0, but just to some intermediate value between the old uptime and 0. Right?

It you look just the data of the last hour, does it go to 0?

I don't think that anything can be easily done. Possibly somebody might build complex logic for separating uptime stats from others, detect 0 and then clear old stats. But does not sound like worth the effort.

rafwes, 2017-09-12:
Hello! Thanks for the swift reply.

Actually I never see the uptime decrease. I left the router overnight without power and reconnected it again this morning. As you can see the uptime value is never close to zero. In fact it continues in a pretty linear fashion as if the router was never rebooted nor turned off but only lost connection to the server. 

I've added some grafana visualizations of what the last 15 minutes / 6 hours look like to me. There is also some [raw influxdb data here](https://pastebin.com/HU9r0ede) of the last 6 hours. 

[image]
[image]

hnyman, 2017-09-12:
Sounds strange, as the source router's collectd process should naturally have restarted when the router got again power, and should have started again from zero as the router's uptime should have been started again from 0.
https://github.com/collectd/collectd/blob/master/src/uptime.c

Is there anything non-standard in your config? Normal collectd plugin running in the router, no special database, just the standard RRD in ramdisk?  (what is influxdb????)

rafwes, 2017-09-12:
I am not using any RDD plugin, only collectd's network plugin to send data to a remote server. This remote server runs influxdb, a time series database that stores the values sent by the network plugin. That's where the database dump I sent you comes from.

On pastebin are all values received by influxdb in the last 6 hours (by then) that were then used to generate the graphs.

hnyman, 2017-09-14:
The really strange part here is how the router would report continuous uptime after being without power for some time.

I think that this is either something in upstream collectd itself, or related to your specific config. Soething in the reporting chain goes wrong.

hnyman, 2017-09-14:
I tested with my own routers, and got the same strange result:
the rrd database in the rebooted router shows uptime continuing from the old counter, like you said, instead of starting from 0.
```
                        <!-- 2017-09-14 23:02:30 EEST / 1505419350 --> <row><v> NaN </v></row>
                        <!-- 2017-09-14 23:03:00 EEST / 1505419380 --> <row><v> NaN </v></row>
                        <!-- 2017-09-14 23:03:30 EEST / 1505419410 --> <row><v> 9.9220000000e+03 </v></row>
                        <!-- 2017-09-14 23:04:00 EEST / 1505419440 --> <row><v> 9.9480000000e+03 </v></row>
                        <!-- 2017-09-14 23:04:30 EEST / 1505419470 --> <row><v> 9.9780000000e+03 </v></row>
```
The same is shown on the router running the network storage:
```
                        <!-- 2017-09-14 22:59:30 EEST / 1505419170 --> <row><v> 9.6780000000e+03 </v></row>
                        <!-- 2017-09-14 23:00:00 EEST / 1505419200 --> <row><v> 9.7080000000e+03 </v></row>
                        <!-- 2017-09-14 23:00:30 EEST / 1505419230 --> <row><v> 9.7380000000e+03 </v></row>
                        <!-- 2017-09-14 23:01:00 EEST / 1505419260 --> <row><v> NaN </v></row>
                        <!-- 2017-09-14 23:01:30 EEST / 1505419290 --> <row><v> NaN </v></row>
                        <!-- 2017-09-14 23:02:00 EEST / 1505419320 --> <row><v> NaN </v></row>
                        <!-- 2017-09-14 23:02:30 EEST / 1505419350 --> <row><v> NaN </v></row>
                        <!-- 2017-09-14 23:03:00 EEST / 1505419380 --> <row><v> 9.8920000000e+03 </v></row>
                        <!-- 2017-09-14 23:03:30 EEST / 1505419410 --> <row><v> 9.9180000000e+03 </v></row>
                        <!-- 2017-09-14 23:04:00 EEST / 1505419440 --> <row><v> 9.9480000000e+03 </v></row>
```
My guess is that somehow the networking functionality in collectd makes things to maintain smooth continuity, and that does not suit the uptime plugin.

hnyman, 2017-09-14:
Looks like the issue is genuine and is mentioned upstream in https://github.com/collectd/collectd/pull/2034

The bottom reason is weak compatibility of the collectd uptime plugin implementation with Openwrt/LEDE embedded routers where the correct time is set with NTP after collectd has already started. When collectd starts, the time is just the "last known time" set by the sysfixtime script, usually the date/time of the last file in /etc dir.

I tested with a single router, and after reboot the uptime plugin showed uptime calculated from the flashing date/time, instead of the recent reboot. I then touched a file in /etc and after reboot the uptime plugin correctly started from 0.

possible fix: https://github.com/collectd/collectd/pull/2034/commits/af01dd6fa3eb458e2fbb272703b0cae37ea54a9b

I will test the proposed upstream patch.

(A quick fix for you might be to touch any file in /etc before a planned reboot as that would provide the correct new "last known time" for the new reboot, or regularly touch a file e.g. every two hours, if you are really monitoring unexpected downtime)

rafwes, 2017-09-14:
Great news you could reproduce the bug and connect it to this merge request. Good job! Whenever I happen to setup a new system (it might take a while), I will do some testing on the proposed temporary fixes you recommended. Let me know if I can be of any further assistance. Thanks again.

hnyman, 2017-09-14:
Should be now fix. I applied the upstream patch.

Works ok for me:

[image]

commit 63a45e6d (2017-09-14): collectd: uptime plugin: apply fix from upstream

Uptime plugin fails to adjust for system time changes after boot.
As Openwrt/LEDE routers usually do not have a RTC, the system time
gets adjusted with NTP possibly after collectd has already started.
But collectd continues to use the initial time set by 'sysfixtime',
which can lead to incorrect uptime calculations.

Apply a proposed fix from upstream that uses /proc/uptime
Reference to https://github.com/collectd/collectd/pull/2034

Fixes #4819
closed 2017-09-14 by hnyman