openwrt/packages #4822: python-pip : error when building python with pip on buildroot missing tcl/tk libs

frymezim, 2017-09-12:
@commodo 
As title says it all.
```
rm -f /home/xxxx/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13/.built
touch /home/xxxx/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13/.built_check
CFLAGS="-Os -pipe -mcpu=xscale -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=soft -iremap/home/xxxx/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13:Python-2.7.13 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -DNDEBUG -fno-inline -I/home/xxxx/lede/source/staging_dir/target-arm_xscale_musl_eabi/usr/include -I/home/xxxx/lede/source/staging_dir/target-arm_xscale_musl_eabi/include -I/home/xxxx/lede/source/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl_eabi/usr/include -I/home/xxxx/lede/source/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl_eabi/include/fortify -I/home/xxxx/lede/source/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl_eabi/include " CXXFLAGS="-Os -pipe -mcpu=xscale -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=soft -iremap/home/xxxx/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13:Python-2.7.13 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -DNDEBUG -fno-inline -I/home/xxxx/lede/source/staging_dir/target-arm_xscale_musl_eabi/usr/include -I/home/xxxx/lede/source/staging_dir/target-arm_xscale_musl_eabi/include -I/home/xxxx/lede/source/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl_eabi/usr/include -I/home/xxxx/lede/source/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl_eabi/include/fortify -I/home/xxxx/lede/source/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl_eabi/include " LDFLAGS="-L/home/xxxx/lede/source/staging_dir/target-arm_xscale_musl_eabi/usr/lib -L/home/xxxx/lede/source/staging_dir/target-arm_xscale_musl_eabi/lib -L/home/xxxx/lede/source/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl_eabi/usr/lib -L/home/xxxx/lede/source/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl_eabi/lib -znow -zrelro -L/home/xxxx/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13" make  -C /home/xxxx/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13/. AR="arm-openwrt-linux-muslgnueabi-gcc-ar" AS="arm-openwrt-linux-muslgnueabi-gcc -c -Os -pipe -mcpu=xscale -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=soft -iremap/home/xxxx/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13:Python-2.7.13 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro" LD=arm-openwrt-linux-muslgnueabi-ld NM="arm-openwrt-linux-muslgnueabi-gcc-nm" CC="arm-openwrt-linux-muslgnueabi-gcc" GCC="arm-openwrt-linux-muslgnueabi-gcc" CXX="arm-openwrt-linux-muslgnueabi-g++" RANLIB="arm-openwrt-linux-muslgnueabi-gcc-ranlib" STRIP=arm-openwrt-linux-muslgnueabi-strip OBJCOPY=arm-openwrt-linux-muslgnueabi-objcopy OBJDUMP=arm-openwrt-linux-muslgnueabi-objdump SIZE=arm-openwrt-linux-muslgnueabi-size CROSS="arm-openwrt-linux-muslgnueabi-" ARCH="arm" CROSS_COMPILE=yes LD="arm-openwrt-linux-muslgnueabi-gcc" PGEN=pgen2 ;
make[4]: Entering directory '/home/xxxx/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13'
running build
running build_ext
building dbm using bdb
INFO: Can't locate Tcl/Tk libs and/or headers

Python build finished, but the necessary bits to build these modules were not found:
_tkinter           bsddb185           dl              
imageop            nis                sunaudiodev     
To find the necessary bits, look in setup.py in detect_modules() for the module's name.

running build_scripts
make[4]: Leaving directory '/home/xxxx/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13'
# Use host pip to install python-setuptools
/home/xxxx/lede/source/staging_dir/hostpkg/bin/pip install --ignore-installed --root=/home/xxxx/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13/install-setuptools --prefix=. /home/xxxx/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13/Lib/ensurepip/_bundled/setuptools-*.whl
bash: /home/xxxx/lede/source/staging_dir/hostpkg/bin/pip: No such file or directory
Makefile:306: recipe for target '/home/xxxx/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13/.built' failed
make[3]: *** [/home/xxxx/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13/.built] Error 127
make[3]: Leaving directory '/home/xxxx/lede/source/feeds/packages/lang/python/python'
package/Makefile:109: recipe for target 'package/feeds/packages/python/compile' failed
make[2]: *** [package/feeds/packages/python/compile] Error 2
make[2]: Leaving directory '/home/xxxx/lede/source'
package/Makefile:105: recipe for target '/home/xxxx/lede/source/staging_dir/target-arm_xscale_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/xxxx/lede/source/staging_dir/target-arm_xscale_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xxxx/lede/source'
/home/xxxx/lede/source/include/toplevel.mk:207: recipe for target 'world' failed
make: *** [world] Error 2
```

commodo, 2017-09-13:
i'll take a look

thanks for reporting

commodo, 2017-09-14:
the error here seems to be
```
bash: /home/xxxx/lede/source/staging_dir/hostpkg/bin/pip: No such file or directory
```

for some reason host pip did not build
i wonder why ?
can you search in the log ?

do you use the SDK build ? or do a full build ?

frymezim, 2017-09-14:
I'm building it through buildroot on latest git build (make menuconfig). Honestly I don't really know what's the cause of it but probably the reason that it can't find the file because as it needs the required library before it can build pip file.

Here's the log from make V=99
```
make[3]: Entering directory '/home/dkmori/lede/source/feeds/packages/lang/python/python'
make[3]: Leaving directory '/home/dkmori/lede/source/feeds/packages/lang/python/python'
make[3]: Entering directory '/home/dkmori/lede/source/feeds/packages/lang/python/python'
rm -f /home/dkmori/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13/.built
touch /home/dkmori/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13/.built_check
CFLAGS="-Os -pipe -mcpu=xscale -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=soft -iremap/home/dkmori/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13:Python-2.7.13 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -DNDEBUG -fno-inline -I/home/dkmori/lede/source/staging_dir/target-arm_xscale_musl_eabi/usr/include -I/home/dkmori/lede/source/staging_dir/target-arm_xscale_musl_eabi/include -I/home/dkmori/lede/source/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl_eabi/usr/include -I/home/dkmori/lede/source/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl_eabi/include/fortify -I/home/dkmori/lede/source/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl_eabi/include " CXXFLAGS="-Os -pipe -mcpu=xscale -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=soft -iremap/home/dkmori/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13:Python-2.7.13 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro -DNDEBUG -fno-inline -I/home/dkmori/lede/source/staging_dir/target-arm_xscale_musl_eabi/usr/include -I/home/dkmori/lede/source/staging_dir/target-arm_xscale_musl_eabi/include -I/home/dkmori/lede/source/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl_eabi/usr/include -I/home/dkmori/lede/source/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl_eabi/include/fortify -I/home/dkmori/lede/source/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl_eabi/include " LDFLAGS="-L/home/dkmori/lede/source/staging_dir/target-arm_xscale_musl_eabi/usr/lib -L/home/dkmori/lede/source/staging_dir/target-arm_xscale_musl_eabi/lib -L/home/dkmori/lede/source/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl_eabi/usr/lib -L/home/dkmori/lede/source/staging_dir/toolchain-arm_xscale_gcc-5.4.0_musl_eabi/lib -znow -zrelro -L/home/dkmori/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13" make  -C /home/dkmori/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13/. AR="arm-openwrt-linux-muslgnueabi-gcc-ar" AS="arm-openwrt-linux-muslgnueabi-gcc -c -Os -pipe -mcpu=xscale -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=soft -iremap/home/dkmori/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13:Python-2.7.13 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro" LD=arm-openwrt-linux-muslgnueabi-ld NM="arm-openwrt-linux-muslgnueabi-gcc-nm" CC="arm-openwrt-linux-muslgnueabi-gcc" GCC="arm-openwrt-linux-muslgnueabi-gcc" CXX="arm-openwrt-linux-muslgnueabi-g++" RANLIB="arm-openwrt-linux-muslgnueabi-gcc-ranlib" STRIP=arm-openwrt-linux-muslgnueabi-strip OBJCOPY=arm-openwrt-linux-muslgnueabi-objcopy OBJDUMP=arm-openwrt-linux-muslgnueabi-objdump SIZE=arm-openwrt-linux-muslgnueabi-size CROSS="arm-openwrt-linux-muslgnueabi-" ARCH="arm" CROSS_COMPILE=yes LD="arm-openwrt-linux-muslgnueabi-gcc" PGEN=pgen2 ;
make[4]: Entering directory '/home/dkmori/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13'
running build
running build_ext
building dbm using bdb
INFO: Can't locate Tcl/Tk libs and/or headers

Python build finished, but the necessary bits to build these modules were not found:
_tkinter           bsddb185           dl              
imageop            nis                sunaudiodev     
To find the necessary bits, look in setup.py in detect_modules() for the module's name.

running build_scripts
make[4]: Leaving directory '/home/dkmori/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13'
# Use host pip to install python-setuptools
/home/dkmori/lede/source/staging_dir/hostpkg/bin/pip install --ignore-installed --root=/home/dkmori/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13/install-setuptools --prefix=. /home/dkmori/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13/Lib/ensurepip/_bundled/setuptools-*.whl
bash: /home/dkmori/lede/source/staging_dir/hostpkg/bin/pip: No such file or directory
Makefile:306: recipe for target '/home/dkmori/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13/.built' failed
make[3]: *** [/home/dkmori/lede/source/build_dir/target-arm_xscale_musl_eabi/Python-2.7.13/.built] Error 127
make[3]: Leaving directory '/home/dkmori/lede/source/feeds/packages/lang/python/python'
package/Makefile:109: recipe for target 'package/feeds/packages/python/compile' failed
make[2]: *** [package/feeds/packages/python/compile] Error 2
make[2]: Leaving directory '/home/dkmori/lede/source'
package/Makefile:105: recipe for target '/home/dkmori/lede/source/staging_dir/target-arm_xscale_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/dkmori/lede/source/staging_dir/target-arm_xscale_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/dkmori/lede/source'
/home/dkmori/lede/source/include/toplevel.mk:207: recipe for target 'world' failed
make: *** [world] Error 2
```

commodo, 2017-09-14:
can you detail the steps ?
it sounds to me like buildroot is something different in your case

frymezim, 2017-09-14:
I just tick the python-pip <*> as it will add the prerequisites as well. The problem persists 3 months back and yeah python3 builds fine with pip. To be sure I did make distclean as well.

Anyway I will also try to build #4830 that was just now merge.

commodo, 2017-09-14:
that's not very explicit :)

1) do you download the LEDE repo from here ? http://github.com/lede-project/source/ 
2) is it trunk or LEDE 17.01 ?
3) did you install packages via `./scripts/feeds update packages ; ./scripts/feeds install -a` ; 
4) does this happen on a clean checkout ? i.e. download a fresh copy of LEDE, and run it ?

frymezim, 2017-09-14:
1)-2) Yes, I'm building through latest `https://git.lede-project.org/?p=source.git` from not the stable release.
3) I do `scripts/feeds/update -a` and `scripts/feeds install -a` as well.
4) Yes this is from clean copy of LEDE repo.

commodo, 2017-09-14:
cool :)
that info was helpful

so, i'm guessing you might have something installed on your system that may be interfering with the build

1) what OS are you running this on ?
2) can you share a full build log ? [from start to finish ]

frymezim, 2017-09-14:
1)
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.3 LTS
Release:	16.04
Codename:	xenial

2) https://pastebin.com/rpZDZTAb

commodo, 2017-09-15:
that log does not look like it's from a clean checkout
can you try a `make dirclean` and retry again ?

error is
```
bash: /home/xxxx/lede/source/staging_dir/hostpkg/bin/pip: No such file or directory
```
and i don't understand why it did not get built
it should have failed earlier

commodo, 2017-09-15:
can you send me your  .config file ?

frymezim, 2017-09-15:
Here
https://pastebin.com/9mKSUPgA

commodo, 2017-09-15:
With just that config I get and endless recursion of:
```
] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length 
```

If I do `make defconfig` it seems to work

frymezim, 2017-09-16:
I always do make defconfig after updating and reinstalling the packages.
It seems that it was building python-pip first before compiling python binaries.

commodo, 2017-09-16:
so, then is this fixed for you ? 
i don't have an idea of what the issue is or could be yet

frymezim, 2017-09-17:
The error was actually introduced by python-setuptools. I was able to compile python without it.

Here's the log
https://paste.ee/p/1wCzJ#WqOf9ThTKEOHoiH8fhiJX2fzL2HoMxik

commodo, 2017-09-18:
so, is there anything we can do from the python package side to address anything ?

if not, maybe we can close this ?

i'm still not sure what the issue is/was

btw: Python 2.7.14 is out, so, I'm planning to update it this week

closed 2017-09-18 by frymezim