openwrt/packages #4829: python-pyserial: and other python packages build fail

nxhack, 2017-09-14:
Maintainer: @commodo
Environment: ar71xx, LEDE head

Description:
python package build issues

python package cannot be built correctly now.

The package's Makefile has the following line.

```
$(call include_mk, python-package.mk)
```

rule.mk https://github.com/lede-project/source/blob/master/rules.mk#L357-L359
```
define include_mk
$(eval -include $(if $(DUMP),,$(STAGING_DIR)/mk/$(strip $(1))))
endef
```
Copy "packages/lang/python/python/files/*. mk" to "$ (STAGING_DIR)/mk/" and build normally.

nxhack, 2017-09-14:
https://github.com/openwrt/packages/issues/4548

https://github.com/openwrt/packages/commit/1588b532fb13393becf323784ef9d4ac2fb59230#diff-15f109d7b82fdaf9bfa1a18b417369ec

nxhack, 2017-09-14:
Hmm. When explicitly running  "python/host/compile" mk was copied. We will deal with package side.

closed 2017-09-14 by nxhack

commodo, 2017-09-14:
@nxhack 
maybe it makes sense to also copy them via `InstallDev` rule.
I'll make a change for that.

But there was a moment in time where it was needed before the `InstallDev`, sometime during some host-side build.
I forget why.

Thanks for reporting

nxhack, 2017-09-14:
Make sure you have the Makefile of python packages / modules
```
PKG_BUILD_DEPENDS:=python/host
```
It became necessary.

It is necessary to fix existing python module packages.

commodo, 2017-09-14:
So, the `python` package has `PKG_BUILD_DEPENDS:=python/host`.

I would hope that `python` building would ensure `python/host` to be built.
Maybe this is not true, in the case of an SDK.

nxhack, 2017-09-14:
> Maybe this is not true, in the case of an SDK.

In the current situation, Travis CI, which runs after pull request, does not work well.