openwrt/packages #4831: youtube-dl - problem dependency

build000, 2017-09-14:
```
make package/symlinks V=s./scripts/feeds update -a
Updating feed 'packages' from 'https://git.lede-project.org/feed/packages.git' ...
remote: Counting objects: 57, done.
remote: Compressing objects: 100% (43/43), done.
remote: Total 57 (delta 33), reused 19 (delta 10)
Unpacking objects: 100% (57/57), done.
From https://git.lede-project.org/feed/packages
   fd0f3aa..aaa6abf  master     -> origin/master
Updating fd0f3aa..aaa6abf
Fast-forward
 lang/python/python-curl/Makefile          | 47 +++++++++++++++++++++++++++++++++++++++++++++++
 lang/python/python/Makefile               | 26 ++++++++++++++------------
 lang/python/python/files/python-host.mk   | 24 +++++++++++++++---------
 lang/python/python3/Makefile              | 26 +++++++++++++-------------
 lang/python/python3/files/python3-host.mk | 24 +++++++++++++++---------
 mail/dovecot/Makefile                     |  6 +++---
 mail/pigeonhole/Makefile                  |  4 ++--
 multimedia/youtube-dl/Makefile            |  6 +++---
 8 files changed, 112 insertions(+), 51 deletions(-)
 create mode 100644 lang/python/python-curl/Makefile
Create index file './feeds/packages.index' 
Collecting package info: done
Updating feed 'luci' from 'https://git.lede-project.org/project/luci.git' ...
Already up-to-date.
Create index file './feeds/luci.index' 
Updating feed 'routing' from 'https://git.lede-project.org/feed/routing.git' ...
Already up-to-date.
Create index file './feeds/routing.index' 
Updating feed 'telephony' from 'https://git.lede-project.org/feed/telephony.git' ...
Already up-to-date.
Create index file './feeds/telephony.index' 
./scripts/feeds install -a
Collecting package info: done
WARNING: Makefile 'package/feeds/packages/youtube-dl/Makefile' has a build dependency on 'zip/host' but 'package/feeds/packages/zip/Makefile' does not implement a 'host' build type
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
```

hnyman, 2017-09-14:
Known problem. See #4828 and #4823

build000, 2017-09-14:
https://github.com/openwrt/packages/commit/7313348033e1bc53f0d4346036b5f98562b9379d
Problem probably fixed - close this issue.

closed 2017-09-14 by build000