openwrt/packages #4837: noddos: don't create /var/lib/noddos at build time

nwf, 2017-09-15:
Maintainer: @StevenHessing 
Environment: LEDE HEAD
Description:

The noddos Makefile currently, in `Package/noddos/install` creates `/var` as a directory, by invoking `$(INSTALL_DIR) $(1)/var/lib/noddos`.  This results in a build error if the package is installed (maybe not if it is only built as a package):

```
make[3]: Entering directory '/tank/openwrt/scratch/builder-usbnetgw2/feeds/packages/net/noddos'
mkdir -p /tank/openwrt/scratch/builder-usbnetgw2/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp
SHELL= flock /tank/openwrt/scratch/builder-usbnetgw2/tmp/.root-copy.flock -c 'cp -fpR /tank/openwrt/scratch/builder-usbnetgw2/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/noddos-0.4.1/.pkgdir/noddos/. /tank/openwrt/scratch/builder-usbnetgw2/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/'
cp: cannot overwrite non-directory '/tank/openwrt/scratch/builder-usbnetgw2/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/./var' with directory '/tank/openwrt/scratch/builder-usbnetgw2/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/noddos-0.4.1/.pkgdir/noddos/./var'
Makefile:85: recipe for target '/tank/openwrt/scratch/builder-usbnetgw2/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp/.noddos_installed' failed
make[3]: *** [/tank/openwrt/scratch/builder-usbnetgw2/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp/.noddos_installed] Error 1
make[3]: Leaving directory '/tank/openwrt/scratch/builder-usbnetgw2/feeds/packages/net/noddos'
package/Makefile:109: recipe for target 'package/feeds/packages/noddos/compile' failed
```

StevenHessing, 2017-09-15:
PR: https://github.com/openwrt/packages/pull/4838

champtar, 2017-09-15:
Thanks @StevenHessing

closed 2017-09-15 by champtar