openwrt/packages #4839: gst1-libav compile error

jhpark555, 2017-09-15:
When I compile in openWrt, I have a compile error like this.

Configuring included Libav instance with args --prefix=/usr         --enable-static --enable-pic         --disable-avdevice --disable-postproc         --disable-programs --disable-ffserver --disable-ffplay --disable-ffprobe --disable-ffmpeg         --disable-encoder=flac --disable-protocols --disable-devices         --disable-network --disable-hwaccels --disable-dxva2 --disable-vdpau         --disable-filters --enable-filter=yadif --disable-doc --disable-vda --disable-d3d11va --disable-dxva2         --disable-iconv --disable-optimizations --enable-cross-compile         --target-os=linux --arch=arm --cross-prefix=arm-openwrt-linux- --ar="$AR" --as="$CC" --cc="$CC" --ld="$CC" --nm="$NM" --disable-ffmpeg --target-os=linux --cpu=cortex-a7+neon-vfpv4 --disable-bsfs --disable-programs --disable-devices --disable-encoders                               --disable-decoders  --enable-decoder='aac'   --enable-decoder='ac3'   --enable-decoder='atrac3'      --enable-decoder='h264'   --enable-decoder='jpegls'      --enable-decoder='mp3'   --enable-decoder='mpeg1video'   --enable-decoder='mpeg2video'   --enable-decoder='mpeg4'   --enable-decoder='mpegvideo'   --enable-decoder='pcm_s16be'   --enable-decoder='pcm_s16le'   --enable-decoder='png'   --enable-decoder='vorbis'   --enable-decoder='wmav1'   --enable-decoder='wmav2'   --enable-decoder='zlib'  --disable-muxers                                  --disable-demuxers  --enable-demuxer='ac3'      --enable-demuxer='h264'   --enable-demuxer='mp3'         --enable-demuxer='mpegvideo'   --enable-demuxer='ogg'              --disable-parsers  --enable-parser='aac'   --enable-parser='ac3'   --enable-parser='h264'   --enable-parser='mpegaudio'   --enable-parser='mpegvideo'   --enable-parser='mpeg4video'  --disable-protocols                   --disable-asm --disable-altivec --disable-amd3dnow --disable-amd3dnowext --disable-mmx --disable-mmxext --disable-sse --disable-sse2 --disable-sse3 --disable-ssse3 --disable-sse4 --disable-sse42 --disable-avx --disable-xop --disable-fma3 --disable-fma4 --disable-avx2 --disable-inline-asm --disable-yasm
arm-openwrt-linux-muslgnueabi-gcc is unable to create an executable file.
C compiler test failed.

If you think configure made a mistake, make sure you are using the latest
version from Git.  If the latest version fails, report the problem to the
ffmpeg-user@ffmpeg.org mailing list or IRC #ffmpeg on irc.freenode.net.
Include the log file "config.log" produced by configure as this will help
solve the problem.
config.status: error: Failed to configure embedded Libav tree
Makefile:198: recipe for target '/home/ppark/Work/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl-1.1.16_eabi/gst-libav-1.8.2/.configured_nnnnnnnnnnyyynyynyyyyyyyyyyyynnnnnnnnnnnnynyynnyynnnnyyyyyynnnnnnyyyyyyyyn' failed

hnyman, 2017-09-16:
> When I compile in openWrt, I have a compile error like this.

In general, test compiling with an up-to-date LEDE and forget about trying to build with stale Openwrt trunk. (Openwrt trunk has been half-dead since May 2016 and will be overwritten with LEDE sources when the planned merge happens...)

jhpark555, 2017-09-21:
After I modified the makefile for gst-libav, the problem solved.

CONFIGURE_ARGS += \
	--without-system-libav \
	--with-libav-extra-configure="--target-os=linux \
	$(LIBAV_CONFIGURE_ENCODERS) \
	--disable-decoders \
	$(LIBAV_CONFIGURE_DECODERS) \
	--disable-muxers \
	$(LIBAV_CONFIGURE_MUXERS) \
	--disable-demuxers \
	$(LIBAV_CONFIGURE_DEMUXERS) \
	--disable-parsers \
	$(LIBAV_CONFIGURE_PARSERS) \
	--disable-protocols \
	$(LIBAV_CONFIGURE_PROTOCOLS) "

thess, 2017-10-29:
I'm actually thinking of dropping this package due to the age of libav being used. Do you really want this version?

thess, 2018-07-24:
Updated version should be OK. Re-open if not.

closed 2018-07-24 by thess