openwrt/packages #4840: apache proxy segfault

lucize, 2017-09-16:
Maintainer @heil 
with --enable-proxy on sun7i target (arm) I get this error and it will loop forewer

```
[Sat Sep 16 13:52:16.047623 2017] [proxy_hcheck:info] [pid 34359738384:tid 3064544580] (70023)This function has not been implemented on this platform: AH03312: apr_thread_pool_create() with 16 threads failed
[Sat Sep 16 13:52:16.047849 2017] [watchdog:debug] [pid -5264627648507674620:tid 3069363352] mod_watchdog.c(585): AH02981: Watchdog: Created worker thread (_proxy_hcheck_).
[Sat Sep 16 13:52:16.048519 2017] [proxy:debug] [pid -5264506015033851904:tid 3069363352] proxy_util.c(1777): AH00925: initializing worker proxy:reverse shared
[Sat Sep 16 13:52:16.048586 2017] [proxy:debug] [pid -5264506015033851904:tid 3069363352] proxy_util.c(1819): AH00927: initializing worker proxy:reverse local
[Sat Sep 16 13:52:17.043593 2017] [core:notice] [pid -4715750481822642040:tid 3069363352] AH00052: child pid 10645 exit signal Segmentation fault (11)
[Sat Sep 16 13:52:17.044001 2017] [core:notice] [pid -4715750481822642040:tid 3069363352] AH00052: child pid 10646 exit signal Segmentation fault (11)
[Sat Sep 16 13:52:17.044068 2017] [core:notice] [pid -4715750481822642040:tid 3069363352] AH00052: child pid 10647 exit signal Segmentation fault (11)
[Sat Sep 16 13:52:17.044124 2017] [core:notice] [pid -4715750481822642040:tid 3069363352] AH00052: child pid 10648 exit signal Segmentation fault (11)
```

diizzyy, 2017-09-16:
I wonder if this is specific to LEDE even though the error message doesn't imply that.
All musl based distros I've looked at doesn't carry any specific patches for Apache 2.4.

https://git.archlinux.org/svntogit/packages.git/tree/trunk?h=packages/apache
https://git.alpinelinux.org/cgit/aports/tree/main/apache2?h=3.6-stable
https://github.com/buildroot/buildroot/tree/master/package/apache

lucize, 2017-09-29:
found [this](https://bugzilla.redhat.com/show_bug.cgi?id=1410883) related to proxy_hcheck, I'll try to disable that module to see the outcome and maybe a bug report to apache

lucize, 2017-09-29:
with                 --disable-proxy-hcheck \

```
[Fri Sep 29 20:38:20.042354 2017] [watchdog:debug] [pid -5261614797669007356:tid 3070067912] mod_watchdog.c(562): AH02980: Watchdog: nothing configured?
[Fri Sep 29 20:38:20.043162 2017] [proxy:debug] [pid -5261489762581086208:tid 3070067912] proxy_util.c(1777): AH00925: initializing worker proxy:reverse shared
[Fri Sep 29 20:38:20.043226 2017] [proxy:debug] [pid -5261489762581086208:tid 3070067912] proxy_util.c(1819): AH00927: initializing worker proxy:reverse local
[Fri Sep 29 20:38:20.043621 2017] [watchdog:debug] [pid -5261614797669007356:tid 3070067912] mod_watchdog.c(562): AH02980: Watchdog: nothing configured?
[Fri Sep 29 20:38:20.044315 2017] [proxy:debug] [pid -5261489762581086208:tid 3070067912] proxy_util.c(1777): AH00925: initializing worker proxy:reverse shared
[Fri Sep 29 20:38:20.044376 2017] [proxy:debug] [pid -5261489762581086208:tid 3070067912] proxy_util.c(1819): AH00927: initializing worker proxy:reverse local
[Fri Sep 29 20:38:20.046226 2017] [watchdog:debug] [pid -5261614797669007356:tid 3070067912] mod_watchdog.c(562): AH02980: Watchdog: nothing configured?
[Fri Sep 29 20:38:20.047078 2017] [proxy:debug] [pid -5261489762581086208:tid 3070067912] proxy_util.c(1777): AH00925: initializing worker proxy:reverse shared
[Fri Sep 29 20:38:20.047141 2017] [proxy:debug] [pid -5261489762581086208:tid 3070067912] proxy_util.c(1819): AH00927: initializing worker proxy:reverse local

```

lucize, 2017-09-29:
maybe the problem is elsewhere 
trying the latest release without proxy-hcheck 2.4.20 

```
[Fri Sep 29 21:47:49.936745 2017] [unique_id:info] [pid 0:tid 3069449416] AH01566: using ip addr 127.0.0.1
[Fri Sep 29 21:47:50.000388 2017] [ssl:info] [pid 4211540143546721650:tid 3069449416] AH01883: Init: Initialized OpenSSL library
[Fri Sep 29 21:47:50.001129 2017] [ssl:info] [pid 4211540143546721650:tid 3069449416] AH01887: Init: Initializing (virtual) servers for SSL
[Fri Sep 29 21:47:50.001206 2017] [ssl:info] [pid 4211540143546721650:tid 3069449416] AH01876: mod_ssl/2.4.20 compiled against Server: Apache/2.4.20, Library: OpenSSL/1.0.2l
AH00558: httpd: Could not reliably determine the server's fully qualified domain name, using 127.0.0.1. Set the 'ServerName' directive globally to suppress this message
[Fri Sep 29 21:47:50.125143 2017] [auth_digest:notice] [pid 124869920658817025:tid 3069449416] AH01757: generating secret for digest authentication ...
[Fri Sep 29 21:47:50.125266 2017] [auth_digest:debug] [pid 124869920658817025:tid 3069449416] mod_auth_digest.c(250): AH01759: done
[Fri Sep 29 21:47:50.129191 2017] [unique_id:info] [pid 6471320296826010328:tid 3069449416] AH01566: using ip addr 127.0.0.1
[Fri Sep 29 21:47:51.001073 2017] [ssl:info] [pid 3539955082619592816:tid 3069449416] AH01883: Init: Initialized OpenSSL library
[Fri Sep 29 21:47:51.001434 2017] [ssl:warn] [pid 683191857840128:tid 3069449416] AH01873: Init: Session Cache is not configured [hint: SSLSessionCache]
[Fri Sep 29 21:47:51.001467 2017] [ssl:info] [pid 335008788922:tid 3069449416] AH01887: Init: Initializing (virtual) servers for SSL
[Fri Sep 29 21:47:51.001509 2017] [ssl:info] [pid 335008788922:tid 3069449416] AH01876: mod_ssl/2.4.20 compiled against Server: Apache/2.4.20, Library: OpenSSL/1.0.2l
[Fri Sep 29 21:47:51.001695 2017] [slotmem_shm:debug] [pid -4707410136396726272:tid 3069449416] mod_slotmem_shm.c(496): AH02301: attach looking for /var/log/slotmem-shm-mod_heartmonitor.shm
[Fri Sep 29 21:47:51.001806 2017] [lbmethod_heartbeat:notice] [pid 5093659415550104:tid 3069449416] AH02282: No slotmem from mod_heartmonitor
[Fri Sep 29 21:47:51.006183 2017] [mpm_worker:notice] [pid -434114893185024000:tid 3069449416] AH00292: Apache/2.4.20 (Unix) OpenSSL/1.0.2l configured -- resuming normal operations
[Fri Sep 29 21:47:51.006331 2017] [mpm_worker:info] [pid -434114893185024000:tid 3069449416] AH00293: Server built: Sep 29 2017 21:46:21
[Fri Sep 29 21:47:51.006415 2017] [core:notice] [pid -5281897969673629144:tid 3069449416] AH00094: Command line: 'httpd -D FOREGROUND -f /etc/apache/httpd.conf -e debug'
[Fri Sep 29 21:47:51.006459 2017] [core:debug] [pid 0:tid 3069449416] log.c(1546): AH02639: Using SO_REUSEPORT: yes (1)
[Fri Sep 29 21:47:51.006486 2017] [mpm_worker:debug] [pid 7:tid 3069449416] worker.c(1874): AH00294: Accept mutex: none (default: sysvsem)
[Fri Sep 29 21:47:51.008058 2017] [proxy:debug] [pid 1483740:tid 3069449416] proxy_util.c(1790): AH00925: initializing worker proxy:reverse shared
[Fri Sep 29 21:47:51.008313 2017] [proxy:debug] [pid 1483740:tid 3069449416] proxy_util.c(1832): AH00927: initializing worker proxy:reverse local
[Fri Sep 29 21:47:51.008466 2017] [proxy:debug] [pid 1483740:tid 3069449416] proxy_util.c(1790): AH00925: initializing worker proxy:reverse shared
[Fri Sep 29 21:47:51.008664 2017] [proxy:debug] [pid 1483740:tid 3069449416] proxy_util.c(1832): AH00927: initializing worker proxy:reverse local
[Fri Sep 29 21:47:51.010888 2017] [proxy:debug] [pid 1483740:tid 3069449416] proxy_util.c(1790): AH00925: initializing worker proxy:reverse shared
[Fri Sep 29 21:47:51.011176 2017] [proxy:debug] [pid 1483740:tid 3069449416] proxy_util.c(1832): AH00927: initializing worker proxy:reverse local
[Fri Sep 29 21:47:52.007755 2017] [core:notice] [pid -4707411991793524600:tid 3069449416] AH00052: child pid 8644 exit signal Segmentation fault (11)
[Fri Sep 29 21:47:52.008147 2017] [core:notice] [pid -4707411991793524600:tid 3069449416] AH00052: child pid 8645 exit signal Segmentation fault (11)
[Fri Sep 29 21:47:52.008214 2017] [core:notice] [pid -4707411991793524600:tid 3069449416] AH00052: child pid 8646 exit signal Segmentation fault (11)
[Fri Sep 29 21:47:53.011931 2017] [proxy:debug] [pid -4707412541578411985:tid 3069449416] proxy_util.c(1790): AH00925: initializing worker proxy:reverse shared
[Fri Sep 29 21:47:53.012371 2017] [proxy:debug] [pid -4707412541578411985:tid 3069449416] proxy_util.c(1832): AH00927: initializing worker proxy:reverse local

[Fri Sep 29 21:47:54.012685 2017] [core:notice] [pid -4707411991793524600:tid 3069449416] AH00052: child pid 8647 exit signal Segmentation fault (11)

```

diizzyy, 2017-09-29:
Have you tried another worker?
https://git.alpinelinux.org/cgit/aports/tree/main/apache2/APKBUILD?h=3.6-stable#n96

lucize, 2017-09-29:
I'll try without worker
on 2.2.34 it's the same, maybe glibc is the lucky one
```
[Fri Sep 29 22:24:52 2017] [notice] Apache/2.2.34 (Unix) mod_ssl/2.2.34 OpenSSL/1.0.2l configured -- resuming normal operations
[Fri Sep 29 22:24:52 2017] [info] Server built: Sep 29 2017 22:07:46
[Fri Sep 29 22:24:52 2017] [debug] worker.c(1774): AcceptMutex: sysvsem (default: sysvsem)
[Fri Sep 29 22:24:52 2017] [debug] proxy_util.c(1916): proxy: grabbed scoreboard slot 0 in child 139911858202765409 for worker \xd8`\xef\x010K\xf3\x01
[Fri Sep 29 22:24:52 2017] [debug] proxy_util.c(1916): proxy: grabbed scoreboard slot 0 in child 139911858202765409 for worker \xd8`\xef\x010K\xf3\x01
[Fri Sep 29 22:24:52 2017] [debug] proxy_util.c(1936): proxy: worker proxy:reverse already initialized
[Fri Sep 29 22:24:52 2017] [debug] proxy_util.c(1916): proxy: grabbed scoreboard slot 0 in child 139911858202765409 for worker \xd8`\xef\x010K\xf3\x01
[Fri Sep 29 22:24:52 2017] [debug] proxy_util.c(1936): proxy: worker proxy:reverse already initialized
[Fri Sep 29 22:24:53 2017] [notice] child pid 10835 exit signal Segmentation fault (11)
[Fri Sep 29 22:24:53 2017] [notice] child pid 10836 exit signal Segmentation fault (11)
[Fri Sep 29 22:24:53 2017] [notice] child pid 10837 exit signal Segmentation fault (11)

```

lucize, 2017-09-30:
glibc is not working either, anyhow disabling proxy leads to 
`protocol.c HTTP Request Line; Unrecognized protocol 'HTTP/0.9' (perhaps whitespace was injected?)`
the last version that is working without proxy is 2.4.23

diizzyy, 2017-12-12:
@lucize 
Still broken?

lucize, 2017-12-12:
I'll try with a pi board, I use the sun7i board on other project, I gave up on apache implementation

lucize, 2017-12-13:
ok, built the today snapshot on  PI 1 B

```
[Wed Dec 13 17:59:09.125840 2017] [watchdog:debug] [pid -5262291718874595328:tid 3069744360] mod_watchdog.c(587): AH02981: Watchdog: Created worker thread (_proxy_hcheck_).
[Wed Dec 13 17:59:09.126572 2017] [proxy:debug] [pid -5262876879200876028:tid 3069744360] proxy_util.c(1777): AH00925: initializing worker proxy:reverse shared
[Wed Dec 13 17:59:09.126696 2017] [proxy:debug] [pid -5262876879200876028:tid 3069744360] proxy_util.c(1819): AH00927: initializing worker proxy:reverse local
[Wed Dec 13 17:59:09.127338 2017] [watchdog:debug] [pid -5262291718874595328:tid 3069744360] mod_watchdog.c(587): AH02981: Watchdog: Created worker thread (_proxy_hcheck_).
[Wed Dec 13 17:59:09.128217 2017] [proxy:debug] [pid -5262876879200876028:tid 3069744360] proxy_util.c(1777): AH00925: initializing worker proxy:reverse shared
[Wed Dec 13 17:59:09.128329 2017] [proxy:debug] [pid -5262876879200876028:tid 3069744360] proxy_util.c(1819): AH00927: initializing worker proxy:reverse local
[Wed Dec 13 17:59:09.128719 2017] [watchdog:debug] [pid -5262291718874595328:tid 3069744360] mod_watchdog.c(587): AH02981: Watchdog: Created worker thread (_proxy_hcheck_).
[Wed Dec 13 17:59:09.129709 2017] [proxy:debug] [pid -5262876879200876028:tid 3069744360] proxy_util.c(1777): AH00925: initializing worker proxy:reverse shared
[Wed Dec 13 17:59:09.129823 2017] [proxy:debug] [pid -5262876879200876028:tid 3069744360] proxy_util.c(1819): AH00927: initializing worker proxy:reverse local
[Wed Dec 13 17:59:09.131139 2017] [watchdog:debug] [pid 0:tid 3065457988] mod_watchdog.c(156): AH02972: Singleton Watchdog (_proxy_hcheck_) running
[Wed Dec 13 17:59:09.131754 2017] [proxy_hcheck:debug] [pid 0:tid 3065457988] mod_proxy_hcheck.c(881): AH03258: _proxy_hcheck_ watchdog started.
[Wed Dec 13 17:59:09.131906 2017] [proxy_hcheck:info] [pid 0:tid 3065457988] (70023)This function has not been implemented on this platform: AH03312: apr_thread_pool_create() with 16 threads failed

```

0anton, 2018-01-19:
apache - 2.4.25-1 package is still broken in LEDE / Reboot (17.01.4, r3560-79f57e422d)

> [Fri Jan 19 22:45:11.002832 2018] [mpm_worker:notice] [pid -5280269970937801572:tid 3070039196] AH00292: Apache/2.4.25 (Unix) OpenSSL/1.0.2m configured -- resuming normal operations
> [Fri Jan 19 22:45:11.002897 2018] [mpm_worker:info] [pid -5280269970937801572:tid 3070039196] AH00293: Server built: Dec 18 2017 10:33:39
> [Fri Jan 19 22:45:11.002948 2018] [core:notice] [pid -5279226517213754864:tid 3070039196] AH00094: Command line: '/usr/sbin/httpd'
> [Fri Jan 19 22:45:11.004396 2018] [proxy_hcheck:info] [pid 0:tid 3065179460] (70023)This function has not been implemented on this platform: AH03312: apr_thread_pool_create() with 16 threads failed
> 

Any page request is answered with "Internal Server Error" 500 and in error_log:

> [Fri Jan 19 22:49:43.138222 2018] [http:error] [pid 0:tid 3064548676] [client 192.168.33.182:58550] AH02429: Response header name 'Content-Length' contains invalid characters, aborting request

0anton, 2018-01-21:
Tried recompiling apache from recent sources (https://github.com/openwrt/openwrt.git). Same problem persists. Even pid is determined by apache incorrectly. Faulty libc?

```
Package: apache
Version: 2.4.28-2

```

```
[Sun Jan 21 22:08:27.016818 2018] [unique_id:info] [pid 0:tid 3069826204] AH01566: using ip addr 192.168.33.1
[Sun Jan 21 22:08:28.001165 2018] [ssl:info] [pid 3329060823161516334:tid 3069826204] AH01887: Init: Initializing (virtual) servers for SSL
[Sun Jan 21 22:08:28.001267 2018] [ssl:info] [pid 3329060823161516334:tid 3069826204] AH01876: mod_ssl/2.4.28 compiled against Server: Apache/2.4.28, Library: OpenSSL/1.0.2n
[Sun Jan 21 22:08:28.012010 2018] [unique_id:info] [pid 1509529588463992:tid 3069826204] AH01566: using ip addr 192.168.33.1[Sun Jan 21 22:08:29.001185 2018] [ssl:warn] [pid 7235419165714114926:tid 3069826204] AH01873: Init: Session Cache is not configured [hint: SSLSessionCache]
[Sun Jan 21 22:08:29.001282 2018] [ssl:info] [pid 326418888909:tid 3069826204] AH01887: Init: Initializing (virtual) servers for SSL
[Sun Jan 21 22:08:29.001304 2018] [ssl:info] [pid 326418888909:tid 3069826204] AH01876: mod_ssl/2.4.28 compiled against Server: Apache/2.4.28, Library: OpenSSL/1.0.2n
[Sun Jan 21 22:08:29.001461 2018] [lbmethod_heartbeat:notice] [pid 1212419:tid 3069826204] AH02282: No slotmem from mod_heartmonitor
[Sun Jan 21 22:08:29.002724 2018] [mpm_worker:notice] [pid -5279337585077439536:tid 3069826204] AH00292: Apache/2.4.28 (Unix) OpenSSL/1.0.2n configured -- resuming normal operations
[Sun Jan 21 22:08:29.002793 2018] [mpm_worker:info] [pid -5279337585077439536:tid 3069826204] AH00293: Server built:
[Sun Jan 21 22:08:29.002837 2018] [core:notice] [pid -5278272639347982336:tid 3069826204] AH00094: Command line: '/usr/sbin/httpd'
[Sun Jan 21 22:08:29.003540 2018] [proxy_hcheck:info] [pid 0:tid 3065494852] (70023)This function has not been implemented on this platform: AH03312: apr_thread_pool_create() with 16 threads failed

```

Any GET request to any page end up with error code 500:

```
Internal Server Error
The server encountered an internal error or misconfiguration and was unable to complete your request.

Please contact the server administrator at you@example.com to inform them of the time this error occurred, and the actions you performed just before this error.

More information about this error may be available in the server error log.

Additionally, a 400 Bad Request error was encountered while trying to use an ErrorDocument to handle the request.
```

Any help on this?

tru7, 2018-08-09:
Installed OpenWrt 18.06.0, r7188-b0b5c64c22 on mpc85xx.
Installed apache with default configuration except for port to avoid conflicts. But it does not start successfully, reports many warnings and throws segmentation faults. 
It even reports ssl:warn even ssl is not activated.
Same for starting with httpd -X

diizzyy, 2018-08-09:
Hmm... Can someone try setting O2 instead of Os when compiling Apache to avoid potential miscompilation?

tru7, 2018-08-09:
just checked with snapshot: same problems with out-of-the-box installation, only setting ServerName

diizzyy, 2018-08-15:
Can you give it a go with O2 instead of Os is it's known to cause miscompilation in some cases?

tru7, 2018-08-15:
@diizzyy do you mean me? I was only talking about packages from 18.06 and snapshots downloads. But I could give it a try building myself, however I need instruction because I don't see Os in the Makefile.

tru7, 2018-08-18:
@diizzyy @0anton @heil @lucize : I did some extensive testing on this one with all the latest (snapshot toolchain, packages master) and found some weired results:
- basically I can confirm all the results from this thread
- no difference when I change compiler options from -Os to -O2 or -O0 or no -O at all
- as soon as --enable-proxy is used the daemon throws segementation fault
- with --disable-proxy it's a non-working server because all requests return 500 with Unrecognized protocol 'HTTP/0.9' in the log.
- httpd.pid is useless because it stores for example 45151774993032 instead of 10512

Then I did the same tests with httpd-2.4.20, leading to the very same results with one exception:
at least the web server works and is returning 'HTTP/1.1 200 OK'

I cannot compile 2.4.34, I guess OpenWrt needs patching for this. But that's beyond my horizon.

I tried the packages from 18.06 and snapshot on three different platforms but none of them work. Looks like there is no working apache package available. That's a little nasty.

Can I do anything else on this?

bilicio, 2018-09-10:
If you change the variable LogLevel  (httpd.conf) to something like "error" you can bypass the Segmentation fault, but it cames with 500 Internal Server Error. Changing the error page to any other page will help show something on the screen. Its just if you are desperate to show a page on this. its only what i could find. :(

jpereira, 2018-11-02:
+1 still failing with the 18.06.1.

lucize, 2019-01-08:
maybe is better to mark this as broken ?

lucize, 2019-01-12:
tested a bit more with loadable modules and I still have the `AH02429: Response header name 'Content-Length' contains invalid characters, aborting request` maybe there is a chance for this

diizzyy, 2019-02-09:
@lucize 
Can you try the latest version with OpenWrt master and the newest version of Apache, also please provide a backtrace using gdb

lucize, 2019-02-09:
I'll test, but for sure apr needs to be modified also to allow dynamic modules, namely for 	--disable-dso \ to be removed

minfrin, 2019-12-27:
Without a backtrace, we can't fix this.

Please follow the guide at https://httpd.apache.org/dev/debugging.html#backtrace if possible, and provide a backtrace of the httpd process after the process has crashed.

Once we know where the crash is occurring, we can isolate the problem and fix it.

PR #10898 (2020-01-12): apache: bump version plus brush up
Maintainer: @heil 
Compile tested: master ath79, 19.07-rc2 ath79+arc770
Run tested: 19.07-rc2 ath90, it starts fine and I get to an error page when visiting the server with my browser

Description:
Hi all,

I guess I got a bit bored over the holidays, honestly.

I'm hoping that this

Fixes #4840 
Fixes #3720

Well, the modules are now installed. I also incorporated a layout file like @minfrin talked about in #10883 (I used his file in the beginning but wasn't totally satisfied, so I made some amendments).

Regarding the segfault when using proxy mode. Well, I [... cut]
closed 2020-01-12 by hnyman