openwrt/packages #4867: luci-app-fwknopd: always regen keys after sysupgrade

DazzyWalkman, 2017-09-22:
Background:
I am using self-compile x86 lede image. No modification to fwknop related sources. 
I began to include luci-app-fwknopd and all its depend packages in my image since lede trunk r4800 or so.
I always keep luci-app-fwknopd and fwknopd in sync with "./scripts/feeds update -a" when compile.

Problem:
Since the deployment of fwknopd, it's working nicely, however, I began to notice fwknopd  keys  always got regenerated  after sysupgrade,  even /etc/conf/fwknopd was on the keep list. This rendered keys in my fwknop clients no longer working. 
Manually restoring the old fwknopd conf and restart fwknopd service would do just fine. and not much of an issue since I did most of upgrades on local console, though I locked myself out of my lan occasionally if I forgot to do so. 

I look into the related sources, and suspect this behavior is defined in /luci/applications/luci-app-fwknopd/root/etc/uci-defaults/40_luci-fwknopd
Line 16 and 17:
uci set fwknopd.@access[0].KEY_BASE64=`fwknopd --key-gen | awk '/^KEY/ {print $2;}'`
uci set fwknopd.@access[0].HMAC_KEY_BASE64=`fwknopd --key-gen | awk '/^HMAC/ {print $2;}'`
It seems like unconditional key regen.
I am not familiar with lede sysupgrade mechanism, so not sure if the above file plays part in this problem.

jp-bennett, 2017-10-14:
@DazzyWalkman Just now saw this bug report, sorry for the delay.  Looking into it now.

jp-bennett, 2017-10-14:
Found the correct fix, the pull request is https://github.com/openwrt/luci/pull/1392

jp-bennett, 2017-10-16:
@DazzyWalkman This should be fixed as the linked PR was accepted.  Could you test a new trunk build when you get a chance and confirm it's fixed?

DazzyWalkman, 2017-10-18:
@jp-bennett
 Confirm fixed.
Tested on lede trunk x86 with budled packages fwknopd - 2.6.9-4 and luci-app-fwknopd - git-17.290.79393-b7be0b9-1
upgrade from r5032 to r5103
with command: `sysupgrade lede-snapshot-r5103-f6c4a9c-x86-generic-combined-ext4.img.gz`

closed 2017-10-18 by DazzyWalkman