openwrt/packages #4878: ddns-scripts: should provide support for POSTed data

dkg, 2017-09-25:
ddns-scripts basically just sends HTTP and HTTPS `GET` requests that include potentially sensitive parameters like passwords.

But `GET` URLs have a tendency to be logged on server logs, stored in browser histories, etc, so a privacy- and security-preserving dynamic DNS service might prefer to use HTTP/S `POST` requests instead, and to send the sensitive parameters as part of the `POST` data.

So i'm proposing a new field for ddns client scripts, named something like `postdata`.  if that field is set, then we send a `POST` instead of a `GET`, and it has the same substitutions available as `url`.

To implement: with `wget`, you can use `--post-data`.  with `curl` you can use `--data`, i think.

dkg, 2017-09-25:
@chris5560 i think this is for you

dibdot, 2018-08-17:
no feedback - close this by now.

closed 2018-08-17 by dibdot