openwrt/packages #4882: mwan3: ifdown / ifup on VLAN interface failing

ratsputin, 2017-09-29:
-------------------------------

Maintainer: @feckert 
Environment: LEDE Reboot SNAPSHOT r4883-ce53c0e
mwan3 version: 2.6.6-1

Description:  Failover configuration using wan_wan2 not failing-over when wan goes down.  Possibly due to wan and wan2 links being associated with VLAN interfaces.

Attempted to reproduce with the following:
```
root@orion:~# /usr/sbin/mwan3 ifdown eth1.99
/sbin/hotplug-call: eval: line 1: syntax error: bad substitution
/sbin/hotplug-call: eval: line 1: syntax error: bad substitution
root@orion:~#
```
mwan3 config:
```
config rule 'default_rule'
        option dest_ip '0.0.0.0/0'
        option proto 'all'
        option sticky '0'
        option use_policy 'wan_wan2'

config policy 'wan_wan2'
        list use_member 'wan_m1_w3'
        list use_member 'wan2_m2_w2'

config policy 'wan_only'
        list use_member 'wan_m1_w3'

config policy 'wan2_only'
        list use_member 'wan2_m1_w2'

config policy 'balanced'
        list use_member 'wan_m1_w3'
        list use_member 'wan2_m1_w2'

config policy 'wan2_wan'
        list use_member 'wan_m2_w3'
        list use_member 'wan2_m1_w2'

config interface 'wan'
        option enabled '1'
        option reliability '2'
        option count '1'
        option timeout '2'
        option interval '5'
        option down '3'
        option up '8'
        option initial_state 'online'
        option family 'ipv4'
        list track_ip '8.8.4.4'
        list track_ip '8.8.8.8'
        option track_method 'ping'
        option size '56'
        option failure_interval '5'
        option recovery_interval '5'
        option flush_conntrack 'never'

config interface 'wan2'
        option count '1'
        option timeout '2'
        option interval '5'
        option down '3'
        option up '8'
        option initial_state 'online'
        option family 'ipv4'
        list track_ip '8.8.8.8'
        list track_ip '8.8.4.4'
        option track_method 'ping'
        option size '56'
        option failure_interval '5'
        option recovery_interval '5'
        option flush_conntrack 'never'
        option enabled '1'
        option reliability '2'

config member 'wan_m1_w3'
        option interface 'wan'
        option metric '1'
        option weight '3'

config member 'wan_m2_w3'
        option interface 'wan'
        option metric '2'
        option weight '3'

config member 'wan2_m1_w2'
        option interface 'wan2'
        option metric '1'
        option weight '2'

config member 'wan2_m2_w2'
        option interface 'wan2'
        option metric '2'
        option weight '2'
```

Relevant portions of network config:
```
config interface 'wan'
        option proto 'dhcp'
        option ifname 'eth1.99'
        option metric '10'
        option peerdns '0'
        option dns '127.0.0.1'
        option broadcast '1'

config interface 'wan2'
        option proto 'dhcp'
        option metric '20'
        option ifname 'eth1.2'
        option peerdns '0'
        option dns '127.0.0.1'
```
I was able to narrow the issue down to the following two instances:
```
+ config_get enabled eth1.99 enabled 0
+ eval export -n -- enabled=${CONFIG_eth1.99_enabled:-${4}}
/sbin/hotplug-call: eval: line 1: syntax error: bad substitution
+ [ -f /etc/hotplug.d/iface/16-mwan3-user ]
+ . /etc/hotplug.d/iface/16-mwan3-user
```
and
```
+ config_get enabled eth1.99 enabled 0
+ eval export -n -- enabled=${CONFIG_eth1.99_enabled:-${4}}
/sbin/hotplug-call: eval: line 1: syntax error: bad substitution
+ [ -f /etc/hotplug.d/iface/16-mwancustombak ]
+ . /etc/hotplug.d/iface/16-mwancustombak
```

feckert, 2017-10-01:
@ratsputin thanks for feedback but this lines

```
root@orion:~# /usr/sbin/mwan3 ifdown eth1.99
/sbin/hotplug-call: eval: line 1: syntax error: bad substitution
/sbin/hotplug-call: eval: line 1: syntax error: bad substitution
root@orion:~#
```
are strange do you know the issue could you explain it maybe there is a bug from my side?

ratsputin, 2017-10-01:
@feckert So, my issue is that with the failure of my primary WAN interface ("wan", in this case eth1.99), the router didn't fail over to wan2 (eth1.2).  This did work with the June release, but I noticed there have been a number of changes to mwan3 since then.  To troubleshoot this further, I attempted to use the "ifup/ifdown" functionality of /usr/sbin/mwan3 to attempt to reproduce the situation manually.  If you check the earlier notes on this ticket, you'll see the details on the errors I received as well as my configuration.

I'm unclear on exactly how the command-line interface using ifdown/ifup is supposed to work so I can troubleshoot this.  If you look at the earlier notes, using ifdown with the physical interface (eth1.2 or eth1.99) generates the error referenced above.  If, however, I try using the network name:

```
/usr/sbin/ifdown wan2
```
I do end up with the following status:
```
Interface status:
 interface wan is online and tracking is active
 interface wan2 is offline and tracking is down

Current ipv4 policies:
balanced:
 wan (100%)

wan2_only:
 unreachable

wan2_wan:
 wan (100%)

wan_only:
 wan (100%)

wan_wan2:
 wan (100%)
```
However, LuCI still shows both interfaces up in the UI, and routing shows the following (note, below shows before and after routing):
```
root@orion:~# netstat -rn
Kernel IP routing table
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
0.0.0.0         98.196.52.1     0.0.0.0         UG        0 0          0 eth1.99
0.0.0.0         104.51.84.1     0.0.0.0         UG        0 0          0 eth1.2
10.1.0.0        0.0.0.0         255.255.255.0   U         0 0          0 eth0.10
98.196.52.0     0.0.0.0         255.255.252.0   U         0 0          0 eth1.99
98.196.52.1     0.0.0.0         255.255.255.255 UH        0 0          0 eth1.99
104.51.84.0     0.0.0.0         255.255.252.0   U         0 0          0 eth1.2
104.51.84.1     0.0.0.0         255.255.255.255 UH        0 0          0 eth1.2
192.168.1.0     0.0.0.0         255.255.255.0   U         0 0          0 br-lan
192.168.3.0     0.0.0.0         255.255.255.0   U         0 0          0 eth0.3
root@orion:~# /usr/sbin/mwan3 ifdown wan2
root@orion:~# netstat -rn
Kernel IP routing table
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
0.0.0.0         98.196.52.1     0.0.0.0         UG        0 0          0 eth1.99
0.0.0.0         104.51.84.1     0.0.0.0         UG        0 0          0 eth1.2
10.1.0.0        0.0.0.0         255.255.255.0   U         0 0          0 eth0.10
98.196.52.0     0.0.0.0         255.255.252.0   U         0 0          0 eth1.99
98.196.52.1     0.0.0.0         255.255.255.255 UH        0 0          0 eth1.99
104.51.84.0     0.0.0.0         255.255.252.0   U         0 0          0 eth1.2
104.51.84.1     0.0.0.0         255.255.255.255 UH        0 0          0 eth1.2
192.168.1.0     0.0.0.0         255.255.255.0   U         0 0          0 br-lan
192.168.3.0     0.0.0.0         255.255.255.0   U         0 0          0 eth0.3
root@orion:~#
```
Notice that the route associated with wan2 (eth1.2) didn't go away?  This is precisely the same problem I encountered when "wan" went away; the default gateway associated with eth1.99 (my primary WAN link) wasn't deleted.

I should note that ping tracking is working fine; it did actually see that the "wan" interface went away and it appears to have attempted to do failover.  The problem is, the routes associated with the "down" interface didn't go away and LuCI seemed to think the interface was still up.

TL;DR - mwan3 failover didn't work when my primary link went down (mwan3 detected it and initiated failover, but didn't delete the routes associated with the downed interface); I'm trying to troubleshoot it, but unsure how using /usr/sbin/mwan3 ifdown _interface_, as the documentation doesn't specify whether _interface_ is supposed to be the WAN alias or the actual physical interface.

Your attention and assistance is sincerely appreciated.

Brett

ratsputin, 2017-10-01:
As a footnote, I did notice the following when I was putting my configuration back after updating the ticket:

```
root@orion:~# /usr/sbin/mwan3 ifup wan2
uci: Entry not found
uci: Entry not found
root@orion:~# /usr/sbin/mwan3 status
Interface status:
 interface wan is online and tracking is active
 interface wan2 is online and tracking is active
```
So, there was a complaint about the command from uci; however, it did bring the interface back online.

I'm happy to dig through what exactly is going on, but I'll need a bit of guidance.

feckert, 2017-10-02:
https://github.com/openwrt/packages/pull/4811 since this PR the route in the main routing table will not get delete anymore. The interface is marked as offline and only the user rules are deleted.

To test if the fail-over scenario works you could block the ping for the interface with the firewall.
start-test:
iptables -I INPUT -i "${iface}" -p icmp  -j DROP
stop-test:
iptables -D INPUT -i "${iface}" -p icmp -j DROP

The variable ${iface} should be substituted with eth1.99 or eth1.2

ezplanet, 2017-10-02:
Reporting a very similar issue, if not the same with BT Home Hub 5. 
WAN is internal ADSL, DMZ is the red Ethernet connected to another router with Internet access.

 The device is BT Home Hub 5
The build is: LEDE Reboot SNAPSHOT r4883-ce53c0e (problem is present also in previous releases) mwan3 module built in.

The configuration is as follows:

dsl pppoa-wan configured as WAN interface, default route metric 2
eth0.2 configured as DMZ interface (red ethernet port), default route metric 1

When WAN interface is down, luci shows the interface as offline, RED colour

# mwan3 status
Interface status:
interface wan is unknown and tracking is active
interface dmz is online and tracking is active

Current ipv4 policies:
balanced:
dmz (100%)

dmz_only:
dmz (100%)

dmz_wan:
dmz (100%)

wan_dmz:
dmz (100%)

wan_only:
unreachable

Current ipv6 policies:
balanced:
unreachable

dmz_only:
unreachable

dmz_wan:
unreachable

wan_dmz:
unreachable

wan_only:
unreachable

All is OK above when WAN interface is down, it is detected, routing goes through DMZ only

When DMZ interface is down instead, luci shows the interface as online tracking, GREEN
however mwan3 status reports:

Interface status:
interface wan is online and tracking is active
interface dmz is offline and tracking is active

Current ipv4 policies:
balanced:
wan (100%)

dmz_only:
unreachable

dmz_wan:
wan (100%)

wan_dmz:
wan (100%)

wan_only:
wan (100%)

Current ipv6 policies:
balanced:
unreachable

dmz_only:
unreachable

dmz_wan:
unreachable

wan_dmz:
unreachable

wan_only:
unreachable

Thus it looks like mwan3 status reports the interface as offline correctly, luci has it as online and routing is still attempted through it (thus failing).

ratsputin, 2017-10-03:
I see the same results @ezplanet mentions above.  I did the following:

```
iptables -I INPUT -i "eth1.2" -p icmp -j DROP
```
Once I had done that, I received the following in syslog:
```
Tue Oct  3 17:56:49 2017 user.notice mwan3track[18041]: Interface wan2 (eth1.2) is offline
Tue Oct  3 17:56:49 2017 user.notice mwan3[11632]: Execute ifdown event on interface wan2 (eth1.2)
Tue Oct  3 17:56:50 2017 user.info mwan3[11632]: connection tracking not flushed on interface wan2 (eth1.2) ifdown
Tue Oct  3 17:56:50 2017 user.info mwan3track[18041]: Detect ifdown event on interface wan2 (eth1.2)
```
As mentioned, routing wasn't updated:
```
root@orion:~# netstat -rn
Kernel IP routing table
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
0.0.0.0         98.196.52.1     0.0.0.0         UG        0 0          0 eth1.99
0.0.0.0         104.51.84.1     0.0.0.0         UG        0 0          0 eth1.2
10.1.0.0        0.0.0.0         255.255.255.0   U         0 0          0 eth0.10
98.196.52.0     0.0.0.0         255.255.252.0   U         0 0          0 eth1.99
98.196.52.1     0.0.0.0         255.255.255.255 UH        0 0          0 eth1.99
104.51.84.0     0.0.0.0         255.255.252.0   U         0 0          0 eth1.2
104.51.84.1     0.0.0.0         255.255.255.255 UH        0 0          0 eth1.2
192.168.1.0     0.0.0.0         255.255.255.0   U         0 0          0 br-lan
192.168.3.0     0.0.0.0         255.255.255.0   U         0 0          0 eth0.3
root@orion:~#
```
Also, mwan3 status is accurate:
```
root@orion:~# /usr/sbin/mwan3 status
Interface status:
 interface wan is online and tracking is active
 interface wan2 is offline and tracking is active

Current ipv4 policies:
balanced:
 wan (100%)

wan2_only:
 unreachable

wan2_wan:
 wan (100%)

wan_only:
 wan (100%)

wan_wan2:
 wan (100%)
```
One core issue is that LuCI's status still shows the following:

MWAN Interface Live Status
wan (eth1.99)
Online (tracking active)
wan2 (eth1.2)
Online (tracking active)

This is, of course, a minor, cosmetic issue, and I demonstrated it on the backup interface.  The larger issue is that mwan3 itself doesn't appear to be failing-over when the primary interface goes down.  In order to demonstrate this, I'll need to post this and then make a separate, independent post.

ratsputin, 2017-10-03:
So, my configuration is as follows:

eth1.99 - Comcast cable modem; when the cable goes down, the ethernet goes from being up on an actual Internet-routed network to being "up" on a local 192.168.100.x network on the modem itself to allow you to do diagnostics.  Obviously, pings to the Internet will fail.
eth1.2 - AT&T U-verse connection

For this test, I will disconnect the cable from the back of the cable modem, much as if cable has gone down.  Previously, this would cause mwan3 to fail over to the AT&T link and all traffic would be routed there until Comcast came back up.

Status prior to failure:
```
root@orion:~# /usr/sbin/mwan3 status
Interface status:
 interface wan is online and tracking is active
 interface wan2 is online and tracking is active

Current ipv4 policies:
balanced:
 wan2 (40%)
 wan (60%)

wan2_only:
 wan2 (100%)

wan2_wan:
 wan2 (100%)

wan_only:
 wan (100%)

wan_wan2:
 wan (100%)

Current ipv6 policies:
balanced:
 unreachable

wan2_only:
 unreachable

wan2_wan:
 unreachable

wan_only:
 unreachable

wan_wan2:
 unreachable

Directly connected ipv4 networks:
 127.0.0.1
 98.196.52.0
 192.168.1.255
 192.168.1.0
 104.51.87.255
 104.51.84.0/22
 104.51.84.0
 192.168.3.0
 104.51.85.38
 10.1.0.0/24
 98.196.55.178
 98.196.55.255
 192.168.1.1
 192.168.3.255
 192.168.3.1
 127.0.0.0/8
 192.168.3.0/24
 10.1.0.0
 127.0.0.0
 192.168.1.0/24
 104.51.84.1
 10.1.0.1
[... 256 lines cut ...]
104.51.84.1     0.0.0.0         255.255.255.255 UH        0 0          0 eth1.2
192.168.1.0     0.0.0.0         255.255.255.0   U         0 0          0 br-lan
192.168.3.0     0.0.0.0         255.255.255.0   U         0 0          0 eth0.3
root@orion:~#
```
And, of course, I have connectivity:
```
root@orion:~# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=0 ttl=56 time=23.183 ms
64 bytes from 8.8.8.8: seq=1 ttl=56 time=22.910 ms
64 bytes from 8.8.8.8: seq=2 ttl=56 time=22.747 ms
^C
--- 8.8.8.8 ping statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 22.747/22.946/23.183 ms
root@orion:~#
```
LuCI's load balancing display is now correct as well:

MWAN Interface Live Status
wan (eth1.99)
Offline
wan2 (eth1.2)
Online (tracking active)

feckert, 2017-10-04:
I have seen that in the log that the flushing is not enbaled

> Tue Oct  3 17:56:50 2017 user.info mwan3[11632]: connection tracking not flushed on interface wan2 (eth1.2) ifdown

On simple failover scenario i have enabled the conntrack flushing. Could you try this? If it is not working i have to do i deeper look.

feckert, 2017-10-04:
I also have seen that the metric of eth1.2 and eth1.99 are the same (metric 1).
The metric should be different! You should change this in this way that the metric should be different

ratsputin, 2017-10-04:
OK, as suggested, I updated /etc/config/mwan3 as follows:
```
config interface 'wan'
        option enabled '1'
        option reliability '2'
        option count '1'
        option timeout '2'
        option interval '5'
        option down '3'
        option up '8'
        option initial_state 'online'
        option family 'ipv4'
        list track_ip '8.8.4.4'
        list track_ip '8.8.8.8'
        option track_method 'ping'
        option size '56'
        option failure_interval '5'
        option recovery_interval '5'
        option flush_conntrack 'always'

config interface 'wan2'
        option count '1'
        option timeout '2'
        option interval '5'
        option down '3'
        option up '8'
        option initial_state 'online'
        option family 'ipv4'
        list track_ip '8.8.8.8'
        list track_ip '8.8.4.4'
        option track_method 'ping'
        option size '56'
        option failure_interval '5'
        option recovery_interval '5'
        option flush_conntrack 'always'
        option enabled '1'
        option reliability '2'
```
I then disconnected the cable from the cable modem on eth1.99 and saw the following in the syslog:
```
Wed Oct  4 16:28:51 2017 user.notice mwan3track[15117]: Interface wan (eth1.99) is offline
Wed Oct  4 16:28:51 2017 user.notice mwan3[17176]: Execute ifdown event on interface wan (eth1.99)
Wed Oct  4 16:28:51 2017 user.info mwan3[17176]: connection tracking flushed on interface wan (eth1.99) ifdown
Wed Oct  4 16:28:51 2017 user.info mwan3track[15117]: Detect ifdown event on interface wan (eth1.99)
```
Checking the mwan3 status, I saw the following:
```
root@orion:/etc/config# /usr/sbin/mwan3 status
Interface status:
 interface wan is offline and tracking is active
 interface wan2 is online and tracking is active
```
Next, attempting to ping resulted in the following:
```
root@orion:/etc/config# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
^C
--- 8.8.8.8 ping statistics ---
33 packets transmitted, 0 packets received, 100% packet loss
root@orion:/etc/config#
```
During this event, LuCI showed the following:

MWAN Interface Live Status
wan (eth1.99)
Online (tracking active)
wan2 (eth1.2)
Online (tracking active)

I did check the gateway priorities in LuCI.  WAN is set to a gateway metric of 10; WAN2 is set to a gateway metric of 20.  I'm not sure where I'd actually set the physical interface metrics; I don't see a way to do that.

feckert, 2017-10-05:
You have to set the metric config under /etc/config/network for this devices.

The mwan3 status output looks ok for me.
I think the luci-app-mwan3 did not show the right state because (https://github.com/openwrt/luci/blob/a876daa4ef390ea0787e2f95b26e87f1b28e1210/applications/luci-app-mwan3/luasrc/controller/mwan3.lua#L70) "ip route table" will not be deleted anymore since PR https://github.com/openwrt/packages/pull/4811 so the interface shows always "online". This needs to get updated.

And an other issue could be that router initiated traffic must be bound to an interface (self interface). If you make a "traceroute" before and after the fail-over behind the router with a lan device (laptop) i assume every thing is fine. Only router initiated traffic will not work. You have to set "local_source" to lan (https://github.com/openwrt/luci/blob/a876daa4ef390ea0787e2f95b26e87f1b28e1210/applications/luci-app-mwan3/luasrc/model/cbi/mwan/globalsconfig.lua#L20) in /etc/config/mwan3 global section.

feckert, 2017-10-05:
@ratsputin replace line https://github.com/openwrt/luci/blob/a876daa4ef390ea0787e2f95b26e87f1b28e1210/applications/luci-app-mwan3/luasrc/controller/mwan3.lua#L70 with 

> if ut.trim(sys.exec("cat /var/run/mwan3/iface_state/" .. interfaceName .. " 2>/dev/null" )) == "online" then

this will fix luci-app-mwan3 status
if you like could you please open a PR?

gonzalodt1, 2017-10-05:
I've Reboot (17.01.2, r3435-65eec8bd5f), which is the tree of files to arrive to that file??
My problem is the same.

ratsputin, 2017-10-06:
@feckert, okay, updating ./usr/lib/lua/luci/controller/mwan3.lua as suggested did address the issue with the LuCI interface properly displaying the interface status.

With regard to the interface priorities, I made the following change to /etc/config/network:

```
config switch_vlan
        option device 'switch0'
        option vlan '2'
        option vid '99'
        option ports '4 6t'
        option metric '10'

config switch_vlan
        option device 'switch0'
        option vlan '3'
        option vid '2'
        option metric '20'
        option ports '3 6t'
```
I don't have a before / after result from an ip route show, but I do have the following once that change was made:
```
root@orion:/# ip route show
default via 98.196.52.1 dev eth1.99 proto static src 98.196.55.178 metric 10
default via 104.51.84.1 dev eth1.2 proto static src 104.51.85.38 metric 20
10.1.0.0/24 dev eth0.10 proto kernel scope link src 10.1.0.1
98.196.52.0/22 dev eth1.99 proto static scope link metric 10
98.196.52.1 dev eth1.99 proto static scope link src 98.196.55.178 metric 10
104.51.84.0/22 dev eth1.2 proto static scope link metric 20
104.51.84.1 dev eth1.2 proto static scope link src 104.51.85.38 metric 20
192.168.1.0/24 dev br-lan proto kernel scope link src 192.168.1.1
192.168.3.0/24 dev eth0.3 proto kernel scope link src 192.168.3.1
root@orion:/#
```
I did test disconnecting the cable and, as mentioned above, LuCI did reflect the proper status and a ping was successful from a host on the br-lan network (192.168.1.x).

I do see, however, what the issue is with mwan3 not flushing routes.  Any service that's running on the router (such as BIND in my case, or the standard DNS server/forwarder) isn't going to work correctly.  While pings and such did work fine from a host on the network behind the router, BIND did not work, as the routing tables on the router are not reflecting the proper routes.

I guess I'm confused as to why mwan3 isn't flushing the routes, or at least the default gateway associated with the "downed" interface.  As long as the primary default gateway is pointing to an interface that has no Internet connectivity, any services running on the router will not be able to connect to the Internet.

Perhaps a fix would either be for mwan3 to manage the default gateways (I'd argue there really should only be one at any given time--"default" implies just one), or, if the routing actually supports, it, perhaps mwan3 could manage the priorities?

feckert, 2017-10-06:
@ratsputin 

> I guess I'm confused as to why mwan3 isn't flushing the routes, or at least the default gateway associated with the "downed" interface. As long as the primary default gateway is pointing to an [...]

You have to set the config option "local_source" in the global section of the mwan3 config file to "lan"

> config globals 'globals'
> 	option local_source 'lan'

This config option will set the ip addresse of your lan interface also to the loopback interface "lo".
Every initiated traffic from the router will now be treated as a normal package from the lan zone or lan interface. And so will go throw the mangle table of the firewall and will be marked with the right mask.

gonzalodt1, 2017-10-06:
@feckert 

> @ratsputin replace line https://github.com/openwrt/luci/blob/a876daa4ef390ea0787e2f95b26e87f1b28e1210/applications/luci-app-mwan3/luasrc/controller/mwan3.lua#L70 with

if ut.trim(sys.exec("cat /var/run/mwan3/iface_state/" .. interfaceName .. " 2>/dev/null" )) == "online" then
this will fix luci-app-mwan3 status
if you like could you please open a PR?

I've had this, but now, the interfaces are always "offline".

ratsputin, 2017-10-06:
Thank you for working with me on this, @feckert.  I have some further information for you.

I made the following configuration change to my /etc/config/mwan3:
```
config globals 'globals'
        option local_source 'br-lan'
```
I used br-lan, as that's my local LAN interface as shown below:
```
root@orion:/etc/config# ifconfig
br-lan    Link encap:Ethernet  HWaddr 62:38:E0:BB:1D:C8
          inet addr:192.168.1.1  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: 2601:2c5:c700:4cb1::1/64 Scope:Global
          inet6 addr: 2602:306:8335:526a::1/64 Scope:Global
          inet6 addr: fdda:5e85:487::1/60 Scope:Global
          inet6 addr: fe80::6038:e0ff:febb:1dc8/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:11215 errors:0 dropped:78 overruns:0 frame:0
          TX packets:7142 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1439616 (1.3 MiB)  TX bytes:2165720 (2.0 MiB)
```
I then disconnected the cable from my cable modem and was still not able to ping 8.8.8.8 from the router command-line.  Looking at /etc/hotplug.d/iface/14-mwan3, I found the logic that handles the routing change you're referring to.  As it appears to not have worked, I went ahead and manually followed the logic:
```
root@orion:/etc/config# ip addr add 192.168.1.1/32 dev lo
root@orion:/etc/config# ip route add default via 192.168.1.1 dev lo
root@orion:/etc/config# cd /usr/sbin
root@orion:/usr/sbin# . /lib/functions.sh
root@orion:/usr/sbin# . /lib/mwan3/mwan3.sh
root@orion:/usr/sbin# . /lib/functions/network.sh
root@orion:/usr/sbin# uci_toggle_state mwan3 globals src_ip 192.168.1.1
root@orion:/usr/sbin# ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=0 ttl=56 time=23.853 ms
64 bytes from 8.8.8.8: seq=1 ttl=56 time=78.677 ms
64 bytes from 8.8.8.8: seq=2 ttl=56 time=24.051 ms
^C
--- 8.8.8.8 ping statistics ---
3 packets transmitted, 3 packets received, 0% packet loss
round-trip min/avg/max = 23.853/42.193/78.677 ms
root@orion:/usr/sbin#
```
As you can see, running the sets manually worked fine.  This struck me as odd.  Apparently, the logic is sound, but it's not being executed.  I went ahead and double-checked that I had the configuration set correctly:
```
root@orion:/usr/sbin# uci_get_state mwan3 globals src_ip
192.168.1.1
root@orion:/usr/sbin# config_load mwan3
root@orion:/usr/sbin# config_get local_source globals local_source
root@orion:/usr/sbin# set | grep local_source
CONFIG_globals_local_source='br-lan'
local_source='br-lan'
root@orion:/usr/sbin#
```
So, as you can see, the configuration change was apparently correct as well.

The only thing I see in the script that might be interfering is the following line:
```
[ -n "$INTERFACE" ] || exit 2
```
I don't know what $INTERFACE is at the point this script is running.  Of course, I'm just speculating at this point.

Your guidance here is greatly appreciated.

Kind regards,

Brett

feckert, 2017-10-06:
@gonzalodt1 This is only for version 17.01.3

feckert, 2017-10-06:
@ratsputin Nice to hear that things are working more or less as expected.

The script will only be executed if the "local_source" ( in your situation br-lan ) is equal to the INTERFACE which is setting up by "netifd" https://github.com/openwrt/packages/blob/139015087cd8408514decc739399b169755302be/net/mwan3/files/etc/hotplug.d/iface/14-mwan3#L20

So if you do a ifdown br-lan && ifup br-lan every thing should work as expected.

The config "src_ip" is only a placeholder if the br-lan gets changed and if so the last ip must be removed from "lo" before the new ip is set.
https://github.com/openwrt/packages/blob/139015087cd8408514decc739399b169755302be/net/mwan3/files/etc/hotplug.d/iface/14-mwan3#L26

ratsputin, 2017-10-06:
@feckert, I tried the following:
```
root@orion:/etc/hotplug.d/iface# ifdown br-lan && ifup br-lan
Interface br-lan not found
Interface br-lan not found
root@orion:/etc/hotplug.d/iface#
```
I'm gathering that's not the expected behavior?  :)

Could this be the source of my problems?

gonzalodt1, 2017-10-10:
Could not fix the problem of updating the table yet?

gonzalodt1, 2017-10-10:
My problem is the same. When you disconnect the interface cable your _ip route_ is:
`default via 147.96.67.1 dev eth1.2  proto static
default via 147.96.25.1 dev eth1.3  proto static  metric 10
default via 147.96.67.1 dev eth1.4  proto static  metric 20
147.96.25.0/24 dev eth1.3  proto static  scope link  metric 10
147.96.67.0/24 dev eth1.2  proto kernel  scope link  src 147.96.67.97
147.96.67.0/24 dev eth1.4  proto static  scope link  metric 20
192.168.1.0/24 dev br-lan  proto kernel  scope link  src 192.168.1.1
`
but when you go to Network<Interfaces in Luci and stop WAN2 interface, your _ip route_ is:

`default via 147.96.67.1 dev eth1.2  proto static
default via 147.96.67.1 dev eth1.4  proto static  metric 20
147.96.67.0/24 dev eth1.2  proto kernel  scope link  src 147.96.67.97
147.96.67.0/24 dev eth1.4  proto static  scope link  metric 20
192.168.1.0/24 dev br-lan  proto kernel  scope link  src 192.168.1.1
`

As it should be doing the other.

feckert, 2017-10-11:
@ratsputin sorry for the wrong information. You have to use the logical interface names in your scenario it should be "ifdown lan && ifup lan". The interface names are the names in the "/etc/config/network"

feckert, 2017-10-11:
@gonzalodt1 FYI if the interface get detected as down by mwan3 it will only execute the down event for the "hotplug-scripts" under "/etc/hotplug.d/iface" it will not take down the interface completely so that the routes already get cleared by netifd as you have done in luci (button).

ratsputin, 2017-10-11:
@feckert I went ahead and ran the "ifdown lan && ifup lan"; unfortunately, it crashed my router, so I'm not sure if it had the intended effect or not, as I'm not sure what the intent of those commands were.

I did, however, disconnect the cable from the back of my cable modem after the router rebooted and received the same behavior as before, where BIND and Tor, which are running on the router, are unable to connect to the Internet.

I did do a bit of digging and found documentation on Hotplug ([https://wiki.openwrt.org/doc/techref/hotplug]).  I believe I see what you're doing now.

I added the following code just before the interface test you mentioned before to /etc/hotplug.d/14-mwan3
```
logger -t mwan3 Hotplug: Device: $DEVICE / Action: $ACTION
```
I then ran the following commands to look at the router config:
```
root@orion:~# . /lib/functions.sh
root@orion:~# . /lib/mwan3/mwan3.sh
root@orion:~# . /lib/functions/network.sh
root@orion:~# config_load mwan3
root@orion:~# config_get local_source globals local_source 'none'
root@orion:~# set | grep local_source
CONFIG_globals_local_source='br-lan'
local_source='br-lan'
root@orion:~#
```
Next, I disconnected the cable from the back of the cable modem and verified $INTERFACE is being set as expected:
```
Wed Oct 11 07:13:42 2017 user.notice mwan3track[8505]: Interface wan (eth1.99) is offline
Wed Oct 11 07:13:42 2017 user.notice mwan3: Hotplug: Device: eth1.99 / Action: ifdown
Wed Oct 11 07:13:42 2017 user.notice mwan3[14188]: Execute ifdown event on interface wan (eth1.99)
Wed Oct 11 07:13:42 2017 user.info mwan3[14188]: connection tracking flushed on interface wan (eth1.99) ifdown
Wed Oct 11 07:13:42 2017 user.notice root: stopping ntpclient
Wed Oct 11 07:13:42 2017 user.info mwan3track[8505]: Detect ifdown event on interface wan (eth1.99)
```
Note, the Hotplug call is for the physical device (eth1.99) of the WAN connection that's been disconnected, not the virtual (br-lan) interface for the LAN.
```
Wed Oct 11 07:15:26 2017 user.notice mwan3track[8505]: Interface wan (eth1.99) is online
Wed Oct 11 07:15:26 2017 user.notice mwan3: Hotplug: Device: eth1.99 / Action: ifup
Wed Oct 11 07:15:26 2017 user.notice mwan3[15147]: Execute ifup event on interface wan (eth1.99)
Wed Oct 11 07:15:26 2017 user.notice mwan3[15147]: Starting tracker on interface wan (eth1.99)
Wed Oct 11 07:15:26 2017 user.info mwan3[15147]: connection tracking flushed on interface wan (eth1.99) ifup
```
You had suggested I make the following change to /etc/config/mwan3
```
config globals 'globals'
        option local_source 'br-lan'
```
The issue with the code, though, is the call to 14-mwan3 is not being made with the name of the LAN interface, but the WAN interface that has gone down.  Therefore, I don't see how the following code can work:
```
[ "${local_source}" = "$INTERFACE" ] || {
        exit 0
}
```
Even if I were to make the following change to /etc/config/mwan3:
```
config globals 'globals'
       option local_source 'eth1.99'
```
So that the test does execute correctly, it wouldn't work in the case of my backup interface (eth1.2) going down.

If the test ("${local_source}" = "$INTERFACE") is correct, could you please explain its purpose so I can troubleshoot further?

Kind regards,

Brett

ratsputin, 2017-10-12:
@feckert I believe I may have determined the source of my issue.  If you look at my previous updates where I listed the routing tables, there was no default gateway pointed to localhost.

I happened to notice the configuration options in LuCI where Globals could be set.  There, I noticed that the Local Source wasn't set (even though I had added it in the mwan3 config file).  I went ahead and set it there.  Since then, I've noticed the following:
```
root@orion:/etc/bind# netstat -rn
Kernel IP routing table
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
0.0.0.0         192.168.1.1     0.0.0.0         UG        0 0          0 lo
0.0.0.0         104.51.84.1     0.0.0.0         UG        0 0          0 eth1.2
0.0.0.0         98.196.52.1     0.0.0.0         UG        0 0          0 eth1.99
10.1.0.0        0.0.0.0         255.255.255.0   U         0 0          0 eth0.10
98.196.52.0     0.0.0.0         255.255.252.0   U         0 0          0 eth1.99
98.196.52.1     0.0.0.0         255.255.255.255 UH        0 0          0 eth1.99
104.51.84.0     0.0.0.0         255.255.252.0   U         0 0          0 eth1.2
104.51.84.1     0.0.0.0         255.255.255.255 UH        0 0          0 eth1.2
192.168.1.0     0.0.0.0         255.255.255.0   U         0 0          0 br-lan
192.168.3.0     0.0.0.0         255.255.255.0   U         0 0          0 eth0.3
root@orion:/etc/bind#
```
Note, the first line now has a default gateway pointed to localhost.

It appears just changing local_source to 'lan' in the globals section of the config file manually isn't sufficient to make this change.

I have further testing to do, but I noticed it was working better when I was rewiring my Internet connections.

I'll follow up with an update once I have an opportunity to do further testing.

ratsputin, 2017-10-21:
@feckert Thank you for your assistance with this issue.  Everything seems to be working fine.  The fix was to set the Local Source Interface to LAN in the LuCI Network > Load Balancing > Configuration > Globals screen.  Apparently, just setting option local_source 'lan' in /etc/config/mwan3 was insufficient.

I did notice that setting that value in the LuCI config also added the line option mmx_mask '0xff00'.  The purpose of that is unclear; however, the end result was the following route being added to the internal routing tables:
```
Destination     Gateway         Genmask         Flags   MSS Window  irtt Iface
0.0.0.0         192.168.1.1     0.0.0.0         UG        0 0          0 lo
```
This addressed the issue with the local services being unable to communicate with the backup link when the primary was down.

Kind regards,

Brett

closed 2017-10-21 by ratsputin

feckert, 2017-10-22:
@ratsputin good to know that everything is working as expected.
mmx_mask '0xff00' is the mask of the firewall in the mangle table.
For mwan3 the default firewall marks is 0xff00 but this overlapped with some other firewall application which uses the same area. So with this option you could configure the firewall mark value.
If the value is 0xff you can track 252 interfaces. If you set the value to 0xf0 you can track 6 interfaces.
Two Firewall marks are always used for blachole/unreachable marking!

On this Page are some more informations.
https://wiki.openwrt.org/doc/howto/mwan3#mwan3_routing