openwrt/packages #4910: mwan3: Correct status not being reflected in LuCI MWAN Interface Live Status

ratsputin, 2017-10-06:
-------------------------------

Maintainer: @feckert 
Environment: 
OpenWrt - LEDE Reboot SNAPSHOT r4883-ce53c0e
LuCI - git-17.260.43176-678c15c
mwan3 - 2.6.6-1
mwan3-luci - git-17.260.43176-678c15c-1

Description:

When an ifdown event takes place, /usr/sbin/mwan3 interfaces reflects the status of the interface appropriately (online/offline); however, the MWAN Interface Live Status in LuCI always shows green and says "Online".

This ticket was opened as requested in issue #4882

Dainiakas, 2017-10-23:
Same issue here everything works, except MWAN Interface Live Status in LuCi

OpenWrt Designated Driver 50139 
LuCI Master (git-17.294.24601-59a4e40)
luci-app-mwan3 | git-17.294.24601-59a4e40-1
mwan3 | 2.6.6-1

/var/run/mwan3track/wan2/STATUS changes accordingly, however in LuCi it's always shows as Online (tracking active).

/cgi-bin/luci/admin/network/mwan/overview/interface_status 
returns status incorrect 
{"status":"online","ifname":"wlan1","name":"wan2","link":"\/cgi-bin\/luci\/admin\/network\/network\/wan2"}

Interface status:
 interface wan is online and tracking is active
 interface wan2 is offline and tracking is active

Current ipv4 policies: (default policies)
balanced:
 wan (100%)

wan2_only:
 unreachable

wan2_wan:
 wan (100%)

wan_only:
 wan (100%)

wan_wan2:
 wan (100%)

ratsputin, 2017-10-23:
Read the notes in Issue #4882.  @feckert provides an edit to fix this as an interim fix before the permanent fix is committed.  It should be about 2/3 of the way down.

feckert, 2017-10-25:
I have added a PR #5012 to get this fixed in lede-17.01.
On Master i think we have to use the new "ubus" interface to get the this state.
So I think it is only a fix for lede-17.01 and should not ported the master.

lucize, 2017-11-10:
do you have an eta, or where should I look for, I would like to use this on a device that is using master

feckert, 2017-11-12:
@lucize i have not talked to @arfett since i am the maintainer of mwan3 ( i am not sure if he is on board anymore?)  but i want to move the status info to the status section in luci. Also i want to change some calls from luci (status, info) to the ubus interface i have added so far. But i have no plans for now when i will do this. For a workaround you could use the fix from branch lede-17.01 on the package feed and apply it to this luci package feed (FYI Package mwan-luci moved to the luci feed luci-app-mwan3 on master)

arfett, 2017-11-15:
@feckert I am using LEDE again and will be playing with mwan3 with IPv6 over the coming months. When you move the status what will you be doing with the detailed status page... move to advanced?

feckert, 2017-11-16:
@arfett nice to talk to you :-) welcome back on board
- luci-app-mwan3
I want to move the status pages of the luci-app-mwan3 to "Network->Loadbalancing->Overview->(Interface Status) && (Detail Status)" to "Status->Loadbalancing" additonal I think the code needs some refactoring for example use the new ubus interface of mwan3.
- Ipv6
This would help me a lot if you could play around with Ipv6. This is also on my roadmap.

feckert, 2017-11-16:
@lucize have a look at #5012

lucize, 2017-11-16:
Yes, the old version is ok, I need to work around the dns issue if the main link is down, the new version has the local route that seems to be a fix for this case

lucize, 2017-11-18:
@feckert maybe it has to do with fast path ? I didn't test without it, the status is ok with old version but not the load balancing

feckert, 2017-12-07:
@lucize what do you mean with  "fast path"?

lucize, 2017-12-07:
The Qualcomm sfe nat driver

camelcamro, 2018-01-21:
additional ... i tested latest version, but still the status are not OK ...

eg: i have a INTERFACE: "4G" and "wan" which is used for internet ...

wan = eth0.2 
4G = 3g-4G

and i have only 4G connected, and wan cable is not connected, but:

after a reboot => wan is shown as OFFLINE and RED, but after a restart of mwan3 it is shown:
--------------
wan (eth0.2)
Online (tracking active)
(GREEN color)
----
Interface status:
 interface wan is unknown and tracking is active
---------------

why is that GREEN ? (cable is not connected to another modem on WAN ?

feckert, 2018-01-23:
Hello

What Proto do you have in the interfaces configurieren?

camelcamro, 2018-01-24:
i guess i found the main problem.
If the firewall will be restarted, that will destroy (flush) all mwan3 iptables rule, and then it is showing error.
(some hotplug scripts needs to "restart" (not fw3 reload) the firewall. (in my case, i did not figure out, but my PPTP server is otherwise not reachable, it seems to be, that mwan3 is manipulating any rule on routing/firewall and then the port 1723 and prtotcol GRE is not reachable. THerefore i have a check if GRE érrors and PPTP clients not pingable then i need to fix it with a firewall restart.

i need to figure out what is really the problem ... i guess there are some mangle rules maybe which are wrong, and then the port 1723 is not reahable from outside, and the packages are using wrong interface.
it is very strange ... but it is something related to mwan3 which was not a problem 6 months ago ..

in any case:

1 question:
which command can i use to create the FW rules again for MWAN3 ?
as a "mwan3 start" or "mwan3 restart" would do too much other steps, which are not wanted.
is there a command to add only all firewall & routing rules ?
how can i show rules + ip routing + after a restart of fw ?
iptables -S is not showing me the mwan3 rules
how can i show rules + ip routing + after a restart of fw + restart of mwan3 to figure out the diference ?

i know, if you really want to support me, i would need to show you all my config and firewall stuff, but i guess it is a little bit more complex:

but i will try to tell:
my main "router2" is using 3 other routers which are connected to "br-lan",
the other routers (router3,4,5) are connecting via ppp client to ppp server "router02"

so, on main router i have:
1.) WAN (wan port - mostly not really connected)
2.) 4G (3g/4g modem on router2)
3.) VPN_ROUTER3 (router3 connected via ppp)
4.) VPN_ROUTER3 (router4 connected via ppp)
5.) VPN_ROUTER3 (router5 connected via ppp)

this 5 devices will be used in mwan3 with different rules for my users ..

this 4 routers are also using "keepalived" if any of the routers are OFFLINE
in logical order (router2,3,4,5)

also router02 is acting as a PPTP server using tcp 1723 and proto GRE ...

additional, the connected PPP clients can reach others ... (that means ppp device to another one ,,)
and therefore i have a lot of rules in /etc/firewall.user to handle it ...

but after a while i have seen, that:
.) ppp clients can't reach internal network from server
.) ppp server can't reach client ppp users ...
but if i ping with INTERFACE it works -> "ping 192.168.3.102 -I ppp_router04"
but if i ping without INTERFACE it DOESN'T works -> "ping 192.168.3.102 -I"
but routing table (route -n) is 100% identically.

then i restart firewall (etc/init.d/firewall restart), and then the PING is working without using interface, too. (routing table is the same, but maybe something wrong with mangle when mwan3 restart a interface or don'T know ....
hmm ...

feckert, 2018-03-15:
Sorry for the late reply

> which command can i use to create the FW rules again for MWAN3 ?
as a "mwan3 start" or "mwan3 restart" would do too much other steps, which are not wanted.

you could use `mwan3 ifup <iface>`

> how can i show rules + ip routing + after a restart of fw ?

`ip rule list table all`

> iptables -S is not showing me the mwan3 rules

`iptables -t mangle -S`

> 1.) WAN (wan port - mostly not really connected)
2.) 4G (3g/4g modem on router2)
3.) VPN_ROUTER3 (router3 connected via ppp)
4.) VPN_ROUTER3 (router4 connected via ppp)
5.) VPN_ROUTER3 (router5 connected via ppp)

Yes this is a very complex setup :-)

What about the `self` interface have setup this up
https://wiki.openwrt.org/doc/howto/mwan3#the_routable_loopback_self

camelcamro, 2018-12-31:
hmm, very complex topic, i guess we should close as mwan3 feckert fixed that much things, and better close that ticket, and lets make a new if still not fixed.

closed 2019-02-27 by ratsputin