openwrt/packages #4912: dovecot - missing doveadm

spljaa, 2017-10-07:
LEDE 17.01.02
Maintainer: @lucize 

In dovecot package there is missing doveadm utility. 

Are there some step-by-step instructions how to build such package on x64 linux? If so I could try to create build with doveadm included. Although I have no experience with cross-compilation.

lucize, 2017-10-07:
you need to rebuild it yourself, doveadm and doveadm-server are compiled during main build but not included in the package, the easy way is to follow the wiki how to make your own build
```
git clone https://git.lede-project.org/source.git lede
cd lede

./scripts/feeds update -a
./scripts/feeds install -a

make defconfig
make menuconfig
```
 and select x86_64 target and dovecot then
`make V=s -j4`
then in
`build_dir/target-x86_64_musl/dovecot-2.2.32/src/doveadm` (2.2.32 is the trunk version maybe you need the 17.01.02 version if you don't want to replace your default version)
You will find .libs and dsync/.libs folder with the binary files you need `doveadm, doveadm-server, libdovecot-dsync.so, libdovecot-dsync.so.0, libdovecot-dsync.so.0.0.0, libdovecot-dsync.so.0.0.0T`
Maybe i'll add an option to enable the doveadm inclusion, but I don't think it will be enabled by default (+1MB size) so anyway you will have to learn hot to build it yourself (no so hard)

spljaa, 2017-10-07:
Thank you.
Yes, that worked well.
I put 2.2.32 inside 17.01.02... so far no issues, but maybe I should learn one day how to retrieve exactly 17.01.02 from git.

Maybe creating additional package with add-on to dovecot could be an option. I do not know how complex it is to define and maintain such package. For me, building package for myself is solution.

Regards
  Jan

PR #5182 (2017-12-02): dovecot: prepare for runtime dependency by plugins
Maintainer: @lucize 
Attention pigeonhole package maintainer: @MikePetullo 
Compile tested: bcm53xx, Buffalo WXR-1900DHP, LEDE Reboot SNAPSHOT r5297-bddffc5
Run tested: bcm53xx, Buffalo WXR-1900DHP, LEDE Reboot SNAPSHOT r5297-bddffc5

Description:

Add ABI_VERSION declaration in order to tell dovecot plugins that are runtime-dependent on dovecot's Application Binary Interface (ABI) version, such as the pigeonhole mail filter, that they need to be rebuilt each time the dovecot's ABI version changes.

This PR is not yet closing issue #5062. A fo [... cut]
commit 12f0caf3 (2017-12-02): dovecot: prepare for runtime dependency by plugins

Add ABI_VERSION declaration in order to tell dovecot plugins that are
runtime-dependent on dovecot's Application Binary Interface (ABI)
version, such as the pigeonhole mail filter, that they need to be
rebuilt each time the dovecot's ABI version changes.

This PR is not yet closing issue #5062. A forthcoming PR for pigeonhole
package will close the issue.

Introduce a new configuration parameter to enable ICU (i18n) support.
Make the ICU disabled by default.

Introduce PKG_CONFIG_DEPENDS with the list of configuration parameters
triggering re [... cut]
closed 2017-12-02 by champtar