openwrt/packages #4917: libiio compile error

stintel, 2017-10-08:
Maintainer: @mhei
Environment: LEDE master r5003

Description:
Unable to build libiio
```
make[2]: Entering directory '/home/stijn/Development/LEDE/openwrt-packages/libs/libiio'
touch /home/stijn/Development/LEDE/source/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/libiio-0.10/.prepared_34b468adedf3b56863cefa041dd52ef3_6664517399ebbbc92a37c5bb081b5c53_check
. /home/stijn/Development/LEDE/source/include/shell.sh; gzip -dc /home/stijn/Development/LEDE/source/dl/libiio-0.10.tar.gz | tar -C /home/stijn/Development/LEDE/source/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/libiio-0.10/.. -xf - 
[ ! -d ./src/ ] || cp -fpR ./src/* /home/stijn/Development/LEDE/source/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/libiio-0.10
touch /home/stijn/Development/LEDE/source/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/libiio-0.10/.prepared_34b468adedf3b56863cefa041dd52ef3_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/stijn/Development/LEDE/source/staging_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/stamp/.libiio_installed
mkdir -p /home/stijn/Development/LEDE/source/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/libiio-0.10
(cd /home/stijn/Development/LEDE/source/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/libiio-0.10; CFLAGS="-Os -pipe -mcpu=arm1176jzf-s -mfpu=vfp -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/stijn/Development/LEDE/source/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/libiio-0.10:libiio-0.10 -Wformat -Werror=format-security -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mcpu=arm1176jzf-s -mfpu=vfp -g3 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap/home/stijn/Development/LEDE/source/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/libiio-0.10:libiio-0.10 -Wformat -Werror=format-security -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wl,-z,now -Wl,-z,relro " LDFLAGS="-L/home/stijn/Development/LEDE/source/staging_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-5.4.0_musl_eabi/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-5.4.0_musl_eabi/lib -znow -zrelro " cmake -DCMAKE_SYSTEM_NAME=Linux -DCMAKE_SYSTEM_VERSION=1 -DCMAKE_SYSTEM_PROCESSOR=arm -DCMAKE_BUILD_TYPE=Release -DCMAKE_C_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_CXX_FLAGS_RELEASE="-DNDEBUG" -DCMAKE_C_COMPILER="/home/stijn/Development/LEDE/source/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-5.4.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc" -DCMAKE_C_COMPILER_ARG1="" -DCMAKE_CXX_COMPILER="/home/stijn/Development/LEDE/source/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-5.4.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-g++" -DCMAKE_CXX_COMPILER_ARG1="" -DCMAKE_ASM_COMPILER="/home/stijn/Development/LEDE/source/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-5.4.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc" -DCMAKE_ASM_COMPILER_ARG1="" -DCMAKE_EXE_LINKER_FLAGS:STRING="-L/home/stijn/Development/LEDE/source/staging_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-5.4.0_musl_eabi/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-5.4.0_musl_eabi/lib -znow -zrelro" -DCMAKE_MODULE_LINKER_FLAGS:STRING="-L/home/stijn/Development/LEDE/source/staging_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-5.4.0_musl_eabi/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-5.4.0_musl_eabi/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_SHARED_LINKER_FLAGS:STRING="-L/home/stijn/Development/LEDE/source/staging_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-5.4.0_musl_eabi/usr/lib -L/home/stijn/Development/LEDE/source/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-5.4.0_musl_eabi/lib -znow -zrelro -Wl,-Bsymbolic-functions" -DCMAKE_AR="/home/stijn/Development/LEDE/source/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-5.4.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc-ar" -DCMAKE_NM="/home/stijn/Development/LEDE/source/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-5.4.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc-nm" -DCMAKE_RANLIB="/home/stijn/Development/LEDE/source/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-5.4.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc-ranlib" -DCMAKE_FIND_ROOT_PATH="/home/stijn/Development/LEDE/source/staging_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/usr;/home/stijn/Development/LEDE/source/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-5.4.0_musl_eabi" -DCMAKE_FIND_ROOT_PATH_MODE_PROGRAM=BOTH -DCMAKE_FIND_ROOT_PATH_MODE_LIBRARY=ONLY -DCMAKE_FIND_ROOT_PATH_MODE_INCLUDE=ONLY -DCMAKE_STRIP=: -DCMAKE_INSTALL_PREFIX=/usr -DDL_LIBRARY=/home/stijn/Development/LEDE/source/staging_dir/target-arm_arm1176jzf-s+vfp_musl_eabi -DCMAKE_PREFIX_PATH=/home/stijn/Development/LEDE/source/staging_dir/target-arm_arm1176jzf-s+vfp_musl_eabi -DCMAKE_SKIP_RPATH=TRUE  /home/stijn/Development/LEDE/source/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/libiio-0.10 )
-- The C compiler identification is GNU 5.4.0
-- Check for working C compiler: /home/stijn/Development/LEDE/source/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-5.4.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc
-- Check for working C compiler: /home/stijn/Development/LEDE/source/staging_dir/toolchain-arm_arm1176jzf-s+vfp_gcc-5.4.0_musl_eabi/bin/arm-openwrt-linux-muslgnueabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test HAS_WPEDANTIC
-- Performing Test HAS_WPEDANTIC - Success
-- Looking for strdup
-- Looking for strdup - found
-- Looking for strerror_r
-- Looking for strerror_r - found
-- Looking for newlocale
-- Looking for newlocale - found
-- Looking for in6addr_any
-- Looking for in6addr_any - found
-- Found Git: /home/stijn/Development/LEDE/source/staging_dir/host/bin/git (found version "2.13.6") 
fatal: Not a git repository (or any parent up to mount point /home)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
-- Performing Test HAS_O_TMPFILE
-- Performing Test HAS_O_TMPFILE - Success
-- Performing Test WITH_NETWORK_EVENTFD
-- Performing Test WITH_NETWORK_EVENTFD - Success
-- Performing Test HAS_PIPE2
-- Performing Test HAS_PIPE2 - Success
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - found
-- Found Threads: TRUE  
-- Found ZLIB: /home/stijn/Development/LEDE/source/staging_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/usr/lib/libz.so (found version "1.2.11") 
-- Found PythonInterp: /home/stijn/Development/LEDE/source/staging_dir/hostpkg/bin/python (found version "2.7.14") 
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Doxygen not found, API documentation won't be generated
-- Found BISON: /home/stijn/Development/LEDE/source/staging_dir/host/bin/bison (found version "3.0.4") 
-- Found FLEX: /home/stijn/Development/LEDE/source/staging_dir/host/bin/flex (found version "2.6.4") 
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of struct usb_functionfs_descs_head_v2
-- Check size of struct usb_functionfs_descs_head_v2 - done
-- Looking for pthread_setname_np
-- Looking for pthread_setname_np - found
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_ASM_COMPILER
    CMAKE_ASM_COMPILER_ARG1
    CMAKE_CXX_COMPILER
    CMAKE_CXX_COMPILER_ARG1
[... 22 lines cut ...]
[ 20%] Building C object CMakeFiles/iio.dir/scan.c.o
[... cut ...]
make[3]: Leaving directory '/home/build/lede/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/libiio-0.10'
make[2]: *** [Makefile:106: /home/stijn/Development/LEDE/source/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/libiio-0.10/.built] Error 2
make[2]: Leaving directory '/home/stijn/Development/LEDE/openwrt-packages/libs/libiio'
make[1]: *** [package/Makefile:111: package/feeds/packages/libiio/compile] Error 2
make[1]: Leaving directory '/home/stijn/Development/LEDE/source'
make: *** [/home/stijn/Development/LEDE/source/include/toplevel.mk:209: package/libiio/compile] Error 2
```

mhei, 2017-10-08:
Maybe my latest upgrade of libxml2 is causing this build error? Will have a look, thanks for reporting...

mhei, 2017-10-11:
Hm, I'm not able to reproduce this error on my system. I checked out the given LEDE revision and used latest package trunk. Maybe you could share your .config via PM. Or can you try a fresh/clean build?

stintel, 2017-10-18:
After removing build_dir, staging_dir and tmp, it still fails to build.

Output of ./scripts/diffconfig.sh:
```
CONFIG_TARGET_brcm2708=y
CONFIG_TARGET_brcm2708_bcm2708=y
CONFIG_TARGET_brcm2708_bcm2708_DEVICE_rpi=y
CONFIG_BRCM2708_SD_BOOT_PARTSIZE=256
CONFIG_DEBUG=y
CONFIG_DROPBEAR_ECC=y
# CONFIG_KERNEL_CC_STACKPROTECTOR_REGULAR is not set
CONFIG_KERNEL_CC_STACKPROTECTOR_STRONG=y
CONFIG_LIBCURL_COOKIES=y
CONFIG_LIBCURL_FILE=y
CONFIG_LIBCURL_FTP=y
CONFIG_LIBCURL_HTTP=y
CONFIG_LIBCURL_NGHTTP2=y
CONFIG_LIBCURL_NO_SMB="!"
CONFIG_LIBCURL_OPENSSL=y
CONFIG_LIBCURL_PROXY=y
CONFIG_LLDPD_WITH_CDP=y
CONFIG_LLDPD_WITH_CUSTOM=y
CONFIG_LLDPD_WITH_DOT1=y
CONFIG_LLDPD_WITH_DOT3=y
CONFIG_LLDPD_WITH_EDP=y
CONFIG_LLDPD_WITH_FDP=y
CONFIG_LLDPD_WITH_LLDPMED=y
CONFIG_LLDPD_WITH_PRIVSEP=y
CONFIG_LLDPD_WITH_SONMP=y
CONFIG_MINIUPNPD_IGDv2=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_EC=y
CONFIG_OPENSSL_WITH_NPN=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_PACKAGE_BRCM80211_DEBUG=y
CONFIG_PACKAGE_bfgminer=y
CONFIG_PACKAGE_block-mount=y
CONFIG_PACKAGE_ca-certificates=y
CONFIG_PACKAGE_cgi-io=y
# CONFIG_PACKAGE_dnsmasq is not set
CONFIG_PACKAGE_e2fsprogs=y
CONFIG_PACKAGE_fontconfig=y
CONFIG_PACKAGE_gdb=y
CONFIG_PACKAGE_gdbserver=y
CONFIG_PACKAGE_iio-utils=y
CONFIG_PACKAGE_iiod=y
CONFIG_PACKAGE_ip-full=y
CONFIG_PACKAGE_iperf=y
CONFIG_PACKAGE_iperf3=y
CONFIG_PACKAGE_ipset=y
CONFIG_PACKAGE_iptables-mod-ipsec=y
CONFIG_PACKAGE_iputils-ping=y
CONFIG_PACKAGE_iputils-ping6=y
# CONFIG_PACKAGE_iwinfo is not set
CONFIG_PACKAGE_jansson=y
CONFIG_PACKAGE_kmod-bluetooth=y
CONFIG_PACKAGE_kmod-crypto-aead=y
CONFIG_PACKAGE_kmod-crypto-authenc=y
CONFIG_PACKAGE_kmod-crypto-cbc=y
CONFIG_PACKAGE_kmod-crypto-cmac=y
CONFIG_PACKAGE_kmod-crypto-deflate=y
CONFIG_PACKAGE_kmod-crypto-des=y
CONFIG_PACKAGE_kmod-crypto-ecb=y
CONFIG_PACKAGE_kmod-crypto-echainiv=y
CONFIG_PACKAGE_kmod-crypto-hash=y
CONFIG_PACKAGE_kmod-crypto-hmac=y
CONFIG_PACKAGE_kmod-crypto-iv=y
CONFIG_PACKAGE_kmod-crypto-manager=y
CONFIG_PACKAGE_kmod-crypto-md4=y
CONFIG_PACKAGE_kmod-crypto-md5=y
CONFIG_PACKAGE_kmod-crypto-null=y
CONFIG_PACKAGE_kmod-crypto-pcompress=y
CONFIG_PACKAGE_kmod-crypto-rng=y
CONFIG_PACKAGE_kmod-crypto-sha1=y
CONFIG_PACKAGE_kmod-crypto-sha256=y
CONFIG_PACKAGE_kmod-crypto-wq=y
CONFIG_PACKAGE_kmod-dma-buf=y
CONFIG_PACKAGE_kmod-drm=y
CONFIG_PACKAGE_kmod-drm-vc4=y
CONFIG_PACKAGE_kmod-fs-cifs=y
CONFIG_PACKAGE_kmod-i2c-core=y
CONFIG_PACKAGE_kmod-iio-bmp280=y
CONFIG_PACKAGE_kmod-iio-bmp280-i2c=y
CONFIG_PACKAGE_kmod-iio-bmp280-spi=y
CONFIG_PACKAGE_kmod-iio-core=y
CONFIG_PACKAGE_kmod-iio-dht11=y
CONFIG_PACKAGE_kmod-ip-vti=y
CONFIG_PACKAGE_kmod-ip6-tunnel=y
CONFIG_PACKAGE_kmod-ip6-vti=y
CONFIG_PACKAGE_kmod-ipsec=y
CONFIG_PACKAGE_kmod-ipsec4=y
CONFIG_PACKAGE_kmod-ipsec6=y
CONFIG_PACKAGE_kmod-ipt-ipsec=y
CONFIG_PACKAGE_kmod-ipt-ipset=y
CONFIG_PACKAGE_kmod-iptunnel=y
CONFIG_PACKAGE_kmod-iptunnel4=y
CONFIG_PACKAGE_kmod-iptunnel6=y
CONFIG_PACKAGE_kmod-lib-crc16=y
CONFIG_PACKAGE_kmod-lib-lzo=y
CONFIG_PACKAGE_kmod-lib-zlib-deflate=y
CONFIG_PACKAGE_kmod-lib-zlib-inflate=y
CONFIG_PACKAGE_kmod-nfnetlink=y
CONFIG_PACKAGE_kmod-regmap=y
CONFIG_PACKAGE_kmod-sched-core=y
CONFIG_PACKAGE_kmod-spi-bitbang=y
CONFIG_PACKAGE_kmod-udptunnel4=y
CONFIG_PACKAGE_kmod-udptunnel6=y
CONFIG_PACKAGE_kmod-usb-acm=y
CONFIG_PACKAGE_kmod-usb-serial=y
CONFIG_PACKAGE_kmod-usb-serial-ftdi=y
CONFIG_PACKAGE_kmod-wireguard=y
CONFIG_PACKAGE_libbase58=y
CONFIG_PACKAGE_libblkid=y
CONFIG_PACKAGE_libbz2=y
CONFIG_PACKAGE_libcares=y
CONFIG_PACKAGE_libcurl=y
CONFIG_PACKAGE_libdbus=y
CONFIG_PACKAGE_libevent2=y
CONFIG_PACKAGE_libexpat=y
CONFIG_PACKAGE_libext2fs=y
CONFIG_PACKAGE_libfreetype=y
CONFIG_PACKAGE_libiio=y
CONFIG_PACKAGE_libipset=y
CONFIG_PACKAGE_libiwinfo-lua=y
CONFIG_PACKAGE_liblua=y
CONFIG_PACKAGE_libmbedtls=y
CONFIG_PACKAGE_libmnl=y
CONFIG_PACKAGE_libmosquitto-ssl=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libnetsnmp=y
CONFIG_PACKAGE_libnghttp2=y
CONFIG_PACKAGE_libopenssl=y
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_libreadline=y
CONFIG_PACKAGE_librpc=y
CONFIG_PACKAGE_librt=y
CONFIG_PACKAGE_libsensors=y
CONFIG_PACKAGE_libsysfs=y
CONFIG_PACKAGE_libubus-lua=y
CONFIG_PACKAGE_libuci-lua=y
CONFIG_PACKAGE_libusb-1.0=y
CONFIG_PACKAGE_libustream-openssl=y
CONFIG_PACKAGE_libuuid=y
CONFIG_PACKAGE_libxml2=y
CONFIG_PACKAGE_lldpd=y
CONFIG_PACKAGE_lsof=y
CONFIG_PACKAGE_lua=y
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-attendedsysupgrade=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-json=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-rpc=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-ssl-openssl=y
CONFIG_PACKAGE_luci-theme-bootstrap=y
CONFIG_PACKAGE_mtr=y
# CONFIG_PACKAGE_odhcpd is not set
CONFIG_PACKAGE_openssl-util=y
CONFIG_PACKAGE_rpcd=y
CONFIG_PACKAGE_rpcd-mod-attendedsysupgrade=y
CONFIG_PACKAGE_rpcd-mod-packagelist=y
CONFIG_PACKAGE_rpcd-mod-rrdns=y
CONFIG_PACKAGE_snmpd=y
CONFIG_PACKAGE_ss=y
CONFIG_PACKAGE_strace=y
CONFIG_PACKAGE_sysfsutils=y
CONFIG_PACKAGE_tc=y
CONFIG_PACKAGE_tcpdump=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_uclibcxx=y
CONFIG_PACKAGE_uhttpd=y
CONFIG_PACKAGE_uhttpd-mod-ubus=y
CONFIG_PACKAGE_usbutils=y
CONFIG_PACKAGE_wpad=y
# CONFIG_PACKAGE_wpad-mini is not set
CONFIG_PACKAGE_zlib=y
CONFIG_PKG_CC_STACKPROTECTOR_STRONG=y
# CONFIG_PKG_FORTIFY_SOURCE_1 is not set
CONFIG_PKG_FORTIFY_SOURCE_2=y
# CONFIG_TARGET_ROOTFS_LN_VAR_TMP is not set
CONFIG_TARGET_ROOTFS_PARTSIZE=512
CONFIG_TARGET_ROOTFS_SQUASHFS=y
CONFIG_TARGET_SQUASHFS_BLOCK_SIZE=256
CONFIG_WPA_SUPPLICANT_INTERNAL=y
```

stintel, 2017-10-20:
Similar problem: https://github.com/cocodelabs/znc-palaver/issues/23
This points to a missing icu dependency. Maybe the libxml2 or libiio detects ICU on the host, causing this problem. I will investigate further.

stintel, 2017-10-20:
At first I thought libxml2 detected host ICU and enabled it due to that, but it seems the problem is really with libiio using host libxml2 headers rather than from staging_dir. This is clearly shown in the compile error:

```
[ 32%] Building C object CMakeFiles/iio.dir/xml.c.o
In file included from /usr/include/libxml2/libxml/parser.h:810:0,
                 from /usr/include/libxml2/libxml/globals.h:18,
                 from /usr/include/libxml2/libxml/threads.h:35,
                 from /usr/include/libxml2/libxml/xmlmemory.h:218,
                 from /usr/include/libxml2/libxml/tree.h:1307,
                 from /home/stijn/Development/LEDE/source/build_dir/target-arm_arm1176jzf-s+vfp_musl_eabi/libiio-0.10/xml.c:23:
/usr/include/libxml2/libxml/encoding.h:31:26: fatal error: unicode/ucnv.h: No such file or directory
compilation terminated.
```

mhei, 2017-10-20:
Thanks for digging into this!

closed 2017-10-20 by mhei

stintel, 2017-10-20:
You're welcome. And thank you for the quick merge ;-)

PR #4981 (2017-10-20): libxml2: install cmake module
Maintainer: @mhei 
Compile tested: brcm2708 on LEDE master r5106.

Description:
Since version 2.9.2, libxml2 provides its own cmake module. Install it
to avoid packages built with cmake and requiring libxml2 to fall back to
cmake's FindLibXml2 module, which seems to detect host libxml2 instead
of the one in staging_dir.

Fixes #4917.
commit 6d4f6a31 (2017-10-20): libxml2: install cmake module

Since version 2.9.2, libxml2 provides its own cmake module. Install it
to avoid packages built with cmake and requiring libxml2 to fall back to
cmake's FindLibXml2 module, which seems to detect host libxml2 instead
of the one in staging_dir.

Fixes #4917.