openwrt/packages #4918: mosquitto-ssl

LeoRX, 2017-10-08:
Maintainer: @karlp
Environment: LEDE Reboot SNAPSHOT r5017-2cb8f0a / LuCI Master (git-17.277.69962-a876daa)

Description:

mosquitto-ssl not available in LEDE snapshot repo.  I installed the mosquitto-ssl 1.4.14-1 and found it looking for libwebsocket.so.8.1

created symlink to libwebsockets.so resolved the issue.  I'm not sure where it reference to libwebsockets.so.8.1.
LEDE snapshot R49xx and R5xxx have libwebsockets.so.10 instead.  this is my first issue logged. hopfully this is the right place to log the issue.

karlp, 2017-10-09:
I'm not quite sure I understand.  mosquttio-ssl is available  here for instance: http://downloads.lede-project.org/snapshots/packages/mipsel_24kc/packages/mosquitto-ssl_1.4.14-1_mipsel_24kc.ipk

Further, downloading and extracting that ipk and running readelf shows it looking for the libwebsockets.so.10:
 0x00000001 (NEEDED)                     Shared library: [libwebsockets.so.10]

Is it possible you have an old lede snapshot install and are trying to update individual packages?  Snapshots are rolling, and libwebsockets and mosquitto have been updated multiple times, are you just out of sync somewhere?

LeoRX, 2017-10-09:
My apologies..  I didn't  think to check the other instruction set.  Not sure why it is not available  for
arm_cortex_a9_vfpv3...  I'll check to see if there are other packages missing and report to LEDE.
Index of /snapshots/packages/arm_cortex-a9_vfpv3/packages/<http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/packages/>
File Name File Size Date../-Thu Sep 14 22:38:20 2017: Packages: 1072.8 KB: Thu Sep 14 22:38:37 2017: Packages.gpg: 0.8 KB: Thu Sep 14 22:38:20 2017: Packages.gz
downloads.lede-project.org

karlp, 2017-10-10:
Ok, I can see that it failed to build for a9-vfp3, I'm looking into it.

karlp, 2017-10-19:
Update on this, jow hypothesized that the libwebsockets-cyassl and liobwebsockets-openssl builds are causing mismatched openssl "compatbilitity" headers to be installed into the staging dir.  No idea why it's only on the cortex-a9 tree.  I poked it again yesterday trying to reproduce it, but I'm currently leaning towards dropping the libwebsockets-cyassl/wolfssl and it's "openssl compatibility" layer, and move to the newer version of libwebsockets, which supports mbedtls instead, and have the two options mbedtls and openssl

LeoRX, 2017-10-24:
either way whichever is easier for you. I can work around it

karlp, 2017-10-24:
I actually tested out both the mbedtls and openssl versions locally yesterday, I'll probably push that as an update later today or tomorrow, and then just cros my fingers that the build issue resolves itself that way.

LeoRX, 2017-10-24:
Thank you Karl.  Appreciate you spending time on it.

karlp, 2017-10-26:
ok. so.... "mosquitto" has now built ok for your target :)

but... libwebsockets-full is now broken.  (libwebsockets-openssl and libwebsockets-mbedtls both seem to work fine)

I'm going to leave this open for a bit longer

karlp, 2017-10-26:
ok, the libwebsockets-full issue is a LWS issue, reported upstream.  mosquitto is back up again, I'm closing this.

closed 2017-10-26 by karlp