openwrt/packages #4931: addblock - pleas add filter to most important sites (Bitcoin Mining)

build000, 2017-10-12:
@dibdot 
Blocking Web Browser Bitcoin Mining 
https://raw.githubusercontent.com/hoshsadiq/adblock-nocoin-list/master/hosts.txt

dibdot, 2017-10-12:
To use this block list, please add this to your config (untested!):

```
config source 'bitcoin'
	option enabled '0'
	option adb_src 'https://raw.githubusercontent.com/hoshsadiq/adblock-nocoin-list/master/hosts.txt'
	option adb_src_rset '\$0 ~/^0\.0\.0\.0[ \t]+([A-Za-z0-9_-]+\.){1,}[A-Za-z]+/{print tolower(\$2)}'
	option adb_src_desc 'focus malicious bitcoin mining sites, infrequent updates, approx. 15 entries'

```

build000, 2017-10-12:
thx., added, working fine.

dibdot, 2017-10-22:
@build000 please close this ticket, changes has been applied.

build000, 2017-10-22:
very sorry - I forgot and in fact is already added filter - I'm already closing :+1: D

closed 2017-10-22 by build000